olikraus/u8g2 #829: Did micropython/python  support?

cx4, 2019-02-25:
i use micropython and python want use u8g2.

olikraus, 2019-02-25:
This projects hosts the u8g2 Arduino Library. 
It least, I did not do any port to micropython.

closed 2019-04-07 by olikraus

cskurdal, 2020-02-17:
@olikraus what level of difficulty do you think it'd be to port this to MicroPython? MP can call C/C++, so might be just updating the classes/methods?

olikraus, 2020-02-17:
I do not know MP and I also do not know the C/C++ interface, so it is difficult to tell.
At least I can try to answer questions regarding the U8g2 API, but I am not able to do any kind of MP integration.

cskurdal, 2020-02-23:
I found and SSD1306 Python driver that works great with the Heltec Wifi Kit 8 https://raw.githubusercontent.com/micropython/micropython/master/drivers/display/ssd1306.py