olikraus/u8g2 #830: How to add more character in Korean

kotran88, 2019-02-26:
Hello :)

I'm now using u8g2_font_unifont_t_korean1 very well 
and wonder how to add more character. 
I"m testing it and found out that character "메", AND "테" is not exist. 
I looked into code ug82_fonts.c and found const uint8_t u8g2_font_unifont_t_korean1 

but don't know how to add above missing chracter on code so that I can print those characters. 

Thank you !

olikraus, 2019-02-26:
Thanks for your feedback.

Please note, that there is u8g2_font_unifont_t_korean2 also available, which should contain more glyphs.

If you want to extend korean1, then the best would be to tell me the required unicodes.
I then will extend this list here: https://github.com/olikraus/u8g2/blob/master/tools/font/build/korean1.map

And rerun the font generation process.

kotran88, 2019-02-27:
Thank you olikraus ! 
missing chracter is below
"카" : %uCE74
 "메" : %uBA54
 "테" :  %uD14C
"쉐" : %uC250
 "듀" : %uB4C0
"쇠": %uC1E0
"케": %uCF00

kotran88, 2019-02-27:
and I can't use  u8g2Fonts.setFont(u8g2_font_unifont_t_korean2);
because when I compile it, 
it cause error

c:/users/kotra/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: address 0x40001840 of C:\Users\kotra\AppData\Local\Temp\arduino_build_458852/displaykorean.ino.elf section `.bss' is not within region `dram0_0_seg'

c:/users/kotra/appdata/local/arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: address 0x40001840 of C:\Users\kotra\AppData\Local\Temp\arduino_build_458852/displaykorean.ino.elf section `.bss' is not within region `dram0_0_seg'

collect2.exe: error: ld returned 1 exit status

olikraus, 2019-03-01:
yes, sure, it will be too large

olikraus, 2019-03-02:
Ceated new beta 2.26.1

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

closed 2019-05-05 by olikraus