olikraus/u8g2 #832: Wiring for an JLXST75256

GavinBrownAus, 2019-03-01:
Hi everyone, just checking in to see if anyone has the wiring for a JLX256160 LCD. I'm hooking it up to a Teensy 3.6 in I2C mode. Any help is greatly appreciated.

olikraus, 2019-03-01:
What exactly is the problem?

GavinBrownAus, 2019-03-01:
Hi Oliver, I've got no display at all...

Just checking what others are doing for wiring. I can go I2C or SPI, just
need to change a couple of resistors on the screen.

Are there any other connections required on the 20 pin connector for either
SPI or I2C?

olikraus, 2019-03-01:
You will need power for the backlight and power for the logic.
SPI requires data, clock, chip select and data/command line (which is usually refered as four-wire-SPI).
I2C only requires clock and data

You can say, that I2C is a little bit more slower, because chip select and command/data information is included in the I2C bit stream, whereas in SPI it runs in parallel on extra lines.
Advantage of I2C is you need only two lines (compared to four lines with SPI).

No other lines are required (provided, you have display module where the display is mounted on a PCB).

GavinBrownAus, 2019-03-02:
Thanks Oliver for the prompt reply.

I've tried using the 4 Wire SPI using the following constructor on a Teensy
3.6

U8G2_ST75256_JLX256160_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/
11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); // Enable U8g2 16 bit mode for
this display

But still no text on the display.

The type of display is a JLX256160G-680 - see attached photo.

I've set the resistors to SPI 1, 3 and 5.

Would you know the wiring between the display and a teensy or a arduino?

Any help would be appreciated.

Kind regards

GB

olikraus, 2019-03-02:
The wiring must match the gpio numbers given as constructor arguments. Best is to create a table which lists all connected pins between display and Arduino board.

GavinBrownAus, 2019-03-02:
Yep, thanks, got that.

I was wondering if you could share the GPIO pins that you used to connect
with the board? And what pins you used on the board/lcd?

olikraus, 2019-03-02:
I have used exactly the same pins as used in the U8g2 example files.
On the display side it will probably look like this, but it might differ for your display.
[image]

On the controller side, you have this SPI constructor:
U8G2_ST75256_JLX256160_F_4W_SW_SPI u8g2(U8G2_R0, /* clock=*/ 13, /* data=*/ 11, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8); 

So, we have the following pins:
clock
data
cs
dc
reset

In the second step, you need to select pins on your uC board. Lets take the ones from the constructor:
clock 	13
data 	11
cs 		10
dc 		9
reset 	8

Now you need to figure out the corresponsing pin on the display. This might require some reading in u8g2 docs and datasheet docs...

clock 	13	SCK		18
data 	11	SDA		15, 16, 17
cs 		10	CS		10
dc 		9	A0 (RS)	8
reset 	8	RST		9

Well, not sure whether this helps. So this is the table for my own 256x64 display. You may probably come out with a different table. So at the end, you have to create such a table for your setup, which includes the display and the uC board selected by you.

olikraus, 2019-03-02:
BTW: There is a nice video on this topic (different display, similar problem): https://www.youtube.com/watch?v=4CD8ERaylmY

GavinBrownAus, 2019-03-02:
Thanks Oliver, thats exactly my connections as well, still no display.
Looks like I may a dead display.

I'll contact the seller to get another display.

Thanks for your help.

olikraus, 2019-03-02:
Hmmm... Do the logic levels fit together?

GavinBrownAus, 2019-03-02:
Yes, just followed through all levels and checked the connectivity between
the 26 pin fpc connector and the 20 pin connector including the resistor
configuration for serial. Still no display.

All voltages are correct as well.

I can get the back light to work and thats about it.

I might put a scope on it a little later today to see what signals are
coming through... However I am leaning towards the display being dead.

GavinBrownAus, 2019-03-02:
Scrap that... Looks like I had pin 13 on the teensy wrong... Now have
something on the display !!!

Yay!

GavinBrownAus, 2019-03-03:
Hi Oliver, last one. Do you know if this display supports
setDisplayRotation ?

olikraus, 2019-03-03:
Yes, setDisplayRotation should be supported, but you should better change the first argument of the constructor.

closed 2019-04-07 by olikraus