olikraus/u8g2 #833: Cannot convert to U8X8 format with bdfconv

ShmulikKraus, 2019-03-01:
I am probably doing something wrong. A nudge in the right direction will be much appreciated.
An example:
`bdfconv.exe -v -f 2 -m "32-127" ../bdf/amstrad_cpc_extended.bdf -o amstrad_cpc_extended_f.c  -n u8x8_font_amstrad_cpc_extended_f -d ../bdf/amstrad_cpc_extended.bdf`

Result: bdfconv either hangs or finishes without error.

olikraus, 2019-03-01:
I only use the linux version of bdfconv, which works correctly with this parameters.
Maybe I should rebuild the windows exe.

ShmulikKraus, 2019-03-01:
Thank you Oli, I'll wait to see if you'll get some free time to do that, because even though I could try to use a Live CD with a virtual machine, I know nothing about Linux so it'll be a slow process.

ShmulikKraus, 2019-03-01:
This might be a stupid question, but where can I download the Linux binary? I'd like to try it with Knoppix.

ShmulikKraus, 2019-03-01:
I tried compiling the code on my WINDOWS 10 machine.
I thing the problem is somewhere here:
`for( i = 0; i <= 0x0ffff; i+=glyphs_per_line )
  {
    x = tga_draw_font_line(y, i, bf_desc_font, bf, glyphs_per_line);
    if ( x > 0 )
    {
      if ( xmax < x )
	xmax = x;
      y +=  tga_get_line_height(bf_desc_font, bf)+1;
      //bf_Log(bf, "Draw TGA, encoding %04x, y %d", i, y);
    }
  }`
After I uncomment the line:
`bf_Log(bf, "Draw TGA, encoding %04x, y %d", i, y);`
this is what I see in stdout:
`C:\bdfconv>bdfconv.exe -v -f 2 -m "32-127" amstrad_cpc_extended.bdf -o amstrad_cpc_extended_f.c  -n u8x8_font_amstrad_cpc_extended_f -d amstrad_cpc_extended.bdf
Parse File amstrad_cpc_extended.bdf: 228 glyph(s) found
Map: map_cmd_list='32-127'
Map: exclude=0 from=32/$20 to=127/$7f map=32/$20
Reduce: Start
Reduce: End
CalculateMaxBBX: x=0, y=0, w=8, h=7
CalculateMaxBBX: Encodings x=32, y=32, w=42, h=33
bf_CalculateMinMaxDWidth: dx_min=3, dx_max=9
bf_CalculateMinMaxDWidth: x_min=0, x_max=3
bf_CalculateMaxBitFieldSize: bbx.x=1, bbx.y=1, bbx.w=4, bbx.h=4, dwidth=5
Generate8x8Font: Tile size width=1 height=1
Generate8x8Font: Font size 764
Monospace Statistics: Max width extention 8, average width extention 2.0
Note: For font format 2 BBX mode has been set to 3
Draw TGA, line height 8
Draw TGA, encoding 0000, y 45
Draw TGA, encoding 0070, y 54

C:\bdfconv>pause
Press any key to continue . . .`

olikraus, 2019-03-01:
So it works?
If not, just use 
```
bdfconv.exe -v -f 2 -m "32-127" ../bdf/amstrad_cpc_extended.bdf -o amstrad_cpc_extended_f.c -n u8x8_font_amstrad_cpc_extended_f 
```
without the -d argument.

ShmulikKraus, 2019-03-01:
Thanks again Oli.
I did not realize that the "tga_draw_font" function was part of the overview picture generation. I thought it was used to convert a bitmap font to the u8g2 encoding.

Without the overview picture parameter it seems to work. I guess I could live without the image... Actually, if it is the same image as U8g2 then I could just use "-f 1" (which does work).

By the way, by the time you posted your comment I already figured how to compile and run the program with Linux but I couldn't get it to convert there either.
If it matters to you, here is the output:

`
knoppix@Microknoppix:~/Downloads/u8g2-master/tools/font/bdfconv$ ./bdfconv -v -f 2 -m '32-127' amstrad_cpc_extended.bdf -o amstrad_cpc_extended_f.c  -n u8x8_font_amstrad_cpc_extended_f -d amstrad_cpc_extended.bdf
Parse File amstrad_cpc_extended.bdf: 228 glyph(s) found
Map: map_cmd_list='32-127'
Map: exclude=0 from=32/$20 to=127/$7f map=32/$20
Reduce: Start
Reduce: End
CalculateMaxBBX: x=0, y=0, w=8, h=7
CalculateMaxBBX: Encodings x=32, y=32, w=42, h=33
bf_CalculateMinMaxDWidth: dx_min=3, dx_max=9
bf_CalculateMinMaxDWidth: x_min=0, x_max=3
bf_CalculateMaxBitFieldSize: bbx.x=1, bbx.y=1, bbx.w=4, bbx.h=4, dwidth=5
Generate8x8Font: Tile size width=1 height=1
Generate8x8Font: Font size 764
Monospace Statistics: Max width extention 8, average width extention 2.0
Note: For font format 2 BBX mode has been set to 3
Draw TGA, line height 8
Segmentation fault
knoppix@Microknoppix:~/Downloads/u8g2-master/tools/font/bdfconv$
`

OK, I'm off to try to covert some fonts.
Thanks for your help and for your great library.

ShmulikKraus, 2019-03-01:
It seems that my joy was short-lived, as I cannot seem to be able to use my newly compiled font.
I'm probably at fault again.
Can you look at my test code and tell me what I'm doing wrong?

`
/**************************************************************************
u8g2 v2.25.10 (https://github.com/olikraus/u8g2/)
 **************************************************************************/

#include <Arduino.h>
#include <U8x8lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif

#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8X8_SSD1306_128X64_NONAME_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE); 

/*
  Fontname: -FreeType-Amstrad CPC extended-Medium-R-Normal--8-80-72-72-P-64-ISO10646-1
  Copyright: Copyright ruboku 2008
  Glyphs: 95/228
  BBX Build Mode: 3
*/
const uint8_t u8x8_font_amstrad_test[762] U8X8_FONT_SECTION("u8x8_font_amstrad_test") = 
  " ~\0\0\0\0\0\0\0\0\0\0\0__\0\0\0\0\7\7\0\7\7\0\0\24\177\177\34\177\177"
  "\24\0\0$*\177\177*\22\0Ff\60\30\14fb\0\60zO]\67zH\0\0\0\0\7\7\0"
  "\0\0\0\0\34>cA\0\0\0\0Ac>\34\0\0\10*>\34\34>*\10\0\10\10>>\10"
  "\10\0\0\0\200\340`\0\0\0\0\10\10\10\10\10\10\0\0\0\0``\0\0\0`\60\30\14\6\3"
  "\1\0>\177QIE\177>\0\0@B\177\177@@\0\0r{IIof\0\0\42aII\177"
  "\66\0\30\24R\177\177P\20\0\0'oIIy\63\0\0>\177II{\62\0\0\3\1q}\17"
  "\7\0\0\66\177II\177\66\0\0&oII\177>\0\0\0\0ll\0\0\0\0\0\200\354l\0"
  "\0\0\0\10\34\66cA\0\0\0$$$$$$\0\0Ac\66\34\10\0\0\0\2\3QY\17"
  "\6\0>\177A]]_\36\0\0|~\23\23~|\0A\177\177II\177\66\0\34>cAAc"
  "\42\0A\177\177Ac>\34\0A\177\177I]Ac\0A\177\177I\35\1\3\0<~CAQs"
  "r\0\0\177\177\10\10\177\177\0\0AA\177\177AA\0\60p@A\177\77\1\0A\177\177\10\34w"
  "c\0A\177\177A@`p\0\177\177\16\34\16\177\177\0\177\177\6\14\30\177\177\0\34>cAc>"
  "\34\0A\177\177I\11\7\6\0<~CQ\63n\134\0A\177\177\11\31\77f\0\0&oII{"
  "\62\0\0\3A\177\177A\3\0\0\77\177@@\177\77\0\0\37\77``\77\37\0\177\177\60\30\60\177"
  "\177\0as\36\14\36sa\0\0\7OxxO\7\0GcqYMgs\0\0\0\177\177AA"
  "\0\0\1\3\6\14\30\60`\0\0\0AA\177\177\0\0\10\14\6\3\6\14\10\0\200\200\200\200\200\200"
  "\200\200\0\0\2\6\14\10\0\0 tTT<x@\0C\77\177DD|\70\0\0\70|DDl"
  "(\0\70|DE\77\177@\0\0\70|TT\134\30\0\0H~\177I\3\2\0\0\230\274\244\244\374"
  "|\0A\177\177\10\4|x\0\0\0D}}@\0\0\0`\340\200\204\374}\0A\177\177\20\70l"
  "D\0\0\0A\177\177@\0\0x|\14\70\14|x\0\4|x\4\4xx\0\0\70|DD|"
  "\70\0\204\374\370\244$<\30\0\30<$\244\370\374\204\0D|xD\14\10\0\0\0H\134TTt"
  " \0\0\4\77\177Dd \0\0<|@@||\0\0\34<``<\34\0<|`\70`|"
  "<\0Dl\70\20\70lD\0\0\234\274\240\240\374|\0\0Ldt\134LD\0\0AAw>\10"
  "\10\0\0\0\0\177\177\0\0\0\0\10\10>wAA\0\2\3\1\3\2\1\1";

void setup() {
    u8x8.begin();
    u8x8.setPowerSave(0);
    u8x8.clear();   
}

void loop() {
      u8x8.setFont(u8x8_font_amstrad_test);
      u8x8.drawString(0, 0, "Hello!%* 123");
      delay(5000);
}
`

olikraus, 2019-03-01:
We

ShmulikKraus, 2019-03-02:
I understand.
It's a shame, I was hoping to compile a U8x8 font with my language.

If I just give you the bdf file and tell you the parameters, would it be quick and easy for you to just convert it for me? (Assuming you are willing to do that)

olikraus, 2019-03-02:
Sure, I can try

ShmulikKraus, 2019-03-02:
Much obliged.
By the way, I made some progress. I think the problem is in the header of the converted font.
For example, in my previous test of the amstrad_cpc, the first few bytes are incorrect:
`
("u8x8_font_amstrad_test") =
" ~\0\0\0\0\0...
`
If I change it to the first bytes of the built-in font (from u8x8_fonts.c), E.G.:
`
("u8x8_font_amstrad_test") =
" \377\1\1\0\0\0\0\0...
`
Then it works.
(Naturally, I also increase the size of the array to accommodate the new characters)

However, there is still a problem with utf-8 text, but first I'll explain what I'm trying to do:
I'm trying to make a u8x8 font with only English, Hebrew and a few special characters. I've converted the file "PxPlus_TandyNew_TV.ttf" to bdf:
[PxPlus_TandyNew_TV.zip](https://github.com/olikraus/u8g2/files/2922020/PxPlus_TandyNew_TV.zip)

And used bdfconv like so:
`
bdfconv.exe -v -f 2 -m "32-127,1488-1518" PxPlus_TandyNew_TV.bdf -o u8x8_font_pptntv_heb.c  -n u8x8_font_pptntv_heb -d PxPlus_TandyNew_TV.bdf
`
[image]

But after the conversion when trying to print anything with this font nothing is displayed on the LCD.
However, if I change the header to `" \377\1\1\0\0...` English/number/special characters work, but Hebrew does not.

E.G. this command:
`u8x8.drawUTF8(0, 0, "Hello! *-+,");`
Will be displayed correctly.
But this:
`u8x8.drawUTF8(0, 0, "!םלוע םולש");`
Will display the exclamation mark followed by some garbled dots.

I should mention that if I convert the same bdf file to u8g2 font instead of u8x8, everything works perfectly.

Can you try the conversion with my parameters and see if it works?

olikraus, 2019-03-05:
I have rebuild the windows exe.

Regarding the u8x8, there is a kind of an issue. Although utf-8 is supported, u8x8 font format is limited to codes 0 to 255. What you could do is, to map the chars into the range, starting from 128.
Moreover, the bdf output is somehow broken. I probably should fix this once.

So a suitable command could be:
```
bdfconv -v -f 2 -m "32-127,1488-1518>128" PxPlus_TandyNew_TV.bdf -o u8x8_font_pptntv_heb.c -n u8x8_font_pptntv_heb
```

After this mapping, in order to output "Alef" (1488) you need to output char 128 ("\x80"). In the same way "Beth" (1489) is mapped to code 129 ("\x81").

Here is the code:

```
/*
  Fontname: -FreeType-PxPlus TandyNew TV-Medium-R-Normal--8-80-72-72-P-70-ISO10646-1
  Copyright: Outline (vector) version (c) 2015 VileR
  Glyphs: 123/781
  BBX Build Mode: 3
*/
const uint8_t u8x8_font_pptntv_heb[988] U8X8_FONT_SECTION("u8x8_font_pptntv_heb") = 
  " \232\1\1\0\0\0\0\0\0\0\0\0\0\6__\6\0\0\0\3\7\0\0\7\3\0\24\177\177\24"
  "\177\177\24\0\0$.kk:\22\0Ff\60\30\14fb\0\60zO]\67zH\0\0\0\4\7"
  "\3\0\0\0\0\0\34>cA\0\0\0\0Ac>\34\0\0\10*>\34\34>*\10\0\10\10>"
  ">\10\10\0\0\0\200\340`\0\0\0\0\10\10\10\10\10\10\0\0\0\0``\0\0\0`\60\30\14"
  "\6\3\1\0\34>cIc>\34\0\0@B\177\177@@\0BcqYIof\0\42cII"
  "I\177\66\0\30\34\26S\177\177P\0/oIIIy\61\0<~KIIx\60\0\3\3qy"
  "\15\7\3\0\66\177III\177\66\0\6OIIi\77\36\0\0\0\0ff\0\0\0\0\0\200\346"
  "f\0\0\0\0\0\10\34\66cA\0\0$$$$$$\0\0\0Ac\66\34\10\0\2\3\1Y"
  "]\7\2\0>\177A]]\37\36\0|~\13\11\13~|\0A\177\177II\177\66\0\34>cA"
  "Ac\42\0A\177\177Ac>\34\0A\177\177I]Ac\0A\177\177I\35\1\3\0\34>cA"
  "Q\63r\0\177\177\10\10\10\177\177\0\0\0A\177\177A\0\0\60p@A\177\77\1\0A\177\177\10"
  "\34wc\0A\177\177A@`p\0\177\177\16\34\16\177\177\0\177\177\6\14\30\177\177\0>\177AA"
  "A\177>\0A\177\177I\11\17\6\0>\177AA\341\377\276\0A\177\177\11\31wf\0\0\42gM"
  "Ys\42\0\0\3A\177\177A\3\0\77\177@@@\177\77\0\37\77`@`\77\37\0\77\177`\70"
  "`\177\77\0cw\34\10\34wc\0\0\7OxxO\7\0GcqYMgs\0\0\0\177\177"
  "AA\0\0\1\3\6\14\30\60`\0\0\0AA\177\177\0\0\10\14\6\3\6\14\10\0\200\200\200\200"
  "\200\200\200\200\0\0\1\3\6\4\0\0 tTT<x@\0A\177\77DD|\70\0\70|DD"
  "Dl(\0\70|DE\77\177@\0\70|TTT\134\30\0H~\177I\11\3\2\0\230\274\244\244"
  "\370|\4\0A\177\177\10\4|x\0\0\0D}}@\0\0\0`\340\200\200\375}\0A\177\177\20"
  "\70lD\0\0\0A\177\177@\0\0||\14x\14|x\0\4|x\4\4|x\0\70|DD"
  "D|\70\0\204\374\370\244$<\30\0\30<$\244\370\374\204\0D|xL\4\14\10\0H\134TT"
  "Tt$\0\4\4\77\177Dd \0<|@@<|@\0\34<`@`<\34\0<|`\70"
  "`|<\0Dl\70\20\70lD\0\234\274\240\240\240\374|\0\0Ldt\134Ld\0\0\10\10>"
  "wAA\0\0\0\0\357\357\0\0\0\0AAw>\10\10\0\2\3\1\3\2\3\1\0pxLF"
  "Lxp\0r~\14\30\60~N\0BBBB~|@\0@@b\62\36|@\0\2\2\2\2"
  "~~\2\0rr\2\2\2~|\0\0\2\2~|\0\0\0\0\0Bn>\22\0\0z~\6\2"
  "\2~|\0>~@Lb>\34\0\0\0\2\22\36\14\0\0\2\2\2\2\362\376\214\0BBBB"
  "B~<\0\3\3\2br\36\14\0z~FBB~|\0r~\14FB~|\0\0\0\2\372"
  "\376\204\0\0\0@BB~|\0\0:~FBb>\34\0@^~`@~>\0\32\36\22\2"
  "\376\374\200\0Z^RBB~|\0\0\376\376\220\20\36\16\0\0BFLX~f\0\2\372\372\2"
  "\62>,\0\2\2\2\2\2~|\0>~P^@~>\0b~~\2\2~|";
```

All in all, it would be better to use u8g2 API, which supports your glyph without remapping.
There are several hebrew fonts available:

u8g2_font_6x13_t_hebrew
u8g2_font_6x13B_t_hebrew
u8g2_font_cu12_t_hebrew
u8g2_font_unifont_t_hebrew

For example "u8g2_font_6x13_t_hebrew":

[image]

ShmulikKraus, 2019-03-08:
Thanks Oliver, the converted font with the new bdfconv works, but the new version still fails when used with the -d parameter.
For me it's OK because I can generate the image with the parameter "-f 1" but it's still technically a bug so I don't know if I should close this issue or not.
What do you think?

> All in all, it would be better to use U8g2 API, which supports your glyph without remapping.

I have used u8g2 successfully before but in this project I have a rotary encoder and I need the fastest LCD printing speed possible, so that I can rotate the encoder quickly and not miss any "clicks".
Also, U8g2 eats up a lot of flash and ram and I might need the memory for my code (haven't finished writing it yet, I'm waiting for some parts to arrive from China).

olikraus, 2019-03-08:
Let's leave it open: fix bug with -d arg for bdfconv.

olikraus, 2019-04-07:
problem was, that the overview picture generator doesn't support font format 2. This is now handled correctly.

closed 2019-04-07 by olikraus

jie326513988, 2021-12-31:
[image]
Hello, I use your latest version of bdfconv on Windows 10 Exe, still unable to convert, can you help me see, thank you!

olikraus, 2021-12-31:
Looks like an internal error. Can you provide the bdf file?

jie326513988, 2022-01-01:
> 看起来像是内部错误。你能提供bdf文件吗？

Thank you very much for your reply. I downloaded the font file on this website

https://stncrn.github.io/u8g2-unifont-helper/

olikraus, 2022-01-01:
For me it works without problem.
I have attached the result

[out.c.gz](https://github.com/olikraus/u8g2/files/7797585/out.c.gz)

jie326513988, 2022-01-01:
Thank you. I don't know what's wrong. I've changed two computers and still make mistakes.
This is some of my attempts. The mistakes are between 34000 and 35000.
An error occurs when executing 20-34349
No problem executing 20-34348
No problem when executing 34348-45000

> 对我来说它没有问题。 我附上了结果
> 
> [输出.c.gz](https://github.com/olikraus/u8g2/files/7797585/out.c.gz)

olikraus, 2022-01-01:
Did you use the latest exe https://github.com/olikraus/u8g2/blob/master/tools/font/bdfconv/bdfconv.exe ?

jie326513988, 2022-01-02:
Yes, I can make sure it's up to dat
[image]
e