olikraus/u8g2 #834: 128x128_F_4W

BastiaanMDC, 2019-03-06:
Hi Mr Kraus,

thanks for the great library! 

i have been reading a few forums about speeding up the refresh rate of the screen. as far as I understand the u8g2 uses a digitalWrite for the Arduino Community. and that Refreshrates can be achieved till 70Hz.

Speed issues/Understanding Issues:
for our project I am using a teensy3.6,  a SSD1327 in SPI Full Buffer Mode to drive 5 Seperate PWM outputs for leds.

what I would like to learn achieve and accomplish is a fast refreshrate that can be controlled.
on the screen I have the mapped potentiometer values are shown and should respond to a change if turned.

on the screen I have the words Temp, LED1,2,3,4,5 which should not change. They should change only if the Temperature is higher than 115C°. (Warning of some sort).

Temp

LED1 (Animated Bar)
LED2 (Animated Bar)
LED3 (Animated Bar)
LED4 (Animated Bar)
LED5 (Animated Bar)

how can I control the refresh rate of the animated Bars and Temperature values on the screen seperately?

`//check if changed and update accordingly if not do nothing. (I technically require a another if statement in the if statement I think based on Millis old State and new State to create a delay on the refresh, but here I am stuck. because I dont know what the SPI Transfer is exactly doing at this moment or at all,

in this example you explain Mr. Andrasfeher https://github.com/olikraus/u8g2/issues/636
how set the Brightness by using the send and receive over the u8x8
would it be faster for the Teensy3.6 to SPI to the SSD1327 screen if the
the u8x8.StartTransfer(u8g2.getu8x8()) and inbetween those lines (a buffer is send over to the screen accordingly holding the below given code?,if so how do I do this?) 

>(giving a call would greatly help to, if your based in Germany that is.)

if(OldReadout_pinA5!=Readout_pinA5)
   {
     u8g2.drawBox(35,30,Bar_LED0,8);
   
   }
OldReadout_pinA5=Readout_pinA5;
`
the code for the screen.
` 
U8G2_SSD1327_MIDAS_128X128_F_4W_SW_SPI u8g2(U8G2_R2, 13, 11, 10, 12, 38);// clock= 13,  data= 11,  cs=10 ,  dc=11,  reset= 8);

void Setup()
{
}
void Loop()
{
       u8g2.firstPage();
  do {
     u8g2.clearBuffer();

    u8g2.setFont(u8g2_font_pressstart2p_8u);
    u8g2.drawStr(0,10,"TEMP"); //x,y,String
    u8g2.setCursor(50,10);
    if(Convers_bitsToTemp<130)
    {
    u8g2.print(Convers_bitsToTemp);
    }
    u8g2.drawStr(0,40,"LED1"); //x,y,String
    u8g2.drawStr(0,55,"LED2"); //x,y,String
    u8g2.drawStr(0,70,"LED3"); //x,y,String
    u8g2.drawStr(0,85,"LED4"); //x,y,String
    u8g2.drawStr(0,100,"BF"); //x,y,String
    u8g2.drawStr(115,0,"C"); //x,y,String

    u8g2.drawRFrame(35,29,90,10,0);// x,y,w,h,_round
    u8g2.drawRFrame(35,44,90,10,0);// x,y,w,h,_round    
    u8g2.drawRFrame(35,59,90,10,0);// x,y,w,h,_round   
    u8g2.drawRFrame(35,74,90,10,0);// x,y,w,h,_round   
    u8g2.drawRFrame(35,89,90,10,0);// x,y,w,h,_round  
      
    if(Convers_bitsToTemp>130)
    {
      //Warning LED Driver is getting Hot
      u8g2.drawStr(0,20,"!!!! WARNING "); //x,y,String
      u8g2.drawStr(0,20,"");
      u8g2.drawStr(0,20,"LEDS GETTING HOT");
    }

//animation is Blinking.... why is that? jitter on the Analog Input?
   if(OldReadout_pinA5!=Readout_pinA5)
   {u8g2.drawBox(35,30,Bar_LED0,8);}
   if(OldReadout_pinA6!=Readout_pinA6)
   {u8g2.drawBox(35,45,Bar_LED1,8);}
    if(OldReadout_pinA7!=Readout_pinA7)
   {u8g2.drawBox(35,60,Bar_LED2,8);}
     if(OldReadout_pinA8!=Readout_pinA8)
   {u8g2.drawBox(35,75,Bar_LED3,8);}
    if(OldReadout_pinA9!=Readout_pinA9)
   {u8g2.drawBox(35,90,Bar_LED4,8);}
    u8g2.setFont(u8g2_font_unifont_t_symbols);
   // u8g2.drawGlyph(115, 0, 0x0024);  // row 0x00b0 this means the column where to find it the b0,1,2,3,4 etc gives away the horizontal position in the symbol U8G2
    u8g2.sendBuffer();
     
  } while ( u8g2.nextPage() );
    OldReadout_pinA5=Readout_pinA5;
    OldReadout_pinA6=Readout_pinA6;
    OldReadout_pinA7=Readout_pinA7;
    OldReadout_pinA8=Readout_pinA8;
    OldReadout_pinA9=Readout_pinA9;
    
}
`

again many thanks, hope to hear from you soon.
[image]

olikraus, 2019-03-06:
For increased performance, use the hardware SPI driver `U8G2_SSD1327_MIDAS_128X128_F_4W_HW_SPI`. This constructor does not receive pin numbers for clock and data lines, instead you have to use the clock and data lines to which you hardware SPI is connected.

This does not make sense:
```
u8g2.sendBuffer();

} while ( u8g2.nextPage() );
```
sendBuffer() will be called by nextPage() also. So you end up sending the buffer twice to the display (which further reduces speed).
In general: You do not need the firstPage/nextPage loop for in full buffer mode. clearBuffer() and sendBuffer() are fully ok. Please consult the example for this.

I think, if you would use HW SPI and remove the firstPage/nextPage loop, then your code will be 3x to 5x faster.

olikraus, 2019-04-07:
no further feedback, closing...

closed 2019-04-07 by olikraus