olikraus/u8g2 #835: Help with constructor please?

exaustgas, 2019-03-13:
I've got a 64x32 SSD1306 display that works (displays text) with the u8x8 library with the following constructor:

`U8X8_SSD1306_64X32_1F_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE); `

I would like to use it with the u8g2 library. Which constructor should I use? My best guess was the this one:

`U8G2_SSD1306_64X32_1F_F_HW_I2C u8g2(U8G2_R0,U8X8_PIN_NONE);`

It compiles and uploads to my UNO but the screen is blank. 

Any ideas?

Ted

olikraus, 2019-03-13:
Did you use hello world example?

exaustgas, 2019-03-13:
> Did you use hello world example?

> 
I did, maybe something not obvious is wrong? Below are two hello world sketches, the one that works with the u8x8 library first and one that does not work using the u8g2 library

`/*
 * This WORKS. 
 */
#include <Arduino.h>
#include <SPI.h>
#include <U8x8lib.h>

/* Constructor */

U8X8_SSD1306_64X32_1F_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE);
/* u8x8.begin() is required and will sent the setup/init sequence to the display */
void setup(void)
{
  u8x8.begin();
}

void loop(void)
{
  //u8x8.setFont(u8x8_font_chroma48medium8_r);    //stock font
 // u8x8.setFont(u8x8_font_inr21_2x4_f);            //my chosen font
 u8x8.setFont(u8x8_font_open_iconic_check_2x2); //just playing with fonts
  
  u8x8.drawString(0, 0, "DCBA");
  delay(1000);
  u8x8.drawString(0, 2, "DCBA");
  delay(1000);
  u8x8.drawString(0, 0, "        ");              //clears the text
  delay(1000);
  u8x8.drawString(0, 2, "        ");              //clears the text
}
`
Below does not work, the screen is just blank:

`/*
 * NOT WORKING! SCREEN BLANK! WORKS WIRED IDENTICAL USING THE U8X8 TEXT ONLY MODE!
 * diy ssd1306 .49" 64x32 oled using hardware i2c on arduino uno. the reset pin is connected to the arduino reset pin. 
 */
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SSD1306_64X32_1F_F_HW_I2C u8g2(U8G2_R0,U8X8_PIN_NONE);     //showing no text on screen. 
//U8X8_SSD1306_64X32_1F_HW_I2C u8x8(/* reset=*/ U8X8_PIN_NONE); //verified works with the U8X8 library (obv not in this sketch though). Exact same wiring above shoes no text on screen

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();					// clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
  u8g2.drawStr(0, 0, "Hello World!");	// write something to the internal memory
  u8g2.sendBuffer();					// transfer internal memory to the display
  delay(1000);
}`

olikraus, 2019-03-15:
`u8g2.drawStr(0, 0, "Hello World!");	`

The reference point of the strings is the lower left by default.

exaustgas, 2019-03-15:
Thank you, that worked! I am sure I changed it to 0,0 thinking it worked fundamentally the same as the u8x8 library, trying not to print off screen.

olikraus, 2019-03-15:
:)

closed 2019-04-07 by olikraus