olikraus/u8g2 #836: u8g2 and ucglib conflict

ditor2, 2019-03-15:
I'm not sure if this is a software problem, but I cannot use two displays together on an ESP32. First display is a 128x128 SSD1327, the second is a 128x160 St7735.
U8g2 is version 2.24.3, ucglib is 1.5.2.
I have the code below in visual micro.  

1) This shows nothing on both displays. "loop u8g2 ok" is never showing up on serial monitor, some of the u8g2 calls freeze the uc, not even watchdog resets.
2) Commenting out (setup section1 and loop section1) or (setup section2 and loop section2) is okay, both displays work alone.
3) Commenting out only the "loop section1" brings back to life the controller, there's Hello world on St7735 (but of course there' nothing on SSD1327).

So it seems that begin() of ucg conflicts the use of u8g2. What should go wrong?

```
#define SSD1327_CS 5
#define SSD1327_DC 0
#define SSD1327_RESET 4

#define ST7735_CS 25
#define ST7735_DC 15
#define ST7735_RESET 2
#define ST7735_BLK 26

#include <U8g2lib.h>
#include <Ucglib.h>

#include <analogWrite.h>

Ucglib_ST7735_18x128x160_HWSPI ucg(/*cd=*/ ST7735_DC, /*cs=*/ ST7735_CS, /*reset=*/ ST7735_RESET);
U8G2_SSD1327_MIDAS_128X128_F_4W_HW_SPI u8g2(U8G2_R0,/* cs=*/ SSD1327_CS, /* dc=*/ SSD1327_DC, /* reset=*/ SSD1327_RESET);

#define icon_wifiap_x 16
#define icon_wifiap_y 16
#define icon_wifiap_width 16
#define icon_wifiap_height 16
const uint8_t icon_wifiap[] PROGMEM = {
	0x00, 0x00, 0xF0, 0x0F, 0xFC, 0x3F, 0x0E, 0x70, 0x03, 0xC0, 0xE1, 0x87,
	0xF8, 0x1F, 0x1C, 0x38, 0x06, 0x60, 0xC2, 0x43, 0xE0, 0x07, 0x30, 0x0C,
	0x18, 0x18, 0x80, 0x01, 0x80, 0x01, 0x00, 0x00, };

void setup(void)
{
	Serial.begin(115200);

	// SETUP SECTION1
	u8g2.begin();
	// END SETUP SECTION1

	// SETUP SECTION2
	ucg.begin(0);
	ucg.clearScreen();
	// END SETUP SECTION2

	analogWriteResolution(ST7735_BLK, 12);
	analogWrite(ST7735_BLK, 128);
}

void loop(void)
{
	// LOOP SECTION1
	u8g2.clearBuffer();					// clear the internal memory
	u8g2.setFont(u8g2_font_ncenB08_tr);	// choose a suitable font
	u8g2.drawStr(0, 10, "Hello World!");	// write something to the internal memory
	u8g2.drawXBMP(icon_wifiap_x, icon_wifiap_y, icon_wifiap_width, icon_wifiap_height, icon_wifiap);
	u8g2.drawLine(0, 0, 127, 127);
	u8g2.drawLine(0, 127, 127, 0);
	u8g2.sendBuffer();					// transfer internal memory to the display
	delay(1000);
	u8g2.clearBuffer();
	u8g2.sendBuffer();
	Serial.println("loop u8g2 ok");
	// END LOOP SECTION1

	// LOOP SECTION2
	ucg.setFont(ucg_font_ncenB14_hf);
	ucg.setPrintPos(0, 25);
	ucg.setColor(255, 255, 255);
	ucg.print("Hello World!");
	delay(1000);
	Serial.println("loop ucg ok");
	// END LOOP SECTION2
}
```

olikraus, 2019-03-16:
Any error message or warning?

ditor2, 2019-03-16:
Do you mean compile time warning? No, there's no any error/warning or unusual message.

olikraus, 2019-03-17:
hmm... I have crosschecked the Ucglib API for SPI, so U8g2 and Ucglib should use the same SPI API.
So at least on a traditional original Arduino Board this should work. I can not say something about other boards, e.g. the ARDUINO internal constant must be >=10600. If for example the ESP32 does not define this variable or defines it with a value lesser than 10600, then it may not work.
Altogether: It is difficult for me to debug this...

ditor2, 2019-03-17:
"It is difficult for me to debug this" - I'm sure... :)

I have this in my build environment:
#define ARDUINO 10807
I'll try this config with another uc in 1-2 weeks. Will close if I cannot reproduce on a non-esp32 uc.

(Just another question: do you plan support for St7735 128x128 display? I have such one as well...)

olikraus, 2019-03-17:
A the moment, I don't have time to work on ucglib :-(

olikraus, 2019-04-07:
closing this for now... feel free to reopen...

closed 2019-04-07 by olikraus