olikraus/u8g2 #837: Font Size gets bigger when using Draw Color 0

danielw63, 2019-03-16:
Hey there,

I´ve got a problem which I can´t explain.
The problem is, that the chosen mode (either off, dry or wet) always appears bigger than the rest although I chose the same font for every character ... also when the select variable is true, I´m having no problems with the visualisation of "ASR" in negative color. The problem only concerns the modes ...

`               if(*select){
			u8g2_DrawRBox(&u8g2, 50, 5, 23, 13, 3);
			u8g2_SetDrawColor(&u8g2, 0);
			u8g2_DrawStr(&u8g2, 53, 15,"ASR");
			u8g2_SetDrawColor(&u8g2, 1);

			u8g2_SetFont(&u8g2, u8g2_font_unifont_t_symbols);
			switch(*modepos){
				case 0:
					//Pfeil über OFF
					u8g2_DrawGlyph(&u8g2, 25, 45, 0x23f7);	
					break;
				case 1:
					//Pfeil über DRY
					u8g2_DrawGlyph(&u8g2, 58, 45, 0x23f7);	
					break;
				case 2: 
					//Pfeil über WET
					u8g2_DrawGlyph(&u8g2, 95, 45, 0x23f7);	
					break;
			}
		}

		u8g2_SetFont(&u8g2, u8g2_font_9x15_tf);

		switch(*mode){
				case 0:
					u8g2_DrawRBox(&u8g2, 18, 44, 31, 14, 2);
					u8g2_SetDrawColor(&u8g2, 0);
					u8g2_DrawStr(&u8g2, 20, 56,"OFF");
					break;
				case 1:
					u8g2_DrawRBox(&u8g2, 51, 44, 31, 14, 2);
					u8g2_SetDrawColor(&u8g2, 0);
					u8g2_DrawStr(&u8g2, 53, 56,"DRY");
					break;
				case 2: 
					u8g2_DrawRBox(&u8g2, 88, 44, 31, 14, 2);
					u8g2_SetDrawColor(&u8g2, 0);
					u8g2_DrawStr(&u8g2, 90, 56,"WET");
					break;
		}
`
https://files.slack.com/files-pri/TDX03S3H7-FH257RW23/img_20190316_152650__01.jpg

olikraus, 2019-03-16:
I do not see the complete code, but there should be a `u8g2_SetDrawColor(&u8g2, 1);` somewhere.

danielw63, 2019-03-16:
Here you see the complete code:

`if(oled_draw_block_count > 0 && *show){
		u8g2_ClearBuffer(&u8g2);
		u8g2_SetFont(&u8g2, u8g2_font_9x15_tf);
		u8g2_SetDrawColor(&u8g2, 1);
		
		drawNavigation(RIGHT);

		u8g2_DrawStr(&u8g2, 55, 15,"DRS");
		u8g2_DrawStr(&u8g2, 20, 56,"OFF");
		u8g2_DrawStr(&u8g2, 55, 56,"ACC");
		u8g2_DrawStr(&u8g2, 88, 56,"AUTO");

		if(*select){
			u8g2_DrawRBox(&u8g2, 52, 5, 23, 13, 3);
			u8g2_SetDrawColor(&u8g2, 0);
			u8g2_DrawStr(&u8g2, 55, 15,"DRS");
			u8g2_SetDrawColor(&u8g2, 1);

			u8g2_SetFont(&u8g2, u8g2_font_unifont_t_symbols);
			switch(*modepos){
				case 0:
					//Pfeil über OFF
					u8g2_DrawGlyph(&u8g2, 21, 45, 0x23f7);	
					break;
				case 1:
					//Pfeil über ACC
					u8g2_DrawGlyph(&u8g2, 55, 45, 0x23f7);	
					break;
				case 2: 
					//Pfeil über AUTO
					u8g2_DrawGlyph(&u8g2, 92, 45, 0x23f7);	
					break;
			}
		}

		u8g2_SetFont(&u8g2, u8g2_font_9x15_tf);

		switch(*mode){
				case 0:
					u8g2_DrawRBox(&u8g2, 18, 44, 31, 14, 2);
					u8g2_SetDrawColor(&u8g2, 0);
					u8g2_DrawStr(&u8g2, 20, 56,"OFF");
					break;
				case 1:
					u8g2_DrawRBox(&u8g2, 51, 44, 31, 14, 2);
					u8g2_SetDrawColor(&u8g2, 0);
					u8g2_DrawStr(&u8g2, 55, 56,"ACC");
					break;
				case 2: 
					u8g2_DrawRBox(&u8g2, 86, 44, 40, 14, 2);
					u8g2_SetDrawColor(&u8g2, 0);
					u8g2_DrawStr(&u8g2, 88, 56,"AUTO");
					break;
		}
	}`

olikraus, 2019-03-16:
hmm.. I am also not able to access the jpg. You could also attach pictures here directly.
All in all it is not clear to me, what is exactly your problem.

danielw63, 2019-03-16:
[image]

danielw63, 2019-03-16:
My problem is, that the marked mode appears bigger than the rest, although I didn´t change anything concerning font ...

olikraus, 2019-03-16:
grummelgrummel... ich scheibe mal auf Deutsch. Also mir scheint der Code immer noch unvollständig. So sehe ich nirgends den String "WET". Ausserdem weiss ich ja nicht was in "drawNavigation()" passiert. Wird dort der Font gesetzt? Meine Empfehlung: Setze einfach vor jedem DrawString Befehl den Font. Wenn alles klappt, kannst Du die nicht benötigten setFont Befehle wieder auskommentieren.

danielw63, 2019-03-16:
Wegen dem "WET": Habe zwei Screens die im Prinzip exakt dasselbe sind, also DRS und ASR. Habe nur vom falschen Screen ein Foto gemacht ...
Aber bin mir jetzt ziemlich sicher, dass es an drawNavigation liegt ... hier wird die Schriftart verändert. Hab das Dokument so übernommen, das war mir bis jetzt nicht aufgefallen.
Am Montag kann ich nochmal testen, schreibe dann ein kurzes Update.
Vielen Dank, dir!

olikraus, 2019-04-06:
I think this is closed

closed 2019-04-06 by olikraus