olikraus/u8g2 #839: Load Only Certain Characters

AcrimoniousMirth, 2019-03-16:
Hi!
Using U8G2 on a small LED screen (128x64) with an ATMega32U4 Pro Micro.
The rest of my code is minimal and doesn't take up much space but I'm trying to get 38 and 18 high text to run and its taking up so much memory it refuses to compile.

I looked at the option of converting a **ttf** to **bdf** then using `bdfconv` and `-m map` but after messing around with a loaf of programs for a while it just seemed way more effort than its worth. Especially for my noonish little project.

I'm just wondering if there's a way to only load a few characters for reuse. I'll only ever be using `0 1 2 3 4 5 6 7 8 9 . V %` so if I could load only those in the two fonts I'd save a tonne of memory. Or is there a way to open a C file, delete out the unneeded and save as, or something?

Any advice on this? Thanks :)

olikraus, 2019-03-17:
which font do want to use?
Is this one of the existing u8g2 fonts?

AcrimoniousMirth, 2019-03-17:
@olikraus 
Hi! Yes, it’s two of them. 
I just opened up the font list to see if I could find them (at work so can’t look them up in the code). I’m noticing that there are loads of fonts available here, when I just clicked on “18 size fonts” yesterday it brought up only 4 but when I scroll down through “all fonts” today there’s loads within the “18” and “38” sections.
When I get home I’ll see about using some of these other fonts

But yes, it’s two existing fonts and I was just wondering if there’s a way to only load selected characters onto the hardware? Like of all the 256 characters per font, I just need 13!

olikraus, 2019-03-17:
> But yes, it’s two existing fonts and I was just wondering if there’s a way to only load selected characters onto the hardware? 

Basically yes, but for the precise advice, i need to know the exact name of the font.

> I just opened up the font list to see if I could find them

You mean this list? https://github.com/olikraus/u8g2/wiki/fntlistall

BTW: Most fonts are available with a different number of glyphs as described her: https://github.com/olikraus/u8g2/wiki/fntlistall#u8g2-font-names

So the last char of the font name indicates the number of chars in the font.

AcrimoniousMirth, 2019-03-17:
@olikraus 
Hey! Yeah, I took the time to find out what those suffixes meant, and searched for ones ending in **MR**.
I've just swapped to these two:
`u8g2_font_inb19_mr`
`u8g2_font_inb38_mr`

It's just under my memory limit, so that's nice, but ideally I'd like to free up more space as this program is going to evolve a bit!

AcrimoniousMirth, 2019-03-17:
I think a function like this could be quite useful:
`u8g2.IncludeOnly(u8g2_font_inb19_mr, (0,1,2,3,4,5,6,7,8,9,.,%);`
So it only loads those files onto the Arduino. Then whenever you set that font and print in it it'll only have access to those characters.
Sadly I'm way too early in my journey to do any work on it :/

olikraus, 2019-03-17:
There is no such thing as "load" on an embedded system. From where shell the "load" happen? The required glyphs have to be defined at compile/link time. This means, any procedure like your suggested "IncludeOnly" can not work. The number of glyphs per font have to be defined during compile time.

In the special case of inb there is also inb19_mn and inb38_mn with numbers only: 
[image]

However, if you need additional chars, you need to use bdfconv.

AcrimoniousMirth, 2019-03-17:
Then it sounds like an applet that simply takes a set font and does the required bdfconv on it would be great... seeing as after 3 hours of fiddling around with different programs like FontForge and several online converters, font format types etc, I ended up giving up because nothing was working.

closed 2019-03-17 by AcrimoniousMirth