olikraus/u8g2 #842: QRCode using u8x8?

Juki48, 2019-03-20:
I have a project using u8x8 to display data on a SSH1106 OLED with a pro micro (can use mega2560 if needed). I would like to put the data in a QR code for easy transfer to a smart phone. Is it possible to display a QR code using u8x8? I have been searching for an example to reference but haven't found anything.

This is the QR library I've been playing with, but I can't figure out how to get it to print to the oled.
[https://github.com/ricmoo/QRCode](url)

olikraus, 2019-03-20:
U8x8 is a simplified text only lib. U8g2 is there to display graphics.

Juki48, 2019-03-21:
Thank you. I have a basic example running now using U8g2. I was also able to strip it down to run on the pro micro, but it's tight.

closed 2019-04-07 by olikraus