olikraus/u8g2 #843: Hardware SPI on ST7920 and the D1 Mini (ESP8266)

Phipli, 2019-03-25:
Hello, I've been trying to set up a [ST7920 128x64 LCD display](https://www.banggood.com/12864-128-x-64-Graphic-Symbol-Font-LCD-Display-Module-Blue-Backlight-p-916218.html) using SPI and a [D1 Mini](https://www.banggood.com/WeMos-D1-mini-V2_2_0-WIFI-Internet-Development-Board-Based-ESP8266-4MB-FLASH-ESP-12S-Chip-p-1143874.html) with the Arduino IDE. So far, I have been able to get software SPI working using "U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ 14, /* data=*/ 13, /* CS=*/ 15, /* reset=*/ 2);".

These pins should also be suitable for hardware SPI as per [the pin-out available online](https://klovimg.com/images/2018/09/29/esp8266-wemos-d1-mini-pinout.png) but when I use "U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, /* clock=*/ 14, /* data=*/ 13, /* CS=*/ 15, /* reset=*/ 2);" instead, I just get a blank screen.

My circuit is [as per this linked example](https://thesolaruniverse.files.wordpress.com/2017/08/27_st7920_fig_01_96.png), but using the D1 Mini instead of the Arduino Nano.

Any thoughts on what I'm doing wrong and might need to change? Does this still relate to your previous [comments here](https://github.com/esp8266/Arduino/issues/2416)? If so, it is a shame they haven't corrected the error you reported yet!

Thanks in advanced, and thank you for creating such a useful and versatile library.

Phipli, 2019-03-25:
Back home (where the circuit is) now, I can confirm that it works if you implement the workaround detailed in this previous "Issue" https://github.com/olikraus/u8g2/issues/322 
I haven't found where the correct version of SPI.cpp is installed on this machine to test your proposed modification to the esp8266 software.

olikraus, 2019-03-25:
so, it works?

Phipli, 2019-03-25:
Yeah - just the issue with SPI mode still present in the ESP8266 hardware description : https://www.youtube.com/watch?v=AffTSqE-TsY

closed 2019-04-04 by Phipli