olikraus/u8g2 #844: Software I2C on AVR

Taimassage, 2019-03-27:
Hello
I have a issue with the Software I2C.
My hardware setup.
I am using a Atmega328p and a SH1106 OLED Display. The frequenc of the Atmega328p is currendly 8MHz.
The SDA is connectet to PC5, and teh SCL is connectet to PC4. For both pins i used a pullup resistor (4.7k). The reset is connected to PC6. I am very sure that the display isnt the issue, because i tested it with SPI and it worked really well. With I2C, I cant see anything on my display.

I could catch a signal from the SDA and the SCL (as you can see in the picture). The only issue i can see is the frequence of the SCL (1kHz, C1 in picture).

So could it be, that the frequency of the SCL ist to low? Do i have to set a prescaler? Or can you see something wrong in my code? 

I look forward to an answer and thank you in advance. 

[image]

```
#include <u8g2.h>
#include <util/delay.h>
#include <u8x8_avr.h>
#include <avr/power.h>

#define DISPLAY_CLK_DIR DDRC
#define DISPLAY_CLK_PORT PORTC
#define DISPLAY_CLK_PIN 5

#define DISPLAY_DATA_DIR DDRC
#define DISPLAY_DATA_PORT PORTC
#define DISPLAY_DATA_PIN 4

u8g2_t u8g2;	
uint8_t u8x8_avr_gpio_and_delay(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
 	switch(msg)
	{
		case U8X8_MSG_GPIO_AND_DELAY_INIT:  // called once during init phase of u8g2/u8x8
			DISPLAY_CLK_DIR |= 1<<DISPLAY_CLK_PIN;
			DISPLAY_DATA_DIR |= 1<<DISPLAY_DATA_PIN;
			
		break;              // can be used to setup pins
		
		case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
			
		break;
		case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
			_delay_us(arg_int*1000UL);
		break;
		
		case U8X8_MSG_DELAY_100NANO:
      /* not used at the moment */
		break;
		
		  case U8X8_MSG_DELAY_NANO:
      /* not required for SW I2C */
		break;
		
		case U8X8_MSG_DELAY_I2C:				
			// arg_int is 1 or 4: 100KHz (5us) or 400KHz (1.25us) 
			_delay_us(arg_int<=2?5:1);
		break;
		
		 case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
			if (arg_int == 0)
			{
				DDRC |= _BV(DDC6);
				PORTC &= ~(_BV(PORTC6));
			}else{	
				DDRC &= ~(_BV(PORTC6));
				PORTC |= _BV(PORTC6);
			}
		break;
			
		case U8X8_MSG_GPIO_I2C_CLOCK:        // Clock pin: Output level in arg_int
			if(arg_int)
			{
				DISPLAY_CLK_PORT |= (1<<DISPLAY_CLK_PIN);
			}else{
				DISPLAY_CLK_PORT &= ~(1<<DISPLAY_CLK_PIN);
			}
		break;
		
		case U8X8_MSG_GPIO_I2C_DATA:        // Data pin: Output level in arg_int		
			if(arg_int)
			{
				DISPLAY_DATA_PORT |= (1<<DISPLAY_DATA_PIN);
			}else{
				DISPLAY_DATA_PORT &= ~(1<<DISPLAY_DATA_PIN);
			}
		break;

		default:
			if (u8x8_avr_delay(u8x8, msg, arg_int, arg_ptr))	// check for any delay msgs
		return 1;
			u8x8_SetGPIOResult(u8x8, 1);      // default return value
		break;
	}
	return 1;
}

void Display_Init (void)
{	
	u8g2_Setup_sh1106_128x64_noname_f(&u8g2, U8G2_R0, u8x8_byte_sw_i2c,    u8x8_avr_gpio_and_delay);
	u8g2_SetI2CAddress(&u8g2, 0x3C*2);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);
}

int main (void)
 {
	CLKPR = (1 << CLKPCE); // enable a change to CLKPR
	CLKPR = 0;
	Display_Init();	
	
	u8g2_ClearBuffer(&u8g2);
	u8g2_SetFont(&u8g2, u8g2_font_sandyforest_tu);
    u8g2_DrawStr(&u8g2, 40, 17, "I2C WORKS");
	u8g2_SendBuffer(&u8g2);
  while (1) ;
  {
	
  }
}
```

olikraus, 2019-03-27:
Independent from the I2C speed, you should be able to see something. 
Did you change the display accordingly (solder bridges need to be changed)?

Taimassage, 2019-03-27:
Thank you for your quick answer.
Yes, i changed the solder bridges, as you can see in the Picture. And i can't even see a single pixel on the display.

So do you think it has to be a hardware problem?

[image]

olikraus, 2019-03-27:
Did you do a correct wiring? D/C and DC MUST be connected to 0V.

Taimassage, 2019-03-27:
If you mean DC and the CS... then yes 

So i connected the 

GND   0V
VCC    5V
DO     SCL (Pullup)
D1      SDA (Pullup)
RST     Reset
DC       0V
CS       0V
[image]

olikraus, 2019-03-27:
Yes, DC and CS. 

hmm... everything looks fine to me...

olikraus, 2019-04-07:
no further feedback, closing...

closed 2019-04-07 by olikraus