olikraus/u8g2 #845: Ram overflow how to solve

rromano001, 2019-03-27:
Platform information:
 linux mint 64 17.3 (plan to move on 19.x but no time to do at this moment)
System Workbench for STM32 - OpenOCD (for Linux 64bits)   Version: 1.17.0.201812190825
STMCube32Mx version 5.1.0
 This moment I am at early stage with u8g2 library learning.
 Goal is to provide ready to use project on some STM32 platform selected from M0 M3 M4 I am using with support of graphics.

 Hi, I just try'd to use some example from various "tutorials"..
 Goal is to test an SPI 4 wire ssd1306 OLED and learn about u8g2 that seem promise good.
 So I prepared a fresh project from STM32CUBEMx, target blue pills stm32f103c8t6.
 First attempt closed with failure on uint16_t refusing to compile CMSIS library (???).

 Cleaned up workspace created another clean project from cubemx same settings , then applied incremental compile  adding line by line u8g2 from tutorial:
https://github.com/leechwort/u8g2-stm32-example

it compile fine but fail on linker with a very huge ram overflow:
Building target: oled1.elf
Invoking: MCU GCC Linker
arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -mfloat-abi=soft -specs=nosys.specs -specs=nano.specs -T"../STM32F103C8Tx_FLASH.ld" -Wl,-Map=output.map -Wl,--gc-sections -o "oled1.elf" @"objects.list"   -lm
/opt/Ac6/SystemWorkbench_2_3/plugins/fr.ac6.mcu.externaltools.arm-none.linux64_1.17.0.201812190825/tools/compiler/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: oled1.elf section `.bss' will not fit in region `RAM'
/opt/Ac6/SystemWorkbench_2_3/plugins/fr.ac6.mcu.externaltools.arm-none.linux64_1.17.0.201812190825/tools/compiler/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/bin/ld: region `RAM' overflowed by 106680 bytes
collect2: error: ld returned 1 exit status
make: *** [oled1.elf] Error 1

09:39:07 Build Finished (took 11s.595ms)

 this point I browsed more issue on github and prebuilt but none has addressed ram issue in depth.
 I need help to configure so I can return documentation on how to configure if not available.

olikraus, 2019-03-27:
Did you enable sectioning on function and data object level with `-ffunction-sections -fdata-sections`?
see also: https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/u8g2_rtc/Makefile#L99

rromano001, 2019-03-27:
Olikraus, thank for immediate and proud support,  you posted link to solution before my return.
Done, blue pills/Stlink stick clone has issues, got working but not able to debug, now ported to STM32F051 DISCOVERY (STM32F051R8Tx) SPI 4 wire it work fine.

 SPI 3 wire has trouble on HAL SPI transmit accept only byte... Next step I try LL version I usually use on my code.

 I am now debugging I2C mode still on F051 board.

 next step enable to work on FREERTOS STM32

enable CUBEMX pin driver for reset and DC Gpio, initialize SPI max speed 10MHz, hardware NSS 8 bit.then
 Prepare GPIO call back as found on example then import on IDE.

 First step: paste library folders to Drivers.
Open Properties for project
 select C/C++ Build-> Setttings -> MCU gcc-Compiler -> includes
 Add include link to workspace or file system to load include library. 
   eg: "${workspace_loc:/${ProjName}/Drivers/u8g2}"

 Same pane select Optimization (close to Includes)
 locate "Other optimization flags" add <<-ffunction-sections -fdata-sections>>
apply it compile fine.
 Thank for now, I refine all and when is usable I post somewhere.
 Best regards
 Roberto

olikraus, 2019-03-27:
ok...

rromano001, 2019-03-27:
Hi Olikraus, on i2C it doesn't work, debugging code I seen data len is forever set to 1 and also on Logical analyzer every transaction is composed of one byte. 
i2C function send byte one at a time, SSD1306 require have control and data on same i2C transaction.
 Reverted to low level function to handle multiple byte transmit on single transaction.

 I got trouble with init function, all parameter are forever passed to external io handler, is possible change internal code to pass length of transfer to init too?
from 
`static void u8x8_i2c_data_transfer(u8x8_t *u8x8, uint8_t arg_int, void *arg_ptr) U8X8_NOINLINE;
static void u8x8_i2c_data_transfer(u8x8_t *u8x8, uint8_t arg_int, void *arg_ptr)
{
    u8x8_byte_StartTransfer(u8x8);    
    u8x8_byte_SendByte(u8x8, 0x040);
    u8x8->byte_cb(u8x8, U8X8_MSG_CAD_SEND_DATA, arg_int, arg_ptr);
    u8x8_byte_EndTransfer(u8x8);
}
`
 to

`static void u8x8_i2c_data_transfer(u8x8_t *u8x8, uint8_t arg_int, void *arg_ptr) U8X8_NOINLINE;
static void u8x8_i2c_data_transfer(u8x8_t *u8x8, uint8_t arg_int, void *arg_ptr)
{
    u8x8_byte_StartTransfer_n(u8x8, arg_int + 1);    
    u8x8_byte_SendByte(u8x8, 0x040);
    u8x8->byte_cb(u8x8, U8X8_MSG_CAD_SEND_DATA, arg_int, arg_ptr);
    u8x8_byte_EndTransfer(u8x8);
}
`

then  start transfer too:
`uint8_t u8x8_cad_StartTransfer(u8x8_t *u8x8)
{
  return u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, 0, NULL);
}
`
duplicate to
`uint8_t u8x8_cad_StartTransfer_n(u8x8_t *u8x8, uint8_t arg_int)
{
  return u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, arg_int, NULL);
}
`

 I think this has no impact on actual code but enable the usage of HAL and LL driver without collecting data on send code...
 
doesn't work, arg_int on init bytes is set to garbage...

olikraus, 2019-03-27:
The transfer size is not known in advance.

Did you read the section on I2C here: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-i2c-communication ?

rromano001, 2019-03-27:
actual I2C hardware.
 It work on data..
 Modify to byte and cad StartTransfer returned garbage on data init....

`uint8_t u8x8_byte_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int,
    void *arg_ptr)
{

	  switch(msg)
	  {
	    case U8X8_MSG_BYTE_SEND:
	      Timeout = 5;
	      while( arg_int > 0 && Timeout)
	      {
			if(LL_I2C_IsActiveFlag_TXIS(I2C1))
			{
	    	  LL_I2C_TransmitData8(I2C1, *((uint8_t *)arg_ptr)); //data);
	    	  //(&hi2c1, (uint16_t)u8x8_GetI2CAddress(u8x8), arg_ptr, arg_int, 10000);
	    	  Timeout = 5;
	    	  arg_ptr++;
	    	  arg_int--;
			}
			if (LL_SYSTICK_IsActiveCounterFlag())
				Timeout--;
	      }
	      break;

	    case U8X8_MSG_BYTE_INIT:
	    	LL_I2C_Enable(I2C1);
	      break;
	    case U8X8_MSG_BYTE_SET_DC:
	      break;
	    case U8X8_MSG_BYTE_START_TRANSFER:
	    	if (!arg_int)
	    		arg_int = 2;

	    	LL_I2C_HandleTransfer(I2C1, u8x8_GetI2CAddress(u8x8), LL_I2C_ADDRSLAVE_7BIT,
	    			arg_int, LL_I2C_MODE_AUTOEND, LL_I2C_GENERATE_START_WRITE);	//  LL_I2C_MODE_SOFTEND, LL_I2C_GENERATE_NOSTARTSTOP); //

	    	//LL_I2C_GenerateStartCondition(I2C1);
	      break;
	    case U8X8_MSG_BYTE_END_TRANSFER:
	    	LL_I2C_ClearFlag_STOP(I2C1);
	    	//LL_I2C_GenerateStopCondition(I2C1);
	      break;
	    default:
	      return 0;
	  }
	  return 1;
}
`

 I need your input on how to proceed, change library or collect data to send after? Collecting unknown amount of data on RTOS can be troubling stack allocation or creating garbage on heap-
 How can I get number of byte in advance to data write?
 Regards
 Roberto

olikraus, 2019-03-27:
> I need your input on how to proceed

This is why I mentioned the above link to the u8g2 i2c template.

> change library or collect data to send after? 

second approach (collect data) as mentioned in the above link.

But (for your convenience),  I will copy the code here:
```
uint8_t u8x8_byte_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  static uint8_t buffer[32];		/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
  static uint8_t buf_idx;
  uint8_t *data;
 
  switch(msg)
  {
    case U8X8_MSG_BYTE_SEND:
      data = (uint8_t *)arg_ptr;      
      while( arg_int > 0 )
      {
	buffer[buf_idx++] = *data;
	data++;
	arg_int--;
      }      
      break;
    case U8X8_MSG_BYTE_INIT:
      /* add your custom code to init i2c subsystem */
      break;
    case U8X8_MSG_BYTE_SET_DC:
      /* ignored for i2c */
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      buf_idx = 0;
      break;
    case U8X8_MSG_BYTE_END_TRANSFER:
      i2c_transfer(u8x8_GetI2CAddress(u8x8) >> 1, buf_idx, buffer);
      break;
    default:
      return 0;
  }
  return 1;
}
```

So, you should place ally your I2C HAL code only at the place where "i2c_transfer" is called.

rromano001, 2019-03-27:
Hi Olikraus, that was my first attempt to get init code working...
 I read few part of library, I find external function got called from pointer stored on u8x8 and mainly I focused these functtions:
`uint8_t u8x8_byte_StartTransfer(u8x8_t *u8x8)
{
  return u8x8->byte_cb(u8x8, U8X8_MSG_BYTE_START_TRANSFER, 0, NULL);
}

uint8_t u8x8_byte_StartTransfer_n(u8x8_t *u8x8, uint8_t arg_int)
{
  return u8x8->byte_cb(u8x8, U8X8_MSG_BYTE_START_TRANSFER, arg_int, NULL);
}
`

and on cad:
`uint8_t u8x8_cad_StartTransfer(u8x8_t *u8x8)
{
  return u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, 0, NULL);
}

uint8_t u8x8_cad_StartTransfer_n(u8x8_t *u8x8, uint8_t arg_int)
{
  return u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, arg_int, NULL);
}`

then I noticed these function get called from here and trasnfer length is know on top of procedure... But I think get called not only by this...
`static void u8x8_i2c_data_transfer(u8x8_t *u8x8, uint8_t arg_int, void *arg_ptr) U8X8_NOINLINE;
static void u8x8_i2c_data_transfer(u8x8_t *u8x8, uint8_t arg_int, void *arg_ptr)
{
    u8x8_byte_StartTransfer_n(u8x8,arg_int+1);
    u8x8_byte_SendByte(u8x8, 0x040);
    u8x8->byte_cb(u8x8, U8X8_MSG_CAD_SEND_DATA, arg_int, arg_ptr);
    u8x8_byte_EndTransfer(u8x8);
}
`
 I also think is time to open another thread, original is solved and this has nothing to do with original overflow.
 Thank in advance for all kind assistance you are providing.
 Roberto

olikraus, 2019-03-27:
:-)

closed 2019-03-27 by olikraus