olikraus/u8g2 #846: SAML21 OLED Project sendBuffer only sending x0 position

jonathanleichner, 2019-03-27:
Hello

I have created a new Atmel Studio 7 project, the display initializes fine, i am using hardware SPI, i have a logic analyzer on the data lines and get correct data.

Device is a ATSAML21J18B.
I'm using the full buffer option. SSD1306 noname _f.

Problem i have is something with the sendBuffer function, when it sends it only sends the X0 positions and never advances to the next X positions.

I used a drawBox for this test. It only draws the box in X0, the Y is correct and changes sizes accordingly, but i can never get the box in X bigger than 1 pixel wide.  I land up with a vertical line in X0 and nothing more. All clipping settings i have checked also.

None of the other functions work either, this is the only one i manged to see something on the display.

Have you got any idea what might be casing this issue?

Thank you.
Jonathan

olikraus, 2019-03-27:
>  sendBuffer function, when it sends it only sends the X0 positions and never advances to the next X positions.

Not sure what you mean. Maybe post some code and pictures.

jonathanleichner, 2019-03-27:
Had a feeling about that.

I have attached a picture and also code.

`int main(void)
{

	atmel_start_init();

	u8g2_Setup_ssd1306_128x64_noname_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_saml21);
	u8g2_InitDisplay(&u8g2);
	u8g2_SetPowerSave(&u8g2, 0);
	

	u8g2_ClearBuffer(&u8g2);
	u8g2_SetDrawColor(&u8g2,1);
	u8g2_DrawBox(&u8g2,0, 0, 40, 64);
	u8g2_SendBuffer(&u8g2);

	while(1)
	{
	}

}`

`//SAML21 specific implementation, initializes SPI peripheral and provides low level functions which allows the u8g2 library to communicate with the display
uint8_t u8g2_gpio_and_delay_saml21(U8X8_UNUSED u8x8_t *u8x8, U8X8_UNUSED uint8_t msg, U8X8_UNUSED uint8_t arg_int, U8X8_UNUSED void *arg_ptr){
	
	switch(msg){
		//Initialize SPI peripheral
		case U8X8_MSG_GPIO_AND_DELAY_INIT:
		spi_m_sync_get_io_descriptor(&OLED_SPI, &io);
		spi_m_sync_enable(&OLED_SPI);
		//initializeIO();
		break;
		
		//Function which implements a delay, arg_int contains the amount of ms
		case U8X8_MSG_DELAY_MILLI:
		delay_ms(arg_int);
		
		break;
		//Function which delays 10us
		case U8X8_MSG_DELAY_10MICRO:
		delay_us(10);
		
		break;
		//Function which delays 100ns
		case U8X8_MSG_DELAY_100NANO:
		delay_us(0.1);
		
		break;
		//Function to define the logic level of the clockline
		case U8X8_MSG_GPIO_SPI_CLOCK:
		//_gpio_set_level(DISPLAY_SCLK, arg_int);
		
		break;
		//Function to define the logic level of the data line to the display
		case U8X8_MSG_GPIO_SPI_DATA:
		//_gpio_set_level(, arg_int);
		
		break;
		// Function to define the logic level of the CS line
		case U8X8_MSG_GPIO_CS:
		gpio_set_pin_level(OLED_CS, arg_int);
		
		break;
		//Function to define the logic level of the Data/ Command line
		case U8X8_MSG_GPIO_DC:
		gpio_set_pin_level(OLED_DC, arg_int);
		
		break;
		//Function to define the logic level of the RESET line
		case U8X8_MSG_GPIO_RESET:
		gpio_set_pin_level(OLED_RST, arg_int);
		
		break;
		default:
		return 0; //A message was received which is not implemented, return 0 to indicate an error
	}
	
	return 1; // command processed successfully.
}
[image]
`

Regards
Jonathan

olikraus, 2019-03-27:
The delay messages receive the delay in the `arg_int` paramenter: https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/u8g2_rtc/u8x8cb.c#L56
At least this is missing.

Additionally, the cases for `U8X8_MSG_GPIO_SPI_CLOCK` and `U8X8_MSG_GPIO_SPI_DATA` are empty.

jonathanleichner, 2019-03-27:
Yes, this is because i am using Hardware SPI.
 `case U8X8_MSG_DELAY_MILLI:
	delay_ms(arg_int);`

Looks correct to me?

olikraus, 2019-03-27:
> Yes, this is because i am using Hardware SPI.

hmm, at least not according to your code:
`u8g2_Setup_ssd1306_128x64_noname_f(&u8g2, U8G2_R0, u8x8_byte_4wire_sw_spi, u8g2_gpio_and_delay_saml21);`

`u8x8_byte_4wire_sw_spi` looks like the original sw emulated code to my. If you have modified this procedure, then posting the same would be cool for further debugging. In fact i assume the bug is inside there.

jonathanleichner, 2019-03-27:
Ah, sorry. I just modified your u8x8_byte_4wire_sw_spi code. Maybe the delays are the issue? Cause i don't call any of them in here.

`uint8_t u8x8_byte_4wire_sw_spi(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
	
	switch(msg)
	{
		case U8X8_MSG_BYTE_SEND:
		data = (uint8_t *)arg_ptr;

		io_write(io,(uint8_t *)data, 1);			//Hardware SPI Write.

		break;
		
		case U8X8_MSG_BYTE_INIT:
		GPIO_set(OLED_CS);
		break;
		case U8X8_MSG_BYTE_SET_DC:
		gpio_set_pin_level(OLED_DC, arg_int);
		break;
		case U8X8_MSG_BYTE_START_TRANSFER:
		//u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_enable_level);
		//u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->post_chip_enable_wait_ns, NULL);
		GPIO_clr(OLED_CS);
		break;
		case U8X8_MSG_BYTE_END_TRANSFER:
		//u8x8->gpio_and_delay_cb(u8x8, U8X8_MSG_DELAY_NANO, u8x8->display_info->pre_chip_disable_wait_ns, NULL);
		//u8x8_gpio_SetCS(u8x8, u8x8->display_info->chip_disable_level);
		GPIO_set(OLED_CS);
		break;
		default:
		return 0;
	}
	return 1;
}`

Let me check if i uncomment delays.

olikraus, 2019-03-27:
> `io_write(io,(uint8_t *)data, 1);`

Again the number of bytes is stored in `arg_int`. For reference look at the template code: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform#hardware-spi-communication

jonathanleichner, 2019-03-27:
That was it, i thought it only passed one byte at a time.

`		case U8X8_MSG_BYTE_SEND:
		data = (uint8_t *)arg_ptr;
		while( arg_int > 0 ) 
		{
			io_write(io,(uint8_t *)data, 1);			//Hardware SPI Write.
			data++;
			arg_int--;
		}
		
		break;`

Thank you so much for your quick response and help. 
I really appreciate it.

olikraus, 2019-03-27:
:)

olikraus, 2019-03-27:
One more comment: I think you could also use `io_write(io,(uint8_t *)data, arg_int);` instead of the while loop.

jonathanleichner, 2019-03-27:
Yea, that would eliminate the loop. Perfect.

Thanks again, this library is amazing.

closed 2019-04-07 by olikraus