olikraus/u8g2 #847: i2c transfer lenght not known on init

rromano001, 2019-03-27:
Hi Olikraus, I open a new issue solved, this is what I changed to get code working without duplicate buffer. Embedded has limited ram and sometimes require not to waste resources so this is what I propose to:

 Target SSD1306 Oled I2C on STM32F051, code generated from CubeMx LL mode, HAL is difficult to adapt, I wish to support FREERTOS so LL is better to do task Yield on busy.

u8x8byte.c:
`uint8_t u8x8_byte_StartTransfer(u8x8_t *u8x8)
{
  return u8x8->byte_cb(u8x8, U8X8_MSG_BYTE_START_TRANSFER, 0, NULL);
}
// added this new function
uint8_t u8x8_byte_StartTransfer_n(u8x8_t *u8x8, uint8_t arg_int)
{
  return u8x8->byte_cb(u8x8, U8X8_MSG_BYTE_START_TRANSFER, arg_int, NULL);
}
`
u8x8cad.c:
`uint8_t u8x8_cad_StartTransfer(u8x8_t *u8x8)
{
  return u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, 0, NULL);
}
// added this code
uint8_t u8x8_cad_StartTransfer_n(u8x8_t *u8x8, uint8_t arg_int)
{
  return u8x8->cad_cb(u8x8, U8X8_MSG_CAD_START_TRANSFER, arg_int, NULL);
}
`

burst transfer modified to export data length:
`static void u8x8_i2c_data_transfer(u8x8_t *u8x8, uint8_t arg_int, void *arg_ptr) U8X8_NOINLINE;
static void u8x8_i2c_data_transfer(u8x8_t *u8x8, uint8_t arg_int, void *arg_ptr)
{
    u8x8_byte_StartTransfer_n(u8x8,arg_int+1); // <<<------- added data length
    u8x8_byte_SendByte(u8x8, 0x040);
    u8x8->byte_cb(u8x8, U8X8_MSG_CAD_SEND_DATA, arg_int, arg_ptr);
    u8x8_byte_EndTransfer(u8x8);
}
`

setup to init code adding transfer length (arg_int is not length here)
`uint8_t u8x8_cad_ssd13xx_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  uint8_t *p;
  switch(msg)
  {
    case U8X8_MSG_CAD_SEND_CMD:
    case U8X8_MSG_CAD_SEND_ARG:
      /* 7 Nov 2016: Can this be improved?  */
      //u8x8_byte_SetDC(u8x8, 0);
      u8x8_byte_StartTransfer_n(u8x8,2);<----- length
`
End of changed library files.

rromano001, 2019-03-27:
At this point what to do is at your choice, this way I2C version is more faster than collecting data on a new array and also data loop can pass away processor resource  with taskYield.
 regards
 Roberto

Here the main Hardware length driven routine:
`uint8_t u8x8_byte_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int,
    void *arg_ptr)
{

	 //uint8_t data;

	  switch(msg)
	  {
	    case U8X8_MSG_BYTE_SEND:
	      Timeout = 5;
	      while( arg_int > 0 && Timeout)
	      {
			if(LL_I2C_IsActiveFlag_TXIS(I2C1))
			{
	    	  //data = (uint8_t)(*arg_ptr);
	    	  LL_I2C_TransmitData8(I2C1, *((uint8_t *)arg_ptr)); //data);
	    	  //(&hi2c1, (uint16_t)u8x8_GetI2CAddress(u8x8), arg_ptr, arg_int, 10000);
	    	  //i2c_write_byte(u8x8, *data);
	    	  //data++;
	    	  Timeout = 5;
	    	  arg_ptr++;
	    	  arg_int--;
			}
			if (LL_SYSTICK_IsActiveCounterFlag())
				Timeout--;
	      }
	      break;

	    case U8X8_MSG_BYTE_INIT:
	    	LL_I2C_Enable(I2C1);
	      break;
	    case U8X8_MSG_BYTE_SET_DC:
	      break;
	    case U8X8_MSG_BYTE_START_TRANSFER:
	    	if (!arg_int)
	    		arg_int = 2;

	    	LL_I2C_HandleTransfer(I2C1, u8x8_GetI2CAddress(u8x8), LL_I2C_ADDRSLAVE_7BIT,
	    			arg_int, LL_I2C_MODE_AUTOEND, LL_I2C_GENERATE_START_WRITE);	//  LL_I2C_MODE_SOFTEND, LL_I2C_GENERATE_NOSTARTSTOP); //

	    	//LL_I2C_GenerateStartCondition(I2C1);
	      //i2c_start(u8x8);
	      //i2c_write_byte(u8x8, u8x8_GetI2CAddress(u8x8));
	      break;
	    case U8X8_MSG_BYTE_END_TRANSFER:
	    	LL_I2C_ClearFlag_STOP(I2C1);
	    	//LL_I2C_GenerateStopCondition(I2C1);
	      //i2c_stop(u8x8);
	      break;
	    default:
	      return 0;
	  }
	  return 1;
`

olikraus, 2019-03-27:
Well, your solution may work for SSD1306 at the moment, but other interfaces might be different and need to be tested also.
One more point: The current code is optimized for Arduino HAL (which actually comes with that 32 byte array). So I do not see a need to change the code. In fact testing effort will be huge and a different solution needs to be implemented for Arduino HAL.

rromano001, 2019-03-27:
Hi Olikraus, I am not interested at all to Arduino, I am professional developing control code for embedded.
 As you can see my proposal has no impact at all to other drivers.
 Again I need faster response and small ram footprint.
 Next step is to port on FPGA too, so expect I propose a lot of code hint or request.
 This code I noticed call reset routine prepared for 4 wire SPI, this can be an issue when other hardware get attached to tampered port.
 I like your code also if it is complex but at last simple to use and quite clean.
 Thank for help and this good code.
 Roberto

olikraus, 2019-03-27:
> I am professional developing control code for embedded.

Wow, thanks for using code from a Arduino hobby developer.

> Again I need faster response and small ram footprint.

Sure, this is open and free code. So any optimization can be applied. Nevertheless, I can not take over all suggestions (probably due to my limited skills but maybe also doubts regarding portability towards Arduino IDE).

> Next step is to port on FPGA too, so expect I propose a lot of code hint or request.

hmmm... sounds like an intelligent display controller.

> I like your code also if it is complex but at last simple to use and quite clean.

Thanks!

olikraus, 2019-03-27:
BTW... probably too late, but I once created an architecture overview: https://github.com/olikraus/u8g2/blob/master/doc/u8g2_software_architecture.png

rromano001, 2019-03-27:
Nothing is late when we are learning about.
 Yes I dissected your code to get it working at best and understand how it was internally, but this last resource may be of great help to deeply understand.
 I wish prepare some ready to use project for STM32 platform.
 Against FPGA it is an emulation of old SED1330 I seen is supported by your library, this is not my goal but help test more in deep.
 My wish was to have an handy small footprint library I2C based to test code when is running, logging is great but sometimes some on time feedback is better than read tons of logged data after issue.
 I think I can port ssd1963 too (not sure of number) and try adapt to STM32F7 display when I collect spare time.
 Saleae Logic 8 plus, debugger and profiling got great control of your code ;)
 First day of test got some good result.
 Best wishes.
 Roberto

closed 2019-04-07 by olikraus