olikraus/u8g2 #850: U8G2_SSD1306_128X64_ALT0_1_HW_I2C

rotacode, 2019-03-29:
hello olikraus
im using you great lib in my project and i have some problem to get my screens work goood and fast.
all my screens connected thru TCA i2c multyplexer and im able to get them work with u8g2.
im conectec with other slave device and i expirience some problem to send data to 2 address when im using U8G2_SSD1306_128X64_NONAME_1_HW_I2C.
in other cases the screen start and all slaves work when i use:
U8G2_SSD1306_128X64_NONAME_F_HW_I2C but every time the screens update data its stops the other slave from working.
then i tryed U8G2_SSD1306_128X64_ALT0_1_HW_I2C. and i figure out that only few lines on top on the screens working but the data run now fast and not freez my other slave.
alsso the creen response very fast.
but now i m not seeing al the screen.
there is any way the config that ALT0 structure to fix this issue ?
or could you give me any hint what is the way to edit that chang in the lib?
thank you
rota

olikraus, 2019-03-29:
>  i figure out that only few lines on top on the screens working

There is a fundamental difference between 1 and F constructors. Did you consider this?

rotacode, 2019-03-29:
yes but i sit to try all the constructure one by one and i mean one by one.
all of them over flow my other slave. and make him fail.
the NONAME is working full screen but is alsso freez my other device.
so only ALT works fevy fast and not over flow the system but only few line in the screen running.
im realy lost here

olikraus, 2019-03-29:
> the NONAME is working full screen but is alsso freez my other device.
> so only ALT works fevy fast and not over flow the system but only few line in the screen running.

Did you try the "U8G2_SSD1306_128X64_ALT0_F_HW_I2C" (note the "F") constructor? I mean you should not compare apples with bananas, U8G2_SSD1306_128X64_ALT0_F_HW_I2C differs a lot from U8G2_SSD1306_128X64_ALT0_1_HW_I2C.

rotacode, 2019-03-29:
i tryed them all 
i i put F in my case its working nice on screen but freezing other slaves that not somthing i can live with
its really big project and the most.
i spent like tow days tring all combinations of structure my problem is that icant use adafruit dir or others dirs coz im using i2c_t3 dir for Wire()

rotacode, 2019-03-29:
by the way allso NONAME 1 work nice and fast but the same just few line of the top of the screen r working

olikraus, 2019-03-30:
> by the way allso NONAME 1 work nice and fast but the same just few line of the top of the screen r working.
any suggestion?

Difficult without seeing the code, but I assume you have used the sendBuffer command with the NONAME 1 constructor (which is wrong, you must use the firstpage/nextpage loop)

rotacode, 2019-03-31:
thanks for your replay olikraus 
you right i use sendBuffer at the end of my void screens commends.
ill try to use firstpage/nextpage like you say and ill update soon.

by the way i compromised by using for now ALT_2_HW and its showing  half (top) of my screens. the system working fine with it so for now i just put all values on top of the screen. hoop your suggestion will solve it.

olikraus, 2019-03-31:
You can also use "ALT_F_HW" if there is enough RAM available, then all the content will be available.

rotacode, 2019-04-01:
hi olikraus  
so i did some tests:
i tryd use all kind of and ALT0 with  firstpage/nextpage loop
in all cases i see full screen and  i got delay thats freez my other i2c slave that control motor faders.
only if i use ALT0_2 with sendBuffer(); i can get it work smoothly. but its shows only half screen

olikraus, 2019-04-01:
It works as it should. If you want simultanous controll of motor faders and display output you probably need a multitasking operating system with proper resource mutual exclusion for th I2C bus. This is far beyound the capabilities of Arduino and is also not a subject or problem of U8g2.

olikraus, 2019-04-01:
ok, I have another idea, which may work for you: You can interlace your code with u8g2 page drawing.
An example is here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/StateBufferLoop/StateBufferLoop.ino

Place your motor fader commands here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/StateBufferLoop/StateBufferLoop.ino#L283

rotacode, 2019-04-02:
nice:) many thanks
ill try to implant it and ill update asap.
thats looks likes the right implantation.

rotacode, 2019-04-10:
hi olikraus
so i try many ways to implant it to my code but not got successful with full screen and fast buffer without system delays. i try to use your example but seens i use TCA multiplexer for 8 i2c connection its the firstpage/nextpage not responding good as send.buffer();
here is my full code for screens and motor controling
`#include <U8g2lib.h>
#include <U8x8lib.h>
#include <i2c_t3.h>
#include <MIDIUSB.h>
#include <Bounce.h>

#define TARGET    0x10              // All faders to target value as fast as possible. PID state is cleared with each command.
#define FOLLOW    0x20 
#define MESSAGE_LEN 17 
U8G2_SSD1306_128X64_ALT0_2_HW_I2C u8g2(U8G2_R0,/* reset=*/ U8X8_PIN_NONE);         // same as the NONAME variant, but may solve the "every 2nd line skipped" problem

byte INT2HEX[8] = {0x00,0x01,0x02,0x03,0x04,0x05,0x06,0x07};
int SCRNUM[8] = {0x07,0x06,0x05,0x04,0x03,0x02,0x01,0x00};
int MUTNUM[8] = {0x07,0x06,0x05,0x04,0x03,0x02,0x01,0x00};
int dbRA[8] = {0,0,0,0,0,0,0,0};

String chname1[8];
String chname2[8];
String chname3[8];
String chname4[8];
int centerword[8];
int hi[8];
int lo[8];
int hibyte[8];
int lobyte[8];
int Gain[8];
String chmute[8] ;
String chsolo[8] ;

boolean solopress = false;
boolean solorelease = false;
elapsedMicros sinceupdatescreenvalues;                         // timer for fader check
unsigned int  updatescreenvaluesInterval = 300000;
unsigned long interval=500; // the time we need to wait
unsigned long previousMillis=0; // millis() returns an unsigned long.  
elapsedMicros sinceButtonmsgRead;                         // timer for fader check
unsigned int  ButtonmsgReadInterval = 20000;
elapsedMicros sinceFaderRead;                         // timer for fader check
unsigned int  faderReadInterval = 1000;
elapsedMicros sinceFaderWrite;                         // timer for fader check
unsigned int   faderWriteInterval = 20000;
bool FaderState = false; // state variable for the LED
int faderTouchCount;
uint8_t panelAddress = 0x66;
byte faderValueData[MESSAGE_LEN];    // 16 bytes for values, 1 byte for touch
byte faderTargetData[MESSAGE_LEN];   // 16 for target values, 1 byte for indicating which faders should be moved.
boolean faderTouched[8];                              // boolean indicating if fader is touched
int     faderValue[8];                                // fader value ranging 0 to 1023
int     faderTarget[8];                               // fader target value ranging 0 to 1023
boolean faderMove[8];                               // fader target value ranging 0 to 1023
boolean updateFaders[8];

//MUX BUTTOMS
// Store the elapsed milliseconds for analog and multiplexed input smoothing/de-bouncing
elapsedMillis msec0 = 0;
long msecMux0[5] = {0, 0, 0, 0, 0};
elapsedMillis msec1 = 0;
long msecMux1[5] = {0, 0, 0, 0, 0};

// Setup mux0 chip pins
const int selectPins0[4] = {43, 44, 45, 46}; // S0~0, S1~1, S2~12
//const int zOutput 
const int zInput0 = 13; // Connect common (Z) to 19 (digital input)
// Setup mux1 chip pins
const int selectPins1[4] = {24, 25, 26, 27}; // S0~0, S1~1, S2~12
//const int zOutput 
const int zInput1 = 28; // Connect common (Z) to 19 (digital input)
[... 978 lines cut ...]
DB =9.1;
}
if (Gain == 121){
DB = 9.5;
}
if (Gain == 122){
DB = 10.0;
}
if (Gain == 123){
DB = 10.2;
}
if (Gain == 124){
DB = 10.6;
}
if (Gain == 125){
DB = 11.0;
}
if (Gain == 126){
DB = 12.0;
}
if (Gain == 127){
DB = 12.00;
}
  return DB;}
  `

olikraus, 2019-04-10:
Hm... I do not get paid for this here... I will not review >1000 lines of code.

It is also not formated and also does not contain firstPage()/nextPage().
If you want me to discuss on a code, make a small piece of code which shows the neccessary part.

If you have a I2C multiplexer you must carefully update the multiplexer before calling firstPage() and nextPage().

rotacode, 2019-04-10:
what do u mean when you say 
 "carefully update the multiplexer before calling firstPage() and nextPage()."

thats the main u8g2 code
`void updatescreenvalues(){
     
     for (int i=0;i<8;i++){
    
     Wire.begin();   
     tcaselect(i);   

     //larg letters names
     chname1[i].toUpperCase();
     chname2[i].toUpperCase();
     chname3[i].toUpperCase();
     chname4[i].toUpperCase();
     
     //page structure
     u8g2.clearBuffer();
     u8g2.drawLine(0, 32, 128, 32);
     u8g2.setFont(u8g2_font_ncenB10_tr); // choose a suitable font
     u8g2.setCursor(90, 60);
     u8g2.print("dbA");
     
     //values update
     u8g2.setFont(u8g2_font_ncenB10_tr);     
     u8g2.setCursor(centerword[i]-5, 15);       
     u8g2.print(String(chname1[i]+chname2[i]+chname3[i]+chname4[i]));            
     u8g2.setFont(u8g2_font_ncenB10_tr); // choose a suitable font
     //u8g2.setCursor(42, 60);
     u8g2.setCursor(02, 15);
     u8g2.print(dbRA[i]);

      //mute update screens      
     if(chmute[i] == "M"){
     u8g2.setFontMode(1);  /* activate transparent font mode */
     u8g2.drawFrame(104,3, 23,13);
     u8g2.setDrawColor(2);
     u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
     u8g2.setCursor(110, 13);    
     u8g2.print(chmute[i]);  
     }
     
     //solo update screens
     if(chsolo[i] == "S"){
     u8g2.setFontMode(1);  /* activate transparent font mode */
//   u8g2.drawBox(01,35, 24, 50);
     u8g2.setDrawColor(2);
     u8g2.drawBox(104,3, 23,13);
     u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
     u8g2.setCursor(112, 13);
     u8g2.print(chsolo[i]);
  
     }
     u8g2.sendBuffer(); 
     }
     
}

//Startup screen
void startscreen(){
  for (int i=0;i<8;i++){
  Wire.begin();
  tcaselect(i);
  u8g2.begin();
  u8g2.clearBuffer();          // clear the internal memory
  //u8g2.drawRFrame(2 ,2,122,62,7);
  u8g2.setFont(u8g2_font_ncenB12_tr); // choose a suitable font
  u8g2.setCursor(34,16);
  u8g2.print("AMC8");  // write something to the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
  u8g2.setCursor(26, 55);
  u8g2.print("HUI Protocol"); // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
 }
delay(7000);
}`

startscreen is an opening screen for 7 seconds and its run from setup.
updatescreens run in the loop in time interval.
i tryd to replace send/clear buffer with firstPage() and nextPage() for startscreen and updatescreens.
but i dont get stable action on screens.

olikraus, 2019-04-10:
I have never tested such a setup and i think your code is still too complicated. I suggest, that you just print the OLED number on each screen in a huge font.

Then simplify the code even further. We need a real simple example.

Once this is done we should start to fix things. For example: Wire.begin() is not required from my point of view. ALso u8g2.begin() should be only done once during setup.

olikraus, 2019-04-22:
No further reply, closing

closed 2019-04-22 by olikraus