olikraus/u8g2 #851: how to startup u8g 2 on nRF52832 i2c with SSD1315

amigomcu, 2019-03-31:
hi all,
ref this issue #553 , the SSD1315 compatible SSD1306
as follow me snippet code , it can't work correctly, which no thing to show

Can help me  revise it . 

//nRF52 i2c init and handler
```
void twi_handler(nrf_drv_twi_evt_t const * p_event, void * p_context)
{
    switch (p_event->type)
    {
        case NRF_DRV_TWI_EVT_DONE:
            if (p_event->xfer_desc.type == NRF_DRV_TWI_XFER_RX)
            {
                //data_handler(m_sample);
            }
                NRF_LOG_INFO("NRF_DRV_TWI_EVT_DONE.");
            break;
        case NRF_DRV_TWI_EVT_ADDRESS_NACK:
            NRF_LOG_INFO("NRF_DRV_TWI_EVT_ADDRESS_NACK.");
        break;
        case NRF_DRV_TWI_EVT_DATA_NACK:
            NRF_LOG_INFO("NRF_DRV_TWI_EVT_DATA_NACK.");
        break;
        default:
            break;
    }
}

static void twi_config(void)
{
    uint32_t err_code;
    nrf_drv_twi_config_t const config = {
       .scl                = ARDUINO_SCL_PIN,
       .sda                = ARDUINO_SDA_PIN,
       .frequency          = NRF_DRV_TWI_FREQ_400K,
       .interrupt_priority = APP_IRQ_PRIORITY_LOWEST,
       .clear_bus_init     = false
    };
    err_code = nrf_drv_twi_init(&m_twi, &config, twi_handler, NULL);
    APP_ERROR_CHECK(err_code);
    nrf_drv_twi_enable(&m_twi);
}
```
//u8g2 delay ,not handle
```
uint8_t u8x8_gpio_and_delay_template(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  switch(msg)
  {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
      break;							// can be used to setup pins
    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
      break;    
    case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
      break;
    case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds
      break;
    case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
      break;
    case U8X8_MSG_DELAY_I2C:				// arg_int is the I2C speed in 100KHz, e.g. 4 = 400 KHz
      break;							// arg_int=1: delay by 5us, arg_int = 4: delay by 1.25us
    case U8X8_MSG_GPIO_D0:				// D0 or SPI clock pin: Output level in arg_int
    //case U8X8_MSG_GPIO_SPI_CLOCK:
      break;
    case U8X8_MSG_GPIO_D1:				// D1 or SPI data pin: Output level in arg_int
    //case U8X8_MSG_GPIO_SPI_DATA:
      break;
    case U8X8_MSG_GPIO_D2:				// D2 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_D3:				// D3 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_D4:				// D4 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_D5:				// D5 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_D6:				// D6 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_D7:				// D7 pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_E:				// E/WR pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_CS:				// CS (chip select) pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_DC:				// DC (data/cmd, A0, register select) pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_RESET:			// Reset pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_CS1:				// CS1 (chip select) pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_CS2:				// CS2 (chip select) pin: Output level in arg_int
      break;
    case U8X8_MSG_GPIO_I2C_CLOCK:		// arg_int=0: Output low at I2C clock pin
      break;							// arg_int=1: Input dir with pullup high for I2C clock pin
    case U8X8_MSG_GPIO_I2C_DATA:			// arg_int=0: Output low at I2C data pin
      break;							// arg_int=1: Input dir with pullup high for I2C data pin
    case U8X8_MSG_GPIO_MENU_SELECT:
      u8x8_SetGPIOResult(u8x8, /* get menu select pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_NEXT:
      u8x8_SetGPIOResult(u8x8, /* get menu next pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_PREV:
      u8x8_SetGPIOResult(u8x8, /* get menu prev pin state */ 0);
      break;
    case U8X8_MSG_GPIO_MENU_HOME:
      u8x8_SetGPIOResult(u8x8, /* get menu home pin state */ 0);
      break;
    default:
      u8x8_SetGPIOResult(u8x8, 1);			// default return value
      break;
  }
  return 1;
}
```
//i2c write cmd/data
```
uint8_t u8x8_byte_nrf_hw_i2c(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int,void *arg_ptr)
{
  uint8_t *data;
  static uint8_t buffer[32];		/* u8g2/u8x8 will never send more than 32 bytes between START_TRANSFER and END_TRANSFER */
  static uint8_t buf_idx;
  switch (msg)
  {
  case U8X8_MSG_BYTE_SEND:
    //HAL_SPI_Transmit(&hspi1, (uint8_t *) arg_ptr, arg_int, 10000);
      data = (uint8_t *)arg_ptr;      
      while( arg_int > 0 )
      {
	buffer[buf_idx++] = *data;
	data++;
	arg_int--;
      }      

    break;
  case U8X8_MSG_BYTE_INIT:
    break;
  case U8X8_MSG_BYTE_SET_DC:
    break;
  case U8X8_MSG_BYTE_START_TRANSFER:
      buf_idx = 0;
    break;
  case U8X8_MSG_BYTE_END_TRANSFER:
      nrf_drv_twi_tx(&m_twi, u8x8_GetI2CAddress(u8x8)>>1, buffer, buf_idx, false);
    break;
  default:
    return 0;
  }
  return 1;
}

```
//u8g2 init and start
```
void u8g2_start(void)
{
u8g2_Setup_ssd1306_i2c_128x64_noname_1(&u8g2, U8G2_R0, u8x8_byte_nrf_hw_i2c, u8x8_gpio_and_delay_template);
u8g2_InitDisplay(&u8g2);
u8g2_SetPowerSave(&u8g2, 0);
u8g2_ClearBuffer(&u8g2);
u8g2_SetFont(&u8g2, u8g2_font_ncenB14_tr);
u8g2_DrawStr(&u8g2, 1, 18, "U8g2 on nordic");
u8g2_SendBuffer(&u8g2);
}
```

HW: nRF52832 ,OLED 0.96" I2C display Driver IC SSD1315
SW:SDK 15.2

olikraus, 2019-03-31:
Some thoughts:
1. `u8g2_Setup_ssd1306_i2c_128x64_noname_1` together with `u8g2_SendBuffer` will not work, use `u8g2_Setup_ssd1306_i2c_128x64_noname_f` ("f" at the end) instead
2. I think you still need to fill some code for the delays, especially for `U8X8_MSG_DELAY_MILLI`
3. Did you check `nrf_drv_twi_tx` for the second argument? Maybe the shift is not required.

amigomcu, 2019-04-01:
hi olikraus,
thank you for your replay.
i have been  done as your advice,but it have issue also.

1.  "u8g2_Setup_ssd1306_i2c_128x64_noname_f"
2. add 
`uint8_t u8x8_gpio_and_delay_template(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  switch(msg)
  {
.....
    case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second
      nrf_delay_ms(1);
      break;

......
`
3. nrf_drv_twi_tx address check do ,the address is correct.

see attach files
https://imgur.com/a/6GeecFx

olikraus, 2019-04-01:
It should be "nrf_delay_ms(arg_int)" instead.

I have not investigated your picture much, but it looks ok to me.

amigomcu, 2019-04-04:
hi olikraus ,
thank your reply,
in my test, it seem not entry u8x8_gpio_and_delay_template 
i2c transaction everytime. 
only first transaction(0xae) , other not done it

olikraus, 2019-04-05:
hmm... maybe there is some issue with the low level HW I2C code.

closed 2019-04-09 by amigomcu