olikraus/u8g2 #852: Overwrite a character with another character

ivanbraga, 2019-04-02:
Hi @olikraus.

As it is said "a picture is worth a thousand words".
I'm using this library for the first time, so there are some tricks I do not know yet.

After much searching, I can not solve this problem.

Numbers and characters overlap one another.

I do not want to use mono-space fonts because it occupies a much larger extent of LCD.

I'm using this code that gets in a loop to write the string.

```
sprintf (str, "% 09.2f",  (cont ++) + res );
u8g2_DrawStr (&u8g2, 60, 40, str);
u8g2_SendBuffer (&u8g2);
```

I tried to use empty string before the actual string, but it does not work.

Here's a short video to make it easier to understand. At the beginning it is very clear what it is.

https://1drv.ms/v/s!AvSeL6qqjaFKgVDUcMAyc1M_032N

Someone had a similar problem, but it was solved in a way that did not help me.

 #335

Thank you for the fantastic library.

Ivan
BR
[image]

ivanbraga, 2019-04-02:
I do not know if it's relevant, but I'm using a 240x64 lcd with the RA6963 clone of the T6963.

u8g2_Setup_t6963_240x64_f (& u8g2, U8G2_R0, u8x8_byte_8bit_8080mode, u8x8_gpio_and_delay)

ivanbraga, 2019-04-02:
After digging a little more I found this.

Option 1.
https://github.com/olikraus/u8g2/issues/490#issuecomment-360178830

I'm doing this right now.

```
sprintf (str, "% 09.2f", res + cont ++);

u8g2_SetDrawColor (& u8g2,0);
u8g2_DrawRBox (& u8g2, 60, 22, u8g2_GetStrWidth (& u8g2, str), 20,0);
u8g2_SetDrawColor (& u8g2,1);

u8g2_DrawStr (& u8g2, 60, 40, str);
u8g2_SendBuffer (& u8g2);
```

However, I do not know how to figure out the height of the highest character in the string, to use within "u8g2_DrawRBox", as I used "u8g2_GetStrWidth".

In the file "u8g2_font.c" I found these references "u8g2_font_GetStrSize" and "u8g2_font_GetStrMinBox", but I think they were not implemented, or did I miss something?

olikraus, 2019-04-02:
Ok, let me start with your initial question and code:
```
u8g2_SetFontMode(&u8g2, 0);
u8g2_SetFont(&u8g2, u8g2_font_logisoso16_tn);

loop 
sprintf (str, "% 09.2f",  (cont ++) + res );
u8g2_DrawStr (&u8g2, 60, 40, str);
u8g2_SendBuffer (&u8g2);
```

Background:
1. All draw operation go to an internal buffer
2. The internal buffer is transfered to the display RAM with the SendBuffer command

You might assume, that SendBuffer not only sends the buffer but also clears the buffer after sending hit. However, this is not the case. As a result, any previous content written into the buffer is kept and during the next loop the string overwrites the existing string with deletion of the same.

In order to solve this problem, there are three solutions in your case:
a) clear the buffer before drawing a new content with the ClearBuffer command 
```
u8g2_SetFontMode(&u8g2, 0);
u8g2_SetFont(&u8g2, u8g2_font_logisoso16_tn);

loop 
u8g2_ClearBuffer(&u8g2);
sprintf (str, "% 09.2f",  (cont ++) + res );
u8g2_DrawStr (&u8g2, 60, 40, str);
u8g2_SendBuffer (&u8g2);
```
b) Use the combination of a monospaced front together with none-transparant font writing method. This is the approach described in the mentioned #335 

c) Finally you can erase a desired rectangle area of the screen like mentioned in your last comment.

> However, I do not know how to figure out the height of the highest character in the string, to use within "u8g2_DrawRBox", as I used "u8g2_GetStrWidth".

See here: https://github.com/olikraus/u8g2/wiki/u8g2reference#getascent
Especially have a look at the picture: The height of your string is calulated with ascent-descent.
Note that descent value usually is negative.
Not tested, but let's assume, you draw the string with `u8g2_DrawStr (&u8g2, 60, 40, str);`

Then the deletion area should be this:
`u8g2_DrawBox(&u8g2, 60, 40-ascent, string width,  ascent-descent)`

ivanbraga, 2019-04-03:
@olikraus 

Thanks for the feedback.
Today I was unable to implement its recommendations. And unfortunately I will be away in the next few days.
Since I will not have access to the display, I will not be able to test.
I'll give you a return as soon as possible.

Ivan
BR

ivanbraga, 2019-04-04:
It worked perfectly, but as the string in question and only numbers and point, it was not necessary "ascent-descen".

> u8g2_DrawBox(&u8g2, 60, 40-ascent, string width, ascent-descent)

Because in this case it would affect some pixels used below the string.

Closed question.

Thankful.

closed 2019-04-04 by ivanbraga

olikraus, 2019-04-04:
:-)