olikraus/u8g2 #853: Weird display problem

edkeefe, 2019-04-03:
Hello,

First of all, thank you for such a useful and powerful library.

I developed a small program using this library on an Arduino Uno. Everything worked great.

I'm now porting to an STM32F401. As a starting point, I'm using the information and source code at:

https://elastic-notes.blogspot.com/2018/10/u8g2-library-usage-with-stm32-mcu.html

I've added the required delays. I am using a CS line, so I added code for the U8X8_MSG_BYTE_START_TRANSFER and U8X8_MSG_BYTE_END_TRANSFER sections.

I have it almost working, but sometimes the display glitches. Here are links to a couple of pictures:

Good...
http://www.cosmacvip.com/OLED/good.jpg

Bad...
http://www.cosmacvip.com/OLED/bad1.jpg
http://www.cosmacvip.com/OLED/bad2.jpg

This happens just by resetting the board using the same code.

Here is the code:

`/**
  ******************************************************************************
  * @file           : main.c
  * @brief          : Main program body
  ******************************************************************************
  ** This notice applies to any and all portions of this file
  * that are not between comment pairs USER CODE BEGIN and
  * USER CODE END. Other portions of this file, whether 
  * inserted by the user or by software development tools
  * are owned by their respective copyright owners.
  *
  * COPYRIGHT(c) 2019 STMicroelectronics
  *
  * Redistribution and use in source and binary forms, with or without modification,
  * are permitted provided that the following conditions are met:
  *   1. Redistributions of source code must retain the above copyright notice,
  *      this list of conditions and the following disclaimer.
  *   2. Redistributions in binary form must reproduce the above copyright notice,
  *      this list of conditions and the following disclaimer in the documentation
  *      and/or other materials provided with the distribution.
  *   3. Neither the name of STMicroelectronics nor the names of its contributors
  *      may be used to endorse or promote products derived from this software
  *      without specific prior written permission.
  *
  * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
  * AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
  * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
  * DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
  * FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
  * DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
  * SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
  * CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
  * OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
  * OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
  *
  ******************************************************************************
  */
/* Includes ------------------------------------------------------------------*/
#include "main.h"
#include "stm32f4xx_hal.h"

/* USER CODE BEGIN Includes */
#include "u8g2.h"
/* USER CODE END Includes */

[... 318 lines cut ...]
  * @retval None
  */
void assert_failed(uint8_t* file, uint32_t line)
{ 
  /* USER CODE BEGIN 6 */
  /* User can add his own implementation to report the file name and line number,
     tex: printf("Wrong parameters value: file %s on line %d\r\n", file, line) */
  /* USER CODE END 6 */
}
#endif /* USE_FULL_ASSERT */

/**
  * @}
  */

/**
  * @}
  */

/************************ (C) COPYRIGHT STMicroelectronics *****END OF FILE****/
`

Any suggestions would be really helpful.

ED

olikraus, 2019-04-03:
To me it looks more like an hardware problem. Can you try reduced transfer speed?

edkeefe, 2019-04-03:
Hello,

As you suggested, the SPI speed was too high. 

I modified the 'hspi1.Init.BaudRatePrescaler' from 'SPI_BAUDRATEPRESCALER_2' to 'SPI_BAUDRATEPRESCALER_8'. This slowed the transfer rate enough for it to run stable.

Thank you!

ED

closed 2019-04-03 by edkeefe

olikraus, 2019-04-03:
:)