olikraus/u8g2 #854: Change contrast for SSD1309 on Teensy

larsrengersen, 2019-04-03:
Hope you can help me, but as a beginner, perhaps getting this to work is out of my league.
I have a Teensy 3.6 and a Buydisplay green OLED with SSD1309 controller.

Found some clues here and there on the web but nothing that helped me to change the brightness / contrast of the screen.

My best bet was the [ContrastTest.ino](https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/ContrastTest/ContrastTest.ino) here on Github.
I can compile that sketch and see the counter and bars on the screen but (as I would have expected) the brightness is not changing.
I already updated to the latest version of the library (2.26.1).
Reading through the capabilities of the 1309 I concluded is should be possible, but perhaps I'm wrong and it's not supported.

Does anyone perhaps have a working example of changing brightness with Arduino with a SSD1309?

olikraus, 2019-04-03:
The contrast (brighness) setting is implemented according to the SSD1309 datasheet: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1309.c#L118

I do not know exactly whether I own a SSD1309 display, so maybe it never got tested.

larsrengersen, 2019-04-04:
Thanks for looking into it Oliver. Well, it the controller does not listen to the instructions indicated in it's own datasheet I am sure it's over my head debugging it.
[image]

Any change you can find time and are interested in getting this combination to work?
I'm happy to contribute by buying another display and have it shipped / delivered to you.
Otherwise I'll either search for another OLED to use in my project or live with quite a bright screen.

olikraus, 2019-04-04:
I have once ordered a SSD1309 display, but I do not know whether this is really a SSD1309 display. 
One difference to other displays seems to be, that the display power supply is generated outside of the display. So maybe this is one reason why the brightness setting is ignored.

This is the display in my lab, which was sold as a SSD1309 display: https://github.com/olikraus/u8g2/wiki/gallery#13-jan-2017-ssd1309-128x64

Hmm I think I never tested contrast change with this...

olikraus, 2019-04-04:
Under the assumtion, that the display, which was sold as a SSD1309 is really a SSD1309 display, I can confirm, that the contrast works. However, the differences are not very huge (as for any other OLED).
I mean, in the range from 0 to 255 the brightness changes is very small.

larsrengersen, 2019-04-05:
Thanks for the addional information. I guess the display I have perhaps than is not a true SSD1309 display.
That reference list of yours is very helpful!
The kind of brightness shown for that SSD1309 is what I am looking for.
I have found a similar green OLED from diymore (https://www.diymore.cc/products/2-42-inch-12864-oled-display-module-iic-i2c-spi-serial-for-arduino-c51-stm32-green-white-blue-yellow) like yours (but only with one row of pins at the side) . I ordered one and will post the results.
From your list I also liked the NHD 3.12 white 256x64 OLED with SSD1322 controller. Perhaps that is an option too.
Thanks for your help and I will share the results once the screen is in.

larsrengersen, 2019-04-29:
The display from diymore has arrived. Unfortunately also no visible change in brightness/contrast.
I published a video on https://imgur.com/a/QuQOzDz
Guess I'll just get some tinting foil used on cars and apply that on the screen for a fixed reduced brightness.

olikraus, 2019-04-29:
well... :-/

closed 2019-05-05 by olikraus