olikraus/u8g2 #855: Change display 128x128 white to 128x128 colour (waveshare / from u8g2 to ucg)

fabianpusch, 2019-04-04:
Hello 
I have build up a remote with the follwing display (a friend of mine helped me a lot to do this).
https://www.waveshare.com/wiki/1.5inch_OLED_Module 
To use this diplay I have used the u8g2 lib.
Now I woul like to change to another display
https://www.waveshare.com/wiki/1.5inch_RGB_OLED_Module
which uses the ucg lib.
I thought it wouldn´t be a big deal, but I am totally confused now.
Is there an easy way to translate the two libaries?
Kindb Regards
Fabian

Robiv8, 2019-04-04:
That would also interest me very much.
Have the same display

olikraus, 2019-04-04:
> Is there an easy way to translate the two libaries?

No. 

There at least two fundamental differences. 
1. Monochrome vs. RGB Color
2. 2KB vs. 32KB Display RAM

These two properties have a huge impact on the library.

fabianpusch, 2019-04-04:
Okay, thank you for the quick answer 😃

So is there a good starting point or do I have to rewrite the whole code?

olikraus, 2019-04-04:
There should be a hello world example ;-)

The huge difference in the display RAM size has one important impact: There is no double buffering like in u8g2. Instead any draw operation in ucglib will directly.

fabianpusch, 2019-04-04:
Oh that is really an argument against this dieplay. So I will keep the old display.
Thank you for your answers 😃

closed 2019-04-22 by olikraus