olikraus/u8g2 #856: Software I2C on STM32DUINO glitches on clock and data

jeffs555, 2019-04-04:
Could not get U8g2 to work with a SSD1306 using software I2C on a STM32F108 with the STM32DUINO core. A logic analyzer showed glitches on the SCL and SDA lines. 

I was able to fix it by editing U8x8lib.cpp to swap the order of every pinMode and digitalWrite command. All through the code, it was doing a pinMode(pin,OUTPUT) and then doing a digitalWrite. That is an invitation for glitches in the output. I swapped all of them to do the digitalWrite first before changing the mode to OUTPUT. I also added a digitalWrite(pin,1) before each pinMode(pin,INPUT_PULLUP) as that will speed up the rise time.

Haven't tried it on any other Arduino processors, but would most likely not hurt any of them.
Here is the U8x8lib.cpp that I modified.
[U8x8lib.zip](https://github.com/olikraus/u8g2/files/3045212/U8x8lib.zip)

olikraus, 2019-04-04:
Hmm... I understand the point, but I am not sure what to do here. In general HW supported I2C should be prefered.
Testing effort would be huge if I would change this.

jeffs555, 2019-04-05:
Probably no need to change anything unless you get a ton of similar issues. This was mainly for info in case someone else had the problem. Still not exactly sure what in the STM32DUINO core makes it glitch, when it works so well on other Arduino cores. Just know that doing pinMode change before digitalWrite looked a little suspicious and changing the order made it work for me. 

I had no success with HW I2C on the STM32DUINO core.

Thanks for all your work on this library.

closed 2019-04-05 by jeffs555

olikraus, 2019-04-05:
ok, good information :-)

mpmarks, 2021-09-13:
Thanks for your updated library jeffs555. I had the same configuration as you, could not get the display to work, saw the gliches on my logic analyzer and switched to your modified code. It solved the problem perfectly.