olikraus/u8g2 #857: Add a function to draw a given function

MicroTransactionsMatterToo, 2019-04-04:
Figured it could be useful to have a function that accepts a function pointer containing draw commands, then runs them in the draw loop for non-full buffer setups. So, something along the lines of

```cpp
void U8G2::draw(void (*drawFunction)(void)) {
    this.firstPage();
   do {
        drawFunction();
    } while ( this.nextPage() );
    return;
}
```

Any thoughts?

MicroTransactionsMatterToo, 2019-04-04:
Assuming the use of C++, this would also allow the use of Lambda functions. It's not really necessary, but I think it'd be a nice touch

olikraus, 2019-04-05:
Ok, but you have already provided such a function in your initial post. It could be easily added by any user.

MicroTransactionsMatterToo, 2019-04-05:
I get your point, I just thought it might be nice to implement in the library. However, if you don't feel there's any reason to, I'll just close this issue :)

closed 2019-04-05 by MicroTransactionsMatterToo

olikraus, 2019-04-05:
yes, but thanks for the idea.