olikraus/u8g2 #858: drawStr() causes crash and reboot when u8g2 is inside class

pman92, 2019-04-05:
Im developing a project in which I would like a seperate class to handle all of my display stuff. I have found another 2 cases of this reported here but no solution was found. 

To overcome the other common issue people have reported I've created an init function in my class in which the u8g2 constructor is (U8G2_ST7920_128X64_F_HW_SPI), and which calls u8g2.begin(). This init function is called within the main arduino setup() function. 
Everything complies and works fine as long as I don't attempt to draw to the display. 

As soon as I call a function of my class to draw something, the Arduino crashes and reboots. 

I've found that the crash happens when drawStr() is called, by placing Serial.print debug messages and then long delay()'s before each step of the drawing function. 
U8g2.clearBuffer() works fine.
U8g2.setFont() works fine.
As soon as I then call drawStr() the arduino reboots. 

I can create a simplified example with only relevant code if required.

Thanks

pman92, 2019-04-05:
I have tried to create a simple example to show the problem, but I'm having a different (unrelated) issue with it (probably with how I am declaring classes or something).
See here: [https://forum.arduino.cc/index.php?topic=608178.0](url)

olikraus, 2019-04-05:
1. "initialize()" allocates a local u8g2 object which is released at the end of initialize() function body, leaving an unititialized u8g2 member object.

2. You should not use the U8G2 class by itself to allocate an object, however defining a pointer to U8G2 objects is ok. Note, that U8G2_ST7920_128X64_F_HW_SPI is derived from U8G2

My suggestion is to use a pointer to U8G2 objects as class member only. Then pass the pointer to the u8g2 object as argument to your constructor function.

pman92, 2019-04-05:
That makes sense.
I've tried again using pointers in my project, but it now stops after setFont() is called. It doesn't crash and reboot, it just freezes and hangs forever. 
I've updated the simple example to show how I'm doing it and posted it again here:
https://forum.arduino.cc/index.php?topic=608178.0
I still cant compile the simple example because of something else I'm doing wrong.
Thanks

olikraus, 2019-04-05:
So it crashes or doesn't compile?
And as AWOL pointed out, you have reused the previously declared identifier: You should carefully read the error message.

pman92, 2019-04-06:
That example doesnt compile (since I've reused the identifier). 
But my actual project (where I haven't reused the identifier) does compile and run, but either hangs or crashes and doesn't reboot after setFont() is called. 
I will change the example sketch object name to something other than DISPLAY, so it should compile fine, and retest to confirm

pman92, 2019-04-07:
Once I changed the identifier in the example sketch it compiles and work's flawlessly. 

However in my real project it still freezes as soon as setFont() is called. 
It never gets past the setFont function, but does not crash and reboot. 

I will do further investigation and try to identify what is different about my real sketch that is causing the problem. 

My actual project uses 45% of RAM (leaving 4440 bytes free) on the Arduino Mega. The example only uses 21% (Leaving 6444 bytes). That should still be plenty available?

pman92, 2019-04-07:
Issue is fixed! Was just another mistake I had made. 
I was passing a pointer to u8g2 to the class in the initialize function. But the test drawing I was trying to do was happening before I ran that function (so my display class didn't have a pointer to u8g2).

All fixed and seems to be working now.

Thanks for your help!

closed 2019-04-07 by pman92

olikraus, 2019-04-07:
:)