olikraus/u8g2 #860: Testing ST7920 at MKR1000

ginanjarfm, 2019-04-07:
Hi. Currently I'm testing ST7920 128x64 display on MKR1000 (SAMD/3.3V) and facing problem the display didn't show anything. Does anyone have solution? [Here is the picture.](https://i.ibb.co/HKpZMjk/Whats-App-Image-2019-04-07-at-7-13-15-AM.jpg)

[image]

Pin connection MKR1000 to ST7920
Pin 9 SCK --> Pin E
Pin 8 MOSI --> Pin R/W
Pin 10 --> Pin RS

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

#define MKR_EC 9
#define MKR_RW 8
#define MKR_RS 10

U8G2_ST7920_128X64_F_SW_SPI u8g2(U8G2_R0, /* clock=*/ MKR_EC /* A4 */ , /* data=*/ MKR_RW /* A2 */, /* CS=*/ MKR_RS /* A3 */, /* reset=*/ U8X8_PIN_NONE);
//U8G2_ST7920_128X64_F_HW_SPI u8g2(U8G2_R0, /* CS=*/ MKR_RS, /* reset=*/ U8X8_PIN_NONE);

void setup(void) {
  u8g2.begin();
}

void loop(void) {
  u8g2.clearBuffer();         // clear the internal memory
  u8g2.setFont(u8g2_font_ncenB08_tr); // choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  delay(1000);  
}
```

Already read [u8gsetup](https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#st7920-128x64-2) but I'm not sure which constructor suit to my problem. Thank you very much.

olikraus, 2019-04-07:
The MKR is a 3.3V board, however the ST7920 is a 5V display: The MKR can not drive the ST7920 directly, you need level shifters for the digital signal lines.

ginanjarfm, 2019-04-07:
Ah, it's weird and true. Since there are two different datasheets [ST7920 VDD (4.5 - 5.5)](http://www.digole.com/images/file/Digole_12864_LCD.pdf) and [old ST7920 VDD (2.7 - 5.5)](http://www.hpinfotech.ro/ST7920.pdf), I thought I use the old one.

NB: Unfortunately the old one itself requires min logical H 0.7 * VDD.

closed 2019-04-07 by ginanjarfm