olikraus/u8g2 #861: No prototypes

Glett, 2019-04-07:
Hello!
I'm trying to port u8g2 library to stm32 with i2c interface.
I'm basing my work in this tutorial: http://elastic-notes.blogspot.com/2018/10/u8g2-library-usage-with-stm32-mcu.html
I just made all the porting functions and code, but when I try to compile , the compiler says that a lot of functions prototypes are missing.
It is possible that some .h is out of date?
Thanks for your answer!
Regards

olikraus, 2019-04-07:
> It is possible that some .h is out of date?

You can get the latest .h from this repo.

Glett, 2019-04-07:
Thank you for your answer.
I believe that I did not explain my well.
I just download the project with the right green button from github webpage  and a lot of prototype functions do not appear into u8g2.h.
For example: 
void u8g2_draw_hv_line_2dir(u8g2_t *u8g2, u8g2_uint_t x, u8g2_uint_t y, u8g2_uint_t len, uint8_t dir);
nor
void u8g2_read_font_info(u8g2_font_info_t *font_info, const uint8_t *font)
Are you sure that all the new code are committed?
Thanks
Regards!

olikraus, 2019-04-08:
These functions are local functions and should not be called from outside the file in which they are defined. In C an extra declaration is not required.
For example "u8g2_read_font_info" appears in "u8g2_font.c". It is also called in the same file. No prototype required.

Glett, 2019-04-08:
Sorry my friend, but this function u8g2_draw_hv_line_2dir is defined in u8g2_hvline.c and used in u8g2_setup.c.
You do not need to declare a function in C99 only if its declaration is above its use, in the same file, also, is a very good practice to define your functions, you can read a lot about that subject on Internet (for example: https://stackoverflow.com/questions/2575153/must-declare-function-prototype-in-c)
Anyway, I defined all the functions and I can't make it work in a oled 0.96" with ssd1306 controller,  using a stm32f103C8 with native Hal libraries.
I will work on it for some days and return with more questions I guess :)) LOL!
Thank you for your time!

olikraus, 2019-04-08:
> Sorry my friend, but this function u8g2_draw_hv_line_2dir is defined in u8g2_hvline.c and used in u8g2_setup.c.

I am also sorry to say, that this is some kind of an exception and the prototype itself is there in u8g2_setup.c:

https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_setup.c#L294

Glett, 2019-04-08:
Oh! I see now.
What compiler are you using?
I think my compiler is trying to help me to write more secure and readable code ;). 
You can close this issue if you want.
Thanks a lot!

closed 2019-04-08 by Glett

olikraus, 2019-04-09:
I have gcc here with usually all warnings activated.