olikraus/u8g2 #862: how to reduce ram size (u8g2_d_memory)

amigomcu, 2019-04-10:
hi all,
in my platform(stm32f105R8, flash / ram size is 64KB) ,       
so , the default setting, the u8g2_d_memory ZI size is big size  128KB
now i try to redefine some function buf[x] size. as follow code.

 12          4          0          0     131672      55280   u8g2_d_memory.o
uint8_t *u8g2_m_48_30_f(uint8_t *page_cnt)
{
  /*static uint8_t buf[11520];*/
  static uint8_t buf[1];
  *page_cnt = 30;
  return buf;
}

https://i.imgur.com/QBM0PWz.png
https://github.com/olikraus/u8g2/wiki/u8g2optimization

olikraus, 2019-04-10:
What exactly do you want to do? Reducing the size of the buffer will not help, because the software expects the buffer of the specified size and will write to the buffer as usual, overwriting any other RAM.
To be more clear: It is a very bad idea to change the buffer sizes in u8g2_d_memory.c.

Use a different constructor instead.

amigomcu, 2019-04-10:
hi  olikraus 

thank you for  reply my question.

constructor for (SSD1306 u8g2_Setup_ssd1306_i2c_128x64_noname_f)
which use u8g2_m_16_8_f() ,i don't revise it . 

other  function(i.e u8g2_m_48_30_f() ) have not called,  But it still occupies space.  
so , i change that buffer size .

/* ssd1306 f */
void u8g2_Setup_ssd1306_i2c_128x64_noname_f(u8g2_t *u8g2, const u8g2_cb_t *rotation, u8x8_msg_cb byte_cb, u8x8_msg_cb gpio_and_delay_cb)
{
  uint8_t tile_buf_height;
  uint8_t *buf;
  u8g2_SetupDisplay(u8g2, u8x8_d_ssd1306_128x64_noname, u8x8_cad_ssd13xx_fast_i2c, byte_cb, gpio_and_delay_cb);
  buf = u8g2_m_16_8_f(&tile_buf_height);
  u8g2_SetupBuffer(u8g2, buf, tile_buf_height, u8g2_ll_hvline_vertical_top_lsb, rotation);

olikraus, 2019-04-10:
> other function(i.e u8g2_m_48_30_f() ) have not called, But it still occupies space.
so , i change that buffer size .

Hmm... Then you must change many more buffers.
Instead of doing so, u8g2 assumes, that you will use the linker garbage collector.

JamesGKent, 2019-04-24:
@olikraus, unfortunately some compilers do not throw away the unused buffers in that file, I've hit this recently myself, the only way to make the library play nice under my environment was to comment out all the unused buffers.
The way I actually did this was to comment them all out and then declare the size I needed at the top of the file along with a comment explaining why.

I can't think of a graceful fix for this, but I know I'm not the first person to hit it, the best "solution" I can think of would be to add a commented out macro near the top of the file along with an explanation that when uncommented uses #ifdef blocks to remove all the buffers and force the user to manually declare their own, I don't know if this is something you would think of including or if you can think of a better fix?

olikraus, 2019-04-24:
1) U8g2 is a Arduino Lib

U8g2 was designed as a Arduino Library. The linker garbage collector is essential part of the Arduino Development IDE, so I think it is valid to assume, that U8g2 can use and build on this feature. 
The only uC line which does not support this (to my knowledge) are the low cost/free PIC1X compilers.

For all the ARM based systems (like the stm32f105R8 mentioned at the start), you could use gcc which includes the linker garbage collector. It just needs to be activated.

B) u8g2_d_memory is not the only problem.

The garbage collector takes care on both: RAM and FLASH-ROM consumption. Your suggestion addresses u8g2_d_memory. But there is much more to consider. So your suggestion would just be a small starting point.

But anyhow: Porting u8g2 to other platforms is ok, but it is not really supported from my end (esprecially because of the garbage collection problem ;-)

closed 2019-05-05 by olikraus

JamesGKent, 2019-05-16:
@olikraus FYI keil MDK: http://www2.keil.com/mdk5/ has this issue if you use the ARM compiler, enabling the various options that *should* strip this out don't seem to have any effect. in order to make this work correctly under keil I had to enable to "single function per object" setting and modify the u8g2_d_memory file.

olikraus, 2019-05-16:
I am not in the position of buying a keil IDE, but the plain&official arm gcc compiler does support this with the correct (and usual) commandline options. No modification is required in the code. 

https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/u8x8_test/Makefile#L99
https://github.com/olikraus/u8g2/blob/master/sys/arm/stm32l031x6/u8x8_test/Makefile#L104

JamesGKent, 2019-05-16:
i shall have to give that a try, although when I previously tried this through keil it didn't behave as expected, that may have been due to working on the eval version.

FYI there is an eval version that is free that reduces some of the capability such as the maximum code size.

olikraus, 2019-05-16:
I once tried to port u8g2 to the microchip PIC compiler suite. It turned out that the free PIC compiler is a reduced/restricted GNU compiler. The garbage collector functionality was intentionally removed and is only actived with the professional compiler version.

I will not waste my time again trying to get a reduced free commercial product to run.