olikraus/u8g2 #863: Using display with SSD 1322

JelleHaas, 2019-04-10:
Hi all,

I am working on a project where it is possible to type with a keyboard and ASCII code on a display. I was using the u8glib library but found some difficulties with the picture loop and what I want to achieve. I have found the u8x8 library and think with this it should be possible.

The problem I have is that I cannot get anything on the display, wich I could with u8g. Is there something wrong with my code?

I use a MKRZERO with this display: 
https://nl.aliexpress.com/item/Real-Oled-display-5-5-256-64-25664-Dots-Grafische-Seral-SPI-LCD-Module-Display-Screen/32918084776.html?spm=a2g0s.9042311.0.0.27424c4dTK8Jiq

This is the code I got something on the display with the u8glib:

_#include "U8glib.h"

U8GLIB_NHD31OLED_GR u8g(12, 8, 10, 9);	// SPI Com: SCK = 13, MOSI = 11, CS = 10, A0 = 9

void draw2(void) {
  // graphic commands to redraw the complete screen should be placed here  
  u8g.setFont(u8g_font_unifont);
  //u8g.setFont(u8g_font_osb21);
  u8g.drawStr( 0, 10, "Regel 1 !");
  u8g.drawStr( 0, 20, "Regel 2 ");
}

void draw3(void) {
  // graphic commands to redraw the complete screen should be placed here  
  u8g.setFont(u8g_font_unifont);
  //u8g.setFont(u8g_font_osb21);
  u8g.drawStr( 0, 10, "Regel 1 !");
  u8g.drawStr( 0, 20, "Regel 2.2 ");
  u8g.drawStr( 0, 30, "Regel 3 Dit is regel 3 ");
}

void setup(void) {
  
  // flip screen, if required
  // u8g.setRot180();
  
  // set SPI backup if required
  //u8g.setHardwareBackup(u8g_backup_avr_spi);

  // assign default color value
  if ( u8g.getMode() == U8G_MODE_R3G3B2 ) {
    u8g.setColorIndex(255);     // white
  }
  else if ( u8g.getMode() == U8G_MODE_GRAY2BIT ) {
    u8g.setColorIndex(3);         // max intensity
  }
  else if ( u8g.getMode() == U8G_MODE_BW ) {
    u8g.setColorIndex(1);         // pixel on
  }
  else if ( u8g.getMode() == U8G_MODE_HICOLOR ) {
    u8g.setHiColorByRGB(255,255,255);
  }
}

void loop(void) {

  u8g.firstPage();  
  do {
    draw2();
  } while( u8g.nextPage() );

  u8g.firstPage();  
  do {
    draw3();
  } while( u8g.nextPage() );
  
  // rebuild the picture after some delay
  delay(50);
}_

This is the new code for u8x8:

_#include <U8x8lib.h>

// Please UNCOMMENT one of the contructor lines below
// U8x8 Contructor List
// The complete list is available here: https://github.com/olikraus/u8g2/wiki/u8x8setupcpp
// Please update the pin numbers according to your setup. Use U8X8_PIN_NONE if the reset pin is not connected

U8X8_SSD1322_NHD_256X64_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 8, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 7);
//U8X8_SSD1322_NHD_256X64_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 7);
//U8X8_SSD1322_NHD_128X64_4W_SW_SPI u8x8(/* clock=*/ 13, /* data=*/ 8, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);
//U8X8_SSD1322_NHD_128X64_4W_HW_SPI u8x8(/* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);

// End of constructor list

void setup(void)
{
  /* U8g2 Project: SSD1306 Test Board */
  //pinMode(10, OUTPUT);
  //pinMode(9, OUTPUT);
  //digitalWrite(10, 0);
  //digitalWrite(9, 0);

  /* U8g2 Project: KS0108 Test Board */
  //pinMode(16, OUTPUT);
  //digitalWrite(16, 0);

  u8x8.begin();
  u8x8.setFont(u8x8_font_8x13_1x2_f);
  u8x8.setPowerSave(0);

}

void loop(void)
{
  u8x8.clear();
  u8x8.setCursor(0, 11);
  u8x8.print("Hello World!");
  //  u8x8.refreshDisplay();    // only required for SSD1606/7
  //delay(100);
}_

closed 2019-04-15 by JelleHaas

olikraus, 2019-04-15:
11 seems to be wrong. This is 88 as pixel value and is outside the display.

JelleHaas, 2019-04-16:
Yes it is, I found it eventually. Sorry for the stupid mistake/question and thank you for the help

olikraus, 2019-04-16:
no problem.