olikraus/u8g2 #864: Change Display size or increase boundaries

josepetridelima, 2019-04-10:
Hi! I am working in a menu with a sliding animation, and each menu item uses all screen size, when I slide the menu to next option an animation happens and a little bit of early option appears for a second in the new option. 
This occurs because the screen has an outside buffer, and the image on the screen still happens after the screen boundaries. But this space is not infinite, and after this end, it starts to appear on the opposite side. 
I would like to stop it. It like a tile effect. When something vanishes in one corner and appears on the opposite side. 
There is a way to avoid that? I tried setMaxClipWindow, setBufferCurrTileRow, updateDisplayArea and it didn't work for me =( 
I sent a video to youtube, so you can understand better my problem here. 

[https://youtu.be/KASK4QrD1As](url)

Thank you for your attention.

olikraus, 2019-04-10:
I personally assume, that there is a software bug in your code, but if not: You could try to activate 16 bit mode (see FAQ) for u8g2. Maybe this works.

josepetridelima, 2019-04-10:
Thank you very much! It's working!!! 
I am improving my code, but I didn't find a way to solve that problem, but with this 16 bit active the problem disappear! 
A think the space outside the screen where the image is formed take bigger with 16 bit active, so the image has space to slide and don't appears again in the opposite side =)

[https://youtu.be/ma4_BcujcBA](url)

olikraus, 2019-04-10:
:)

closed 2019-04-22 by olikraus