olikraus/u8g2 #865: LD7032 in I2C mode not working with u8g2

Majestix1970, 2019-04-11:
Hi,

my 0.5" OLED display with LD7032 driver is not working in I2C mode (R5 resolder accordingly...)
SPI mode (R4 connected) is working only in SW mode on my Arduino Mirco and MicroPro.

Can anybody tell me please how I need to set it up to get it working in I2C HW modewith u8g2 lib
Thanks!

note: only the constructor for SW SPI mode is working (if uncommented :-)) none of the HW constructors are workin!?

here is my code:

#include <U8g2lib.h>

//U8G2_LD7032_60X32_F_4W_SW_SPI u8g2(/*rotation*/U8G2_R0, /* clock=*/ 3, /* data=*/ 2, /* cs=*/ 7, /* dc(A0)=*/ 6, /* reset=*/ 8);
//U8G2_LD7032_60X32_F_2ND_4W_HW_SPI u8g2(/*rotation*/U8G2_R0,/* cs=*/ 7,/* dc=*/ 6,/* reset=*/ 8);

//U8G2_LD7032_60X32_F_SW_I2C u8g2(/*rotation*/U8G2_R0,/* clock=*/ 3,/* data=*/ 2); //, /*reset=*/ 8)
U8G2_LD7032_60X32_F_HW_I2C u8g2(/*rotation*/U8G2_R0, /*reset=*/ 8,/* clock=*/ 3,/* data=*/ 2);

void setup(void) {
  u8g2.begin();
}  

void loop(void) {
  u8g2.clearBuffer(); 
  u8g2.sendBuffer();
  u8g2.setFont(u8g2_font_profont10_tf);
  u8g2.drawStr(0,20,"Hello World!");
  u8g2.sendBuffer();
  delay(200);
}

Majestix1970, 2019-04-11:
Sorry I forgot to mention which display I have exactly...

https://www.dfrobot.com/wiki/index.php/0.5_Inch_OLED_Display_SKU:DFR0376#How_to_connect_Oled_PCB_and_Display

[image]

Majestix1970, 2019-04-11:
[OCELL_use_guide2.pdf](https://github.com/olikraus/u8g2/files/3070442/OCELL_use_guide2.pdf)

v1993, 2019-04-11:
AFAIK, on most arduino boards HW I2C pins can not be remapped (passed to constructor values do actually nothing), so you should find which ones are correct on your board.

Qucik googling tells that they are `2/SDA and 3/SCL`. Note that you can actually use any pin for `reset`. But because it looks like you use them already, you probably should try omiting last two arguments like this:

```CXX
U8G2_LD7032_60X32_F_HW_I2C u8g2(U8G2_R0, 8);
```

P.S.: don't wanna sound rude, but can you, please, read messages before sending? My eyes hurt a little because of your punctuation… Also, use ``` to format code pls.

olikraus, 2019-04-11:
Indeed, for hardware driven I2C or SPI you must figure out which pins are forseen on Arduino Mirco and MicroPro boards for these kind of communication. You can only use these pins.

v1993, 2019-04-12:
BTW @olikraus do you remember note that SW I2C don't wait for ACK and HW one do? It also may take place here I think.

Majestix1970, 2019-04-12:
Hi All
Thanks for your replies.

Pin 2 and 3 are the correct HW pins since other I2C devices are responding correctly.
Also tried already "omiting last two arguments" => no success....

Either my Display's I2C  is defect or the lib is not working correctly....
I will have a look into the cpp code to check what is the root cause ;-)

Just wanted to double check if I did some "big dummy" mistake...
Looks like I did not. Any way.

Sorry for my "punctuation" => I am not a native speaker.

Thanks so far!

olikraus, 2019-04-12:
I crosschecked this. I2C was never tested. We need to check whether the interface has been implemented correctly. Is the I2C protocol the same as for ssd1306, then it should work

Majestix1970, 2019-04-12:
Thanks for your reply!
Dont know... I need to check as well. but good hint.

Hope I finde some time on the weekend :-)
Lets see who is faster.
Thanks

Majestix1970, 2019-04-14:
Hi olikraus,

I did some test with LD7032 and HW SPI => does also not work.
How do you define which HW pins are used?
Thanks

olikraus, 2019-04-14:
Hw pins are defined by your board.

Majestix1970, 2019-04-22:
Hi  olikraus
I am sorry, but somehow I am unable to find the file/code location where you define the I2C address used for the LD7032 OLED

According to the SSD1306 datasheet the LD7032 has the same setup protocoll...

Found some difference in your init sequence (in u8x8_d_ld7032_60x32.c) and the init sequence suggested by the manufacture. changed it, butstill no success...

Would be great if you could tell me in which file of your lib I can find the definition of the I2C address or how you define the I2C address
Thanks a lot!

Majestix1970, 2019-04-22:
one more thing: the display itself does not have a possibility to set the I2C address via jumper... according to datasheet must be setup at the I2C startup via SW
format of the address definition (LD7032: b7:b0 -> 0110 ID2 ID1 ID0 R/W#) is different to SSD1306(b7:b0 -> 011110 SA0 R/W#)
Might this be the reason why the display does not come up? Unfortunately I do not find the location in your code where this is defined...
Thanks for your help 
And the great LIB you share with us!!

olikraus, 2019-04-22:
You can use https://github.com/olikraus/u8g2/wiki/u8g2reference#seti2caddress to assign a different address. Maybe the default address is wrong for the LD7032. I assume you need to use `u8g2.setI2CAddress(0x60);`

Majestix1970, 2019-04-23:
Hi olikraus,

thanks for your fast reply! Unfortunately it did not help..
the code:
U8G2_LD7032_60X32_F_HW_I2C u8g2(U8G2_R0, U8X8_PIN_NONE);  // SCL=3, SDA= 2;

void setup(void) {
  u8g2.setI2CAddress(0x60); //also tried e.g. 0x66... no success
  u8g2.begin();
}  

Can you please tell me where you define the default I2C address used for LD7032

Thanks

olikraus, 2019-04-23:
> Can you please tell me where you define the default I2C address used for LD7032

Here: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_cad.c#L650

Majestix1970, 2019-04-23:
Thanks(Danke!)
But so far nothing is working…. I will double check the HW itself

Do you know out of your head if you did program the graphic mem by respecting the 64x32 bytes

Beste Regards
Eike

olikraus, 2019-04-23:
I definitly tested this device. Can you try SW I2C instead? Will software emulated I2C work? 
Are you able to observe the traffic on the I2C bus?

Majestix1970, 2019-04-24:
Hi olikraus,

SW I2C does also not work. I tried both HW and SW.
In general I2C works fine. If I use another device it responds fine ☹

Maybe I do something wrong in the wiring the display.
Do you have a wiring diagram? Did you only connect SDA,SCL,3.3V and GND. Or did you tie the other pins(A0, CS, Reset) high or low too?
Thanks for all your help so far.
Best Regards
Eike

olikraus, 2019-04-24:
> Do you have a wiring diagram? Did you only connect SDA,SCL,3.3V and GND. Or did you tie the other pins(A0, CS, Reset) high or low too?

Usually SDA and SCL should be good enough. All other connections should be described by the display vendor.

Majestix1970, 2019-04-24:
Found It!!!
there was an error in the code sending data to the display.
after changing the code in 8x8x_cad.c (see attached picture) the number of bytes send in a strem from 24 to 64 it works! took a while to get through your code ...
[image]

Would be great if you could double check and distribute it with your next update.
Best Regrads
Eike

Majestix1970, 2019-04-24:
one more thing....

There must be another error in your code since the HW Constructor is not working. Only the SW Constructor is working on my Arduino Micro and MicrPro....

Any idea what that might be?
Or some hints where I need to look within your code (saves time searching :-) )

olikraus, 2019-04-24:
The Arduino Code itself restricts any I2C transfer to 32 bytes. This is why val is set to 24 (which is less than 32). Actually the purpose of the above code is to cut down code into smaller pieces if there is a need of transfering more than 32 bytes.

In other words: Increasing the size to 64 will not solve the problem, because of the 32 byte restriction in the Arduino Code. However size of 64 may work with SW I2C because SW I2C bypasses the Arduino Code (and its restrictions).

Majestix1970, 2019-04-24:
Sorry I forgot:
I also changed the init sequence in u8x8_d_ld7032_60x32.c (line 47 to 70) for the LD7032 OLED with 60x32 pixel:
(sorry do not now how to post the code correctly ....
U8X8_CA(0x02, 0x00),		  		/* Dot Matrix Display OFF */
  U8X8_CA(0x14, 0x00),		  		/* Stand-by OFF, OSCA Start */
  U8X8_CA(0x1a, 0x04),		  		/* Dot Matrix Frame Rate,  special value for this OLED from manual 4 => 120Hz*/
  U8X8_CA(0x1d, 0x00),		  		/* Graphics Memory Writing Direction: reset default (right down, horizontal) */
  U8X8_CA(0x09, 0x00),	      		/* Display Direction:  reset default (x,y: min --> max) */
  U8X8_CAA(0x30, 0x00, 0x3B),  		/* Display Size X, Column Start - End 0-0x3b(59)*/
  U8X8_CAA(0x32, 0x00, 0x1F),  		/* Display Size Y, Row Start - End 0-0x1f(31)*/
  U8X8_CA(0x34, 0x00),				/* Data Reading/Writing Box X start */
  U8X8_CA(0x35, 0x07),				/* Data Reading/Writing Box X end */
  U8X8_CA(0x36, 0x00),				/* Data Reading/Writing Box Y start */
  U8X8_CA(0x37, 0x1F),				/* Data Reading/Writing Box Y end */
  U8X8_CA(0x38, 0x00),        		/* Display Start Address X */
  U8X8_CA(0x39, 0x00),        		/* Display Start Address Y */
  U8X8_CA(0x10, 0x00),		  		/* Peak Pulse Width Set: 0 SCLK */
  U8X8_CA(0x16, 0x00),		  		/* Peak Pulse Delay Set: 0 SCLK */
  U8X8_CA(0x12, 0x40),		  		/* 0x32, 0x50 or 0x40 Dot Matrix Current Level Set: 0x050 * 1 uA = 80 uA */
  U8X8_CA(0x18, 0x03),		  		/* Pre-Charge Pulse Width: 3 SCLK */
  U8X8_CA(0x44, 0x02),		  		/* Pre-Charge Mode: Every Time */
  U8X8_CA(0x48, 0x03),		  		/* Row overlap timing: Pre-Charge + Peak Delay + Peak boot Timing */
  U8X8_CA(0x17, 0x00),          	/* Row Scan */
  U8X8_CA(0x13, 0x00),          	/* Row Scan Sequence Setting */
  U8X8_CA(0x1C, 0x00),          	/* Data Reverse */
  U8X8_CA(0x3f, 0x11),		  		/* VCC_R_SEL: Internal Regulator enabled(D4=1) and VCC_R=VCC_C*0.7(D0=1) */
  U8X8_CA(0x3d, 0x00),		  		/* VSS selection: 2.8V */

olikraus, 2019-04-24:
hmmm... why was there a need to change the init code? It did work for me.

Majestix1970, 2019-04-24:
yes it did.... but it seems to create "display pixel errors" 
With my init sequence it looks more stable => no pixel errors while updating a page.

Majestix1970, 2019-04-24:
I just double checked... downloaded your lib again
updated the "24" with "64" and your old init sequence => works but shows display errors
update also the init sequence => no display erros any more

Majestix1970, 2019-04-24:
any idea why the "HW Cunstructor" is not working?

olikraus, 2019-04-24:
init sequence: ah, ok.... i guess i need to takeover your sequence

"HW Cunstructor": I tried to explain this above. I did not invent the "24", instead there is a "32" limit in the Arduino Code. You can google for the 32 byte limit for I2C on Arduino....
So again: Your increase from 24 to 64 will might solve the problem for software emulated I2C (because this will bypass the Arduino TWI lib), but you will run into the 32 byte limit of the Arduino Lib.

olikraus, 2019-04-24:
ToDo:
1. Takeover new init sequence
2. Check, why the I2C code is wrong and figure out how this can be fixed (changing 24 to 64 is NOT a solution)

Majestix1970, 2019-04-24:
on my Arduino Micro and MicroPro
the following does work/does not work:

U8G2_LD7032_60X32_F_4W_SW_SPI u8g2(U8G2_R0, 3, 2,  7,  6, 8) => works;

U8G2_LD7032_60X32_F_2ND_4W_HW_SPI u8g2(U8G2_R0,7, 6,8); => does NOT work

U8G2_LD7032_60X32_F_SW_I2C u8g2(U8G2_R0,/* clock=*/ 3,/* data=*/ 2, /*reset=*/8); => works

U8G2_LD7032_60X32_F_HW_I2C u8g2(/*rotation*/U8G2_R0, /*reset=*/8);  => does NOT work

any idea why or in which file of your code I have to look for the error?

Majestix1970, 2019-04-24:
Thanks for taking over!
waiting for an update. so far I live with "64 is no solution" :-)

olikraus, 2019-04-24:
Actually I am a little bit confused.

> U8G2_LD7032_60X32_F_2ND_4W_HW_SPI u8g2(U8G2_R0,7, 6,8); => does NOT work

Clear, because there is no 2nd HW SPI in your system.

> U8G2_LD7032_60X32_F_SW_I2C u8g2(U8G2_R0,/* clock=/ 3,/ data=*/ 2, /reset=/8); => works

This is where I have doubts. It works with val=24 or with val=64 or with both? 
I assume: It works only with val=64, correct?

> U8G2_LD7032_60X32_F_HW_I2C u8g2(/rotation/U8G2_R0, /reset=/8); => does NOT work

Same questions: It does not work with any value, correct?

This is what I assume: There are two problems: One is the I2C protocol, which probably has an issue (which can be solved by using val=64, although i do not know why). The second problem could be the level shifter. It probably blocks the ACK response from the display, disallowing proper I2C. Software I2C in U8g2 ignores the ACK response (this is why it works), but HW I2C will expect the ACK response and will reviews to work. 

Actually the PCB of your display was designed by me. here is the schematic, which I created long back:
[image]

Once I handled over the schematic to the fab in China it got modified and extended with the I2C option. However, I guess, these people did not consider, that the ACK signal will not be transmitted in opposite direction due to the HC4050 on the PCB. 

Can you check whether there is a HC4050 chip on your board? If so, HW I2C will never work.

Majestix1970, 2019-04-25:
Hi Oli
1st: sorry there was a copy past error.  I did not use "_2ND_" did just ccheck on another Arduino.
2nd: No there is no HC4050 on the pcb. Only a power converter LT3494 and the LD7032.
[image]
[0.5OLED_Schematic.pdf](https://github.com/olikraus/u8g2/files/3118244/0.5OLED_Schematic.pdf)
Also I2C is working. But not stabel since it shows also with my init sequence display errors... much more then in SPI mode. makes me crazy...
By the way: I use a 5V to 3.3V converter inbetween the Arduino and the Display.

Did you had a chance to look into your I2C code?
Why is the "64" Solution I so far identified and make the display work "no good"?

Thanks
Eike

Majestix1970, 2019-04-25:
Logik Level Converter is bi-directional :-)

Majestix1970, 2019-04-25:
do you know how sensitive the display driver is regarding 5V?
Does it actually hurt it. Read on several sides that it can be used with 5V. Datasheet of the LD7032 says 3.3V max.  What is correct...

olikraus, 2019-04-25:
5V signals will probably destroy  the OLED. I would trust more the 3.3V from the datasheet.
Probably you need a special I2C level shifter.

> Did you had a chance to look into your I2C code?

Yes, it still looks good to me. I probably need to test this. But for testing, I have to find my own LD7032 in my lab, and I didn't start to search for it :-( It might take a while until I will find it... my lab is a little but crowded... filled up with many things... :-p

Majestix1970, 2019-04-25:
my lab too... know what you mean ;-)
What do you mean with "special I2C level shifter"
I have this one (marked for I2C)...
https://www.amazon.de/gp/product/B07F3P942R/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1

any other suggestion?

Majestix1970, 2019-04-25:
may be I order me a new display to be sure it is not a HW defect we chase....
was thnking to switch display with an ssd1306, but need the 0.5inch size and could not find somthing similar to this I currently have....
Any idea for a different display in this small size?

olikraus, 2019-04-26:
> I have this one (marked for I2C)...

Looks ok. Of course I have not tested this.
But: Are you sure that there is no other level shifter on the pcb? At least in the pic of your second post, there is a IC which looks like a level shifter.

> Any idea for a different display in this small size?

THe LD7032 is indeed the smallest OLED I know.

Majestix1970, 2019-04-26:
Yepp unfortunate…

I will see if I find the time to write my own tiny I2C graphic lib for this display…
I keep you posted.
But would love to use your lib in the future.… do not like to event wheels  all the time new ☺

Any idea why the “_HW_” constructor is not working either?

olikraus, 2019-04-26:
> Any idea why the “_HW_” constructor is not working either?

you probably mean HW_I2C because HW 4-wire SPI will work nicely.
The reason why HW_I2C will not work is the unidirectional level shifter, which will block the ACK response from display to uC.

0.5" OLED Module in my lab (official device): Includes unidirectional level shifter --> HW I2C can not work

0.5" OLED Module schematic (my version):  Includes unidirectional level shifter --> HW I2C can not work

0.5" OLED Module schematic (your version):  No level sifter --> HW I2C will work, but you must not use 5V uC (like Uno)

0.5" OLED Module schematic (picture from your second post): Some level shifter is visible on the PCB: Type of level shifter (uni/bidirectional) is unknown --> Not sure whether HW I2C works or not.

As you can see: THere is no reliable information. So you ask, whether HW I2C works or not. Answer is: I do not know. The information is not consistent.
Maybe you should put a picture for your device (both sides) here, to allow a precise analysis.

At the end it is like this: If there is a bidirectional communication between display and uC possible, then also HW I2C is possible.

> I will see if I find the time to write my own tiny I2C graphic lib for this display…

Sure, but it will not solve the problem of HW I2C....

Majestix1970, 2019-04-27:
Hi Oli

just to make sure...

Yes I am taking regarding HW_I2C.
(HW_SPI is also not working for whatever reason but I do not care about that right now...)

the level shifter is a bidirectional ( should be possible to see via the details I posted . let me know if you can not access the posted information)

The level shifter is used in other setups on the HW I2C pins with other devices and works perfectly fine... only the display using u8g2 in the HW_I2C config does not work...

My display is mounted into a housing(hot glued....) therefore I hesitated to remove it to make a photo. Perhaps I try to get it out and post you sone more detailed pics.
But I can 100% ensure you there is NO level shifter on the PCB at all. Only the components you also can see on the schematic I did post as well. Nothing more nothing less.

Outlook für iOS<https://aka.ms/o0ukef> beziehen

Majestix1970, 2019-04-27:
by the way... which second post of mine is showing a level shifter? As I can see in the whole thread there is only one post of the display. There is no level shifter. 

Also I posted the link to the level shifter I use and the description and HW setup of it shows it is bidirectional.

Please help me to under stand where my provided info is not consistent.

olikraus, 2019-04-28:
> by the way... which second post of mine is showing a level shifter? As I can see in the whole thread there is only one post of the display. There is no level shifter.
>
> Also I posted the link to the [...]

In your second post, the PCB is visible with a black blackground. Like this:

[image]

On this PCB there is a IC with 16 pins. This is the level shifter (i think). What is printed on this IC? Can you read the label on the IC? Or maybe you can post a close up picture of the label on that IC.

Majestix1970, 2019-04-29:
Hi Oli

Outlook für iOS<https://aka.ms/o0ukef> beziehen

[https://sha256systems.eu/WebRoot/Store27/Shops/5b7c8826-3528-4d6f-bd2d-c024fcdbd435/5991/9360/E165/8A61/7538/0A48/3670/6749/6W4A9451-900x600.jpg]<https://camo.githubusercontent.com/574ec1974573e05dffe8341a93ae7819a3c455cb/68747470733a2f2f73686132353673797374656d732e65752f576562526f6f742f53746f726532372f53686f70732f35623763383832362d333532382d346436662d626432642d6330323466636462643433352f353939312f393336302f453136352f384136312f373533382f304134382f333637302f363734392f36573441393435312d393030783630302e6a7067>

Majestix1970, 2019-04-29:
Hi Oli

you are right! I checked with the chinese manufacturer and he did first send me wrong schematic (a new one...) having no level shifter at all. The IC on the pcb is labled, but not readable. Even not with a magnifier glas... Considering the schematic from China I did wrongly assume that this IC is the LD7032. But the LD7032 is directly build into the display itself (Am I right?)
This means now it is clear why the HW Constructor is not working... ACK is blocked by the HC4050
Leaves only the SW issue open.
Thanks for all your help!
Best Regards
Eike

Outlook für iOS<https://aka.ms/o0ukef> beziehen

[https://sha256systems.eu/WebRoot/Store27/Shops/5b7c8826-3528-4d6f-bd2d-c024fcdbd435/5991/9360/E165/8A61/7538/0A48/3670/6749/6W4A9451-900x600.jpg]<https://camo.githubusercontent.com/574ec1974573e05dffe8341a93ae7819a3c455cb/68747470733a2f2f73686132353673797374656d732e65752f576562526f6f742f53746f726532372f53686f70732f35623763383832362d333532382d346436662d626432642d6330323466636462643433352f353939312f393336302f453136352f384136312f373533382f304134382f333637302f363734392f36573441393435312d393030783630302e6a7067>

olikraus, 2019-04-29:
> But the LD7032 is directly build into the display itself (Am I right?)

Yes correct. 

So, what is my todo? I need to update the init sequence, correct?

Majestix1970, 2019-04-30:
Hi Oli

Outlook für iOS<https://aka.ms/o0ukef> beziehen

Majestix1970, 2019-04-30:
Hi Oli
- update init sequence does not hurt
- fixing problem that SW_I2C for 60x32 LD7032 OLED is only working after changing „24 bytes to 64 bytes “ ( you said that this is just a workaround but not the real fix)
Thanks
Eike

Outlook für iOS<https://aka.ms/o0ukef> beziehen

olikraus, 2019-05-19:
At the moment I can not test the I2C interface. I think the code was never tested. Added a comment into the code...

olikraus, 2019-05-19:
Updated the init sequence. Tested with my own device: No visible change, so keeping the new sequence.

olikraus, 2019-05-19:
Not sure how to continue from here: The I2C cad driver is probably broken, but further analysis requires a I2C LD7032 test setup...

Majestix1970, 2019-08-29:
Hi Oli,

can I help you with something here?

olikraus, 2019-08-29:
Not sure... You have already suggested a workaround, so it works for you, right?

I think the problem is more, that I do not understand the workaround and my fear is, that the bug is somewhere else.

olikraus, 2020-01-18:
closing this for now...

closed 2020-01-18 by olikraus