olikraus/u8g2 #866: setCursor error

edkeefe, 2019-04-12:
Hello,

I developed a program using the U8G2 library on an Arduino Uno with an SSD1306 display. It worked perfectly.

I ported the code to an STM32F407VGT6. Everything works as expected except one command. 

In the Arduino code the line is (works):

u8g2.setCursor(20, 40);

On the STM32 I tried the following (doesn't work):

u8g2_setCursor(&u8g2, 20, 40);

I get the following error in Keil uVision:

Error: L6218E: Undefined symbol u8g2_SetCursor (referred from main.o).

I'm missing a file that contains this function. What is the name of the file that contains the missing code so I can included it?

Or is there something else I'm missing or doing wrong?

Thank you.

ED

v1993, 2019-04-13:
Here's a clue:

Learn C. Learn what comiling and linkage is and what are differences between them, learn what does "symbol" mean. Also, please, note that putting your multiline code into ``` makes it formated nicely and single ` does so for one-line code.

`Undefined symbol` means that you include definition but don't link implementation into projec somehow. It most likely mean that you miss some `.c` file and/or forgot to reconfigure your build system somehow.

This also may be (not likely tho) a problem in u8g2 if they forgot to update some source list, but who knows.

olikraus, 2019-04-14:
Note that the lower case uppercase spelling differs between u8x8 and u8g2. In this case "set" is different.

closed 2019-05-05 by olikraus

WangShuoran, 2024-06-15:
> Note that the lower case uppercase spelling differs between u8x8 and u8g2. In this case "set" is different.

I search your code and https://github.com/olikraus/u8g2/wiki/u8g2reference information, I do not find this function, Please confirm!

olikraus, 2024-06-16:
I think my comment was little bit wrong: The difference is between C and C++
The C++ part follows the Arduino naming convention, however for C i follow my own naming convention.
In the above mentioned link always both names (C and C++) are provided, like here: https://github.com/olikraus/u8g2/wiki/u8g2reference#setcontrast

WangShuoran, 2024-06-18:
> I think my comment was little bit wrong: The difference is between C and C++ The C++ part follows the Arduino naming convention, however for C i follow my own naming convention. In the above [...]

Think you for your quickly reply. But my question is setCursor have not C function, does it have a replace?

[image]

olikraus, 2024-06-19:
> Thank you for your quickly reply. But my question is setCursor have not C function, does it have a replace?

No, because the setCursor function will set the draw position for the "print()" command which is actually from the Arduino IDE (u8g2 print() function is the same as the serial.print() function). So indeed there is another difference between C and C++: Additionally has the print() and setCursor() commands.

WangShuoran, 2024-06-20:
Okay, can I understand that because there is an API directly in the C code to specify the display location, there is no need for setCursor and Print functions to display it?

olikraus, 2024-06-20:
Exactly. The C drawStr function expects the position as part of its arguments.
Moreover the C functions are intended to work without Arduino Environment, as a consequence there is no print() (and not setCursor()) command, because the print() function is part of the Arduino IDE.