olikraus/u8g2 #867: Something (setFont?) causing the compiler to stop

machmar, 2019-04-13:
when I try to upload the hello world sketch, It wont upload

`#include <U8g2lib.h>
U8G2_SSD1306_128X64_NONAME_F_SW_I2C u8g2(U8G2_R0, SCL, SDA, U8X8_PIN_NONE);

void setup() {
  u8g2.begin();
}

void loop() {
  u8g2.clearBuffer();          // clear the internal memory
  u8g2.setFont(u8g2_font_u8glib_4_tf); // choose a suitable font
  u8g2.drawStr(0,10,"Hello World!");  // write something to the internal memory
  u8g2.sendBuffer();          // transfer internal memory to the display
  delay(1000);  
}`

` C:\Users\marec\OneDrive\Dokumenty\Arduino\libraries\U8g2\src\clib\u8g2_setup.c: In function 'u8g2_draw_l90_r0':

C:\Users\marec\OneDrive\Dokumenty\Arduino\libraries\U8g2\src\clib\u8g2_setup.c:303:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files/windowsapps/arduinollc.arduinoide_1.8.21.0_x86__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

Using library U8g2 at version 2.25.10 in folder: C:\Users\marec\OneDrive\Dokumenty\Arduino\libraries\U8g2 
Using library SPI at version 1.0 in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\libraries\SPI 
Using library Wire at version 1.0 in folder: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\libraries\Wire 
exit status 1
Error compiling for board Arduino Pro or Pro Mini. `

But when I upload different sketch (fontUage for example), It works and runs perfectly.
Help

machmar, 2019-04-13:
I used the code thing so it isn't just like random text but It doesn't seem to work properly (here on GitHub)

v1993, 2019-04-13:
Hint: use ``` and not ` around multiline code.

v1993, 2019-04-13:
"Segmentation fault"… it means that there is something wrong  with your compiler, quite literally. Segmentation fault is serious failure inside crashing program for ANY possible input. U8g2 may cause this somehow, but it is compiler's bug (or probably not enough memory problem?).

machmar, 2019-04-13:
I forgot to ssay this but it stopped working after I updated it ower night. I might have broke it,

v1993, 2019-04-13:
Another hint: you can edit your messages and even preview them!

> I forgot to ssay this but it stopped working after I updated it ower night. I might have broke it,

Then it looks like you just ran out of memory. That's pretty much nobody's fault then.

machmar, 2019-04-13:
like physical memory in my pc ? (I have 7gb free)

v1993, 2019-04-13:
Yep. If it is fixed now on exactly the same source I can't guess any other reason for compiler to fail with such message.

Whatever, segfaulting compiler is not u8g2's issue anyways…

machmar, 2019-04-13:
gonna try to clean up my ssd than

v1993, 2019-04-13:
Nono! That problem could be caused only by RAM overflow, disk space is not related thing.

I guess you can close this issue because it is irrelevant to u8g2.

machmar, 2019-04-13:
pretty sure it's not ram (16gigs and only Firefox and Spotify are open ~8gigs used) Where should I ask than ?

v1993, 2019-04-13:
I thing that best place would be something like arduino forums or arduino bug tracker. However, you'll need to be able to describe how to reproduce it.

machmar, 2019-04-13:
Restarted my PC and everything is now working.

closed 2019-04-13 by machmar