olikraus/u8g2 #868: Please explain the incrementing syntax - "u8g2.drawBox(10+a,15,30,7);"

dlion88, 2019-04-14:
Hello,
Thanks for the great library and examples, but please explain the use of the incrementing function to animate the box drawing, ie the 10+a entry.
I cannot find an explanation in library reference, or in net search.
Thanks.

olikraus, 2019-04-14:
Not exactly sure what you mean. Without knowing the context a+10 just seems to be an expression which is evaluated at runtime.

dlion88, 2019-04-14:
Hi,
Thanks for your quick response. Here is full context of my query.

The following is taken from U8g2 GraphicsTest.ino example file:

void u8g2_box_frame(uint8_t a) {
  u8g2.drawStr( 0, 0, "drawBox");
  u8g2.drawBox(5,10,20,10);
  u8g2.drawBox(10+a,15,30,7);

Please explain how 10+a moves the box.
The box moves incrementally on the screen. It does not just jump to the new 10+a position.
Is this animation action built in to U8g2?
Thanks again.
Cheers.

olikraus, 2019-04-15:
A increases by 1 with every frame. This is just plain c coding and has nothing to do with u8g2.

dlion88, 2019-04-15:
OK. Thanks for your patience.
Cheers.

closed 2019-04-15 by dlion88