olikraus/u8g2 #869: Unable to fully display ascender/descenders of glyphs

mpc3k, 2019-04-14:
Hi all,

I am using the following [code example](https://robotzero.one/heltec-wifi-kit-8/) on a Heltec WiFi Kit 8 (ESP 8266 with a 0.96inch 128*32 OLED display), using u8g2 with #define U8G2_16BIT enabled, the current is u8g2_font_logisoso32_tf. The descenders of the small letters "y, g" etc. and the upper part of the Asterisk are cut off, when choosing a font size of 32 (which should work on a 32 px high display). Trial and error brought the font size down to 22, all descender are there, but no matter how much smaller I take it, Commas always show up as dots and Asterisks remain cut by half.

This is the code I'm using:
> #include <U8g2lib.h>
> 
> //U8g2 Contructor
> U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 16, /* clock=*/ 5, /* data=*/ 4);
> // Alternative board version. Uncomment if above doesn't work.
> // U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ 4, /* clock=*/ 14, /* data=*/ 2);
> 
> u8g2_uint_t offset;     // current offset for the scrolling text
> u8g2_uint_t width;      // pixel width of the scrolling text (must be lesser than 128 unless U8G2_16BIT is defined
> const char *text = "Gg Yy ,,, *** "; // scroll this text from right to left
> 
> 
> void setup(void) {
>   u8g2.begin();
> 
>   u8g2.setFont(u8g2_font_logisoso32_tf); // set the target font to calculate the pixel width
>   width = u8g2.getUTF8Width(text);    // calculate the pixel width of the text
> 
>   u8g2.setFontMode(0);    // enable transparent mode, which is faster
> }
> 
> 
> void loop(void) {
>   u8g2_uint_t x;
> 
>   u8g2.firstPage();
>   do {
> 
>     // draw the scrolling text at current offset
>     x = offset;
>     u8g2.setFont(u8g2_font_logisoso32_tf);   // set the target font
>     do {                // repeated drawing of the scrolling text...
>       u8g2.drawUTF8(x, 32, text);     // draw the scolling text
>       x += width;           // add the pixel width of the scrolling text
>     } while ( x < u8g2.getDisplayWidth() );   // draw again until the complete display is filled
> 
>     u8g2.setFont(u8g2_font_logisoso32_tf);   // draw the current pixel width
>     u8g2.setCursor(0, 64);
>     u8g2.print(width);          // this value must be lesser than 128 unless U8G2_16BIT is set
> 
>   } while ( u8g2.nextPage() );
> 
>   offset -= 1;            // scroll by one pixel
>   if ( (u8g2_uint_t)offset < (u8g2_uint_t) - width )
>     offset = 0;             // start over again
> }

Would love to get this fixed, because apart from that bug, u2g8 seems to work great!

Thank you!
Pat

olikraus, 2019-04-15:
I have only mobile connection. Please exclude the brief answers.

Ref postponed by default is the baseline of the glyphs, this means, that the comma will always appear as dot for any font size as long as the reference pos is 0,32. Choose a little bit smaller value to fix this. Also have a look to the pics in the reference manual!

mpc3k, 2019-04-20:
Thanks Oli, but which line in the lib or code are you referencing to? I don't have access to the hardware atm … is it
>   u8g2.drawUTF8(x, 32, text);     // draw the scolling text
?

olikraus, 2019-04-20:
Hm.. I think the autocompletion on my smartphone also destroyed the answer partly.
The term baseline comes out of typography and does not refer to a line in a code.
Have a look here:
![https://camo.githubusercontent.com/f2464e3471043c20ac1129134eef16b77283e10e/68747470733a2f2f7261772e6769746875622e636f6d2f466f6e744275726561752f66624f70656e546f6f6c732f6d61737465722f53686f774d6f757365436f6f7264696e617465732f73686f774d6f757365436f6f7264696e617465735f707265766965772e706e67](https://camo.githubusercontent.com/f2464e3471043c20ac1129134eef16b77283e10e/68747470733a2f2f7261772e6769746875622e636f6d2f466f6e744275726561752f66624f70656e546f6f6c732f6d61737465722f53686f774d6f757365436f6f7264696e617465732f73686f774d6f757365436f6f7264696e617465735f707265766965772e706e67)

The baseline goes through the middle of the comma. If you display has a height of 32 pixel, and you place a comma at (0,32) and the reference point of the glyph is the baseline itself, then only the part above baseline will be visible: `u8g2.drawUTF8(x, 32, ",");` 

So, better use something like `u8g2.drawUTF8(x, 29, ",");`

mpc3k, 2019-04-20:
Thanks again :) 
I can follow you as far as your explanation goes … then again, from a Typographer point of view, when setting a specific font size, one would expect to see the full given size from ascender to descender height. 

Additional irritation stems from the display hardware size, which is considerably bigger than the area in use. I can't tell if this is a hardware or software limitation.

olikraus, 2019-04-20:
Logisoso32 means, that the ascent will be 32 pixel. "Again"  will have a total of more than 40 pixel height. You need to select a smaller font for your 128x32 pixel display.

closed 2019-05-05 by olikraus