olikraus/u8g2 #872: u8x8 iconic font not declared in this scope

OscarVanL, 2019-04-16:
Hi, I'm trying to use the icon font 'u8x8_font_open_iconic_embedded_8x_t' from https://github.com/olikraus/u8g2/wiki/fntgrpiconic#open_iconic_embedded_8x

When I set the font and compile, I get 

```
error: 'u8x8_font_open_iconic_embedded_8x_t' was not declared in this scope
```

Here is my code:

```
#include <Arduino.h>
#include <SPI.h>
#include <U8g2lib.h>

U8G2_SSD1327_EA_W128128_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);

void setup() {
    u8g2.begin();
    u8g2.clearBuffer();
    u8g2.setFont(u8x8_font_open_iconic_embedded_8x_t);
    u8g2.drawGlyph(20, 20, 71);
    u8g2.sendBuffer();
}
```

Any idea why this is happening?

OscarVanL, 2019-04-16:
Nevermind, I solved this. I made a typo and wrote u8x8 rather than u8g2. Oops.

olikraus, 2019-04-16:
:-)

closed 2019-04-16 by olikraus