olikraus/u8g2 #873: Is there any way to reduce the sendBuffer processing time?

woodencase01, 2019-04-16:
Hi!

In my current loop, I'm using the Full Frame Buffer with sendBuffer every 750ms. Each call takes around 70ms.

Is there a way to reduce the processing time?
Could I use the firstPage/nextPage every 100ms to get a similar refresh rate with smaller call duration?
If not, could I change anything (such as font type) to reduce the amount of data transferred?

Here is my set-up:
ATMEGA2560
ST7920_128X64
Full Frame Buffer
Software SPI

Thanks!

olikraus, 2019-04-16:
> Is there a way to reduce the processing time?

probably not.

> Could I use the firstPage/nextPage every 100ms to get a similar refresh rate with smaller call duration?

The refresh rate will go down, but the call duration is also reduced.

JamesGKent, 2019-04-24:
@woodencase01 from your first post you say you are using software spi, this possibly is the real reason its very slow, if you are able try to switch to hardware SPI.

woodencase01, 2019-04-24:
@JamesGKent  Thanks for pointing that out! I'll definitely give it a try. The screen I'm using already had a header with the SPI used for an SD card and the screen was wired to digital pins. I'll try to switch a few of them to get the hardware SPI working.

closed 2019-05-05 by olikraus