olikraus/u8g2 #874: u8g2 on ATXmega

Chriserke, 2019-04-18:
Hello,

do you have any experience using this on a ATXmega? 
And if so, could i use the avr 328 example and remap the ports and go from there or is there someting else that needs to be done?

I'm not quite sure as there is no ATXmega in the supported devices.

Kind regards,

olikraus, 2019-04-19:
In general any none-Arduino boards are not supported. But for sure you can use any existing examples and write your own low level driver.

elektronek, 2019-04-24:
Hello Mr. Olikraus,
you have simple example of one HW SPI low level driver ? 

i.e. for AVR what's here?
https://github.com/olikraus/u8glib/blob/master/csrc/u8g_com_atxmega_hw_spi.c

olikraus, 2019-04-24:
Did you check this page: https://github.com/olikraus/u8g2/wiki/Porting-to-new-MCU-platform

closed 2019-05-05 by olikraus