olikraus/u8g2 #875: Why do I implement u8g2 on stm32f1 is not work?

huuchinh02, 2019-04-19:
Hi,

I'm working on stm32f1. I tried to port u8g2 lib but nothing shows in OLED. Can you help me?

My code is below

`uint8_t u8g2_gpio_and_delay_stm32(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr)
{
  switch(msg)
  {
    case U8X8_MSG_GPIO_AND_DELAY_INIT:	// called once during init phase of u8g2/u8x8
      break;							// can be used to setup pins
    case U8X8_MSG_DELAY_NANO:			// delay arg_int * 1 nano second
      break;    
    case U8X8_MSG_DELAY_100NANO:		// delay arg_int * 100 nano seconds
      break;
    case U8X8_MSG_DELAY_10MICRO:		// delay arg_int * 10 micro seconds      
      break;
    case U8X8_MSG_DELAY_MILLI:			// delay arg_int * 1 milli second   
      break;
    default:
      return 0;
  }
  return 1;
}

uint8_t u8x8_byte_4wire_sw_spi_stm32(u8x8_t *u8x8, uint8_t msg, uint8_t arg_int, void *arg_ptr) 
{
  switch(msg) 
  {
    case U8X8_MSG_BYTE_SEND:
      sh1106_bsp_drv->Write((uint8_t *)arg_ptr, arg_int);
      break;
    case U8X8_MSG_BYTE_INIT:
      SH1106_CS_DEASSERT();
      break;
    case U8X8_MSG_BYTE_SET_DC:
      !arg_int ? SH1106_DC_CMD() : SH1106_DC_DATA();
      break;
    case U8X8_MSG_BYTE_START_TRANSFER:
      SH1106_CS_ASSERT();
      break;
    case U8X8_MSG_BYTE_END_TRANSFER: 
      SH1106_CS_DEASSERT();
      break;
    default:
      return 0;
  }  
  return 1;
}`

`/* Set up */
  u8g2_Setup_sh1106_128x64_noname_1(&u8g2_handle, U8G2_R0, 
  u8x8_byte_4wire_sw_spi_stm32, u8g2_gpio_and_delay_stm32);
  u8g2_InitDisplay(&u8g2_handle); // send init sequence to the display, display is in sleep mode after this,
  u8g2_SetPowerSave(&u8g2_handle, 0); // wake up display
  
  u8g2_FirstPage(&u8g2_handle);
	do
	{
		u8g2_SetFont(&u8g2_handle, u8g2_font_crox1cb_tf);
		u8g2_DrawStr(&u8g2_handle, 0, 0, "Hello World!");
	} while (u8g2_NextPage(&u8g2_handle));`

My hardware ensures work fine because I show successful a bitmap using my basic code that set pixels into OLED. I using SPI 4 wire for this application.

Thank you so much

huuchinh02, 2019-04-19:
`ui8tashihasoifhoashoashfoas`

huuchinh02, 2019-04-19:
`ẹlksadjfdsf
fkjsdklfjsd`

huuchinh02, 2019-04-19:
`fdhfjksdhfjksd`
`fsdkfjsd`

huuchinh02, 2019-04-19:
> fwefwefwefwe
fewfwefwefwe

olikraus, 2019-04-19:
You must implement the delay procedures.

huuchinh02, 2019-04-19:
I tried it, but nothing change. I think delay procedures don't affect because my direct code doesn't have any delay between set/clear CS and DC pin. I debugged and view data while your library conversion, all of them is zero that makes nothing to show on OLED

olikraus, 2019-04-19:
> I debugged and view data while your library conversion, all of them is zero that makes nothing to show on OLED

Then maybe your "sh1106_bsp_drv->Write((uint8_t *)arg_ptr, arg_int);" does not work.

huuchinh02, 2019-04-19:
No, it works fine because I using that function to setup and display a bitmap successful. I see your buffer (buf[128]) on debug viewer is zero all elements when run
>u8g2_DrawStr(&u8g2_handle, 0, 0, “Hello world!”);
Where do I mistake?

olikraus, 2019-04-20:
ah, yes, true, the reference point of glyphs is the lower left edge of the glyph. use

```
u8g2_DrawStr(&u8g2_handle, 0, 20, “Hello world!”);
```

olikraus, 2019-04-20:
Created an FAQ entry for this (i think this question came up multiple times recently)

huuchinh02, 2019-04-20:
Hi olikraus,

You right, that is work, but I don't understand what is your reference `y` mean? Is it a y pixel position for starting a string? Could you explain to me? Tks!

huuchinh02, 2019-04-20:
I understood it. (x,y) coordinate is left bottom of a string although the original coordinate (0,0) is the left top of the screen. Why you don't match us to easily use?

olikraus, 2019-04-20:
U8g2 follows same conversation as 99% of all other serious graphics libs.

huuchinh02, 2019-04-22:
Yes, I got it. Many thanks for your kindly support. You can close this topic

olikraus, 2019-04-22:
:)

closed 2019-04-22 by olikraus