olikraus/u8g2 #876: AVD-TM57QV-NW-001-A with ST7511 driver

Dawid-GDA, 2019-04-19:
Hello,

I would like to use u8g2 on the TFT LDC 320x240 monochrome display AVD-TM57QV-NW-001-A. It has a ST7511 driver, which has no list. The manufacturer of this display for communication only used the 8080 parallel protocol (technical note AVD-TM57QV-NW-001-B-spec).
Can I match this driver to an existing one? Do you need to write a completely new ST7511 driver support for this display?

[AVD-TM57QV-NW-001-B-spec.pdf](https://github.com/olikraus/u8g2/files/3098010/AVD-TM57QV-NW-001-B-spec.pdf)
[ST7511_V0_3_20120723 for AV-Display only.pdf](https://github.com/olikraus/u8g2/files/3098013/ST7511_V0_3_20120723.for.AV-Display.only.pdf)

greetings
David

olikraus, 2019-04-19:
This needs to be implemented. Is there any startup command sequence available?

Dawid-GDA, 2019-04-20:
I have Atmega128 programmed by the seller who presents the display.
And it looks like this:

[image]

The seller promised me the code which is uploaded to this Atemga128. The code is written in assembler. Will it help? But because of Easter, I will have it only on Tuesday.
I will also try to describe the Atmeta128 connections with the screen.

olikraus, 2019-04-20:
Yes, assembler code would be ok.

Dawid-GDA, 2019-04-23:
The seller sent me the code. He said that this is responsible for screen initiation.

`
ldi temp,0xAE;  // SWreset
call intel_ins_out2
ldi temp,0xA5
call intel_data_out2

ldi temp,0x61;  // all PWR on
call intel_ins_out2
ldi temp,0x0F call intel_data_out2
ldi temp,0x04 call intel_data_out2
ldi temp,0x02 call intel_data_out2
ldi temp,0xA5 call intel_data_out2

ldi temp,0x62;  // Electronic Volumn Set 1
call intel_ins_out2
ldi temp,0x0A call intel_data_out2 ;  // VCOM[6:0] 0~127 => -0.4250-(0.0125xVCOM) 
ldi temp,0x06 call intel_data_out2 ;  // VGHREG[5:0] 0~63 => 1.5+(0.1x(VGHREG-1))
ldi temp,0x0F call intel_data_out2 ;  // VGLREG[4:0] 0~31 => 2.4+(0.1xVGLREG)
ldi temp,0xA5 call intel_data_out2

ldi temp,0x63;  // Electronic Volumn Set 2
call intel_ins_out2
ldi temp,0x0F call intel_data_out2 ;  // GVDD[4:0] 0~31 => 3.1+(0.1xGVDD)
ldi temp,0x0F call intel_data_out2 ;  // GVCL[4:0] 0~31 => -3.1-(0.1xGVCL)
ldi temp,0xA5 call intel_data_out2 ;  
ldi temp,0xA5 call intel_data_out2

ldi temp,0x66;  // scan mode
call intel_ins_out2
ldi temp,0x00 call intel_data_out2 ;  // 00000  10 revers
ldi temp,0xA5 call intel_data_out2 ; 
ldi temp,0xA5 call intel_data_out2 ; 
ldi temp,0xA5 call intel_data_out2 ; 

ldi temp,0x12;  // SLP OUT
call intel_ins_out2
ldi temp,0xA5 call intel_data_out2 ; 

ldi temp,0x15;  // DISPLAY_ON
call intel_ins_out2
ldi temp,0xA5 call intel_data_out2 ; 

ldi temp,0x22;  // bit per pixel selection
call intel_ins_out2
ldi temp,0x00 call intel_data_out2 ;  // 00 -1  01 -2 10 -4
ldi temp,0xA5 call intel_data_out2 ; 
ldi temp,0xA5 call intel_data_out2 ; 
ldi temp,0xA5 call intel_data_out2 ; 

ldi temp,0x24;  // memory control directions inversions
call intel_ins_out2
ldi temp,0x01 call intel_data_out2 ;  // 00000 MV MY MX
ldi temp,0xA5 call intel_data_out2 ; 
ldi temp,0xA5 call intel_data_out2 ; 
ldi temp,0xA5 call intel_data_out2 ; 

ldi temp,0x26;  // col. addr. setting 
call intel_ins_out2
ldi temp,0x00 call intel_data_out2 ;  // CSA[9:8]
ldi temp,0xA0 call intel_data_out2 ;  // CSA[7:0] 
ldi temp,0x01 call intel_data_out2 ;  // CEA[9:8] 319
ldi temp,0xDF call intel_data_out2 ;  // CEA[7:0] 

ldi temp,0x25;  // page addr. setting
call intel_ins_out2
ldi temp,0x00 call intel_data_out2 ;  // PSA[7:0] 
ldi temp,0x1D call intel_data_out2 ;  // PEA[7:0] Max = 159  9F
ldi temp,0x00 call intel_data_out2 ;  
ldi temp,0xA5 call intel_data_out2 ;  

;ldi temp,0x1B;  // INVERSJA
;call intel_ins_out2
;ldi temp,0xA5 call intel_data_out2 ; 

ldi temp,0x2C;  // write data command 
call intel_ins_out2
ldi temp,0xA5 call intel_data_out2 ;
 `

On my eye, although I do not know assembler, this is a part of what is uploaded to this atmega128. Write or just enough. If not, what else is needed and I will try to settle?
I also wrote the connections between the screen and Atmega128.
Screen -> Atmega128
1. PEL -> PB0 (backlight)
2. RESET -> RESET
3. GND -> GND
4. VDD -> V + (3.3V)
5. WR -> PB4
6. RD -> PB5
7. CS -> PB6
8. RS -> PB7
9. NC -> GND
10. DB0 -> PD0
11. DB1 -> PD1
12. DB2 -> PD2
13. DB3 -> PD3
14. DB4 -> PD4
15. DB5 -> PD5
16. DB6 -> PD6
17. DB7 -> PD7
18. GND -> GND
19. GND -> GND
20. VDD -> V + (3.3V)

olikraus, 2019-04-24:
Constructor:

```
U8G2_ST7511_AVD_320X240_1_8080 u8g2(U8G2_R0, 13, 11, 2, 3, 4, 5, 6, A4, /*enable/WR=*/ 7, /*cs=*/ 10, /*dc=*/ 9, /*reset=*/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V
U8X8_ST7511_AVD_320X240_8080 u8g2(13, 11, 2, 3, 4, 5, 6, A4, /*enable/WR=*/ 7, /*cs=*/ 10, /*dc=*/ 9, /*reset=*/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V
```

I have created beta release 2.26.3:

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

Can you do some testing?
Start with the HelloWorld.ino example.
If this works, please let me know about FlipMode.ino examples.
I am also interessted in pictures of the display (if anything is visible).

Of course you may need to modify the constructor arguments.

Dawid-GDA, 2019-04-25:
I downloaded U8g2 2.26.3. I removed the old version and installed a new one.
I connected the screen to AdruinoUNO through logic converters.

DB0 -> 13
DB1 -> 11
DB2 -> 2
DB3 -> 3
DB4 -> 4
DB5 -> 5
DB6 -> 6
DB7 -> A4
WR -> 7
CS -> 10
RS -> 9
RD -> 3.3V

I chose the example U8g2-> u8x8-> HelloWord. I uncommented the constructor:

> U8X8_ST7511_AVD_320X240_8080 u8g2(13, 11, 2, 3, 4, 5, 6, A4, /*enable/WR=*/ 7, /*cs=*/ 10, /*dc=*/ 9, /*reset=*/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V

In the example, I changed u8x8 to u8g2. In addition to the backlight nothing displays.

```
void setup(void)
{
  /* U8g2 Project: SSD1306 Test Board */
  //pinMode(10, OUTPUT);
  //pinMode(9, OUTPUT);
  //digitalWrite(10, 0);
  //digitalWrite(9, 0);		
  
  /* U8g2 Project: KS0108 Test Board */
  //pinMode(16, OUTPUT);
  //digitalWrite(16, 0);	
  
  u8g2.begin();
  u8g2.setPowerSave(0);
}

void loop(void)
{
  u8g2.setFont(u8x8_font_chroma48medium8_r);
  u8g2.drawString(0,0,"Hello World!");
  u8g2.refreshDisplay();		// only required for SSD1606/7  
  delay(2000);
}
```

[image]

I thought it might be 3.3V logic, so I connected it directly to ArduinoUNO. Unfortunately, nothing has changed.

In the same example, I changed the constructor to:

> U8G2_ST7511_AVD_320X240_1_8080 u8g2(U8G2_R0, 13, 11, 2, 3, 4, 5, 6, A4, /*enable/WR=*/ 7, /*cs=*/ 10, /*dc=*/ 9, /*reset=*/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V

But there were only error:

> HelloWorld:197: error: 'U8G2_ST7511_AVD_320X240_1_8080' does not name a type
> 
>  U8G2_ST7511_AVD_320X240_1_8080 u8g2(U8G2_R0, 13, 11, 2, 3, 4, 5, 6, A4, /*enable/WR=*/ 7, /*cs=*/ 10, /*dc=*/ 9, /*reset=*/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V
>  ^
> 
> C:\Users\Dawid\AppData\Local\Temp\arduino_modified_sketch_513002\HelloWorld.ino: In function 'void setup()':
> 
> HelloWorld:216: error: 'u8g2' was not declared in this scope
> 
>    u8g2.begin();
>    ^
> 
> C:\Users\Dawid\AppData\Local\Temp\arduino_modified_sketch_513002\HelloWorld.ino: In function 'void loop()':
> 
> HelloWorld:224: error: 'u8g2' was not declared in this scope
> 
>    u8g2.setFont(u8x8_font_chroma48medium8_r);
>    ^
> 
> exit status 1
> 'U8G2_ST7511_AVD_320X240_1_8080' does not name a type

I found an example of U8g2 -> full_buffer -> HelloWord.
With the constructor who was there:

> U8G2_ST7511_AVD_320X240_F_8080 u8g2(U8G2_R0, 13, 11, 2, 3, 4, 5, 6, A4, /*enable/WR=*/ 7, /*cs=*/ 10, /*dc=*/ 9, /*reset=*/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V

Memory error:

> Sketch uses 8678 bytes (26%) of program storage space. Maximum is 32256 bytes.
> Global variables use 10100 bytes (493%) of dynamic memory, leaving -8052 bytes for local variables. Maximum is 2048 bytes.

So I changed to:

> U8G2_ST7511_AVD_320X240_1_8080 u8g2(U8G2_R0, 13, 11, 2, 3, 4, 5, 6, A4, /*enable/WR=*/ 7, /*cs=*/ 10, /*dc=*/ 9, /*reset=*/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V

But nothing showed up.

I do not know where I'm making a mistake. Is it my fault to connect the screen to ArduinoUNO. Maybe the example I used is wrong. Is there any screen communication diagnosis with Arduino?

olikraus, 2019-04-25:
> In the example, I changed u8x8 to u8g2. In addition to the backlight nothing displays.

Ups. Will fix this in the next beta.

> I thought it might be 3.3V logic, so I connected it directly to ArduinoUNO. Unfortunately, nothing has changed.

No, the level shifters are required. ArduinoUNO is a 5V device.

> But there were only error:

Strange, maybe the old lib was not removed completely. It should actually work. But you can continue with u8x8 example. If we get this to work, then the rest is not a problem.

> Memory error:

Obviously. You screen is very huge, so a lot RAM is required for the F constructor. Let's continue with u8x8.

> Is it my fault to connect the screen to ArduinoUNO.

Uno should be ok

> Maybe the example I used is wrong.

U8x8 Hello World is perfect.

> Is there any screen communication diagnosis with Arduino?

No :-(

Obviously I might have made software mistakes here, but a solderless protoboard is also a great source of problems because the plug in wires may not have reliable contact. Can you monitor the signals at the display (scope or logic analyzer)? Do all signals reach the display correctly?

Dawid-GDA, 2019-04-25:
I do not have a logic analyzer, but I have measured the voltage meter at the output of Atmega128 and ArduinoUNO.

display -> Atmega128 ->ArduinoUNO
WR -> 4,55 -> 4,8
RD -> 4,6 -> 4,8
CS -> 0 -> 4,8
RS -> 4,55 -> 4,8
DB0 -> 0,07 – 0,25 -> 0 – 0,05
DB1 -> 0,04 – 0,16 -> 0 – 0,05
DB2 -> 0,11 – 0,4 -> 0 – 0,1
DB3 -> 0,15 – 0,5 -> 0 – 0,08
DB4 -> 0,11 – 0,4-> 0 – 0,05
DB5 -> 2,6 – 4,2 -> 0 – 0,17
DB6 -> 0,01 – 0,09 -> 0 – 0,06
DB7 -> 0,04 – 0,12 -> 0 – 0,09

It's a bit weird. I will try ArduninoUNO and ArduinoMEGA with others. Maybe something is wrong with that.

olikraus, 2019-04-25:
I does not make much sense to use a voltage meter. Disconnect all power supply and measure the ohm resistance for all wires from Arduino board to level shifter input and level shifter output to your display. All these wires must have 0 ohm.

Dawid-GDA, 2019-04-26:
I connected the display directly under ArduinoMEGA with the constructor set to:

> U8X8_ST7511_AVD_320X240_8080 u8x8(13, 11, 2, 3, 4, 5, 6, A4, /enable/WR=/ 7, /cs=/ 10, /dc=/ 9, /reset=/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V

And the thread did not display as before.
When I changed the settings in the constructor to:

> U8X8_ST7511_AVD_320X240_8080 u8x8(8, 9, 2, 3, 4, 5, 6, 7, /enable/WR=/ A1, /cs=/ A3, /dc=/ A2, /reset=/ 10); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V

Or

> U8X8_ST7511_AVD_320X240_8080 u8x8(9, 8, 2, 3, 4, 5, 6, 7, /enable/WR=/ 10, /cs=/ 12, /dc=/ 13, /reset=/ 11); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V

This display shows:

[image]

Small success is. Any ideas what could be wrong?

olikraus, 2019-04-26:
You have different pin numbers and in both cases the same output is generated?
Strange.

Well, if I would have the board in my lab, I would start to write at the memory locations of the display to see whether something changes. Somehow it looks like, that the wrong BPP mode is selected, but I exactly followed the instructions from the assembler listing.
Can you test with the u8x8 graphics test? Will you see something else?

Dawid-GDA, 2019-04-26:
> You have different pin numbers and in both cases the same output is generated?

Yes, the same is generated.

Examples GraphicsTest:

https://youtu.be/0TYDsFmAARk

The film shows something shifted.

I talked to the seller and apparently you have to implement the whole initiation code that I gave earlier. Maybe there is a problem here.

olikraus, 2019-04-26:
Wow amazing. This looks already quite good!!!

Can you locate u8x8_d_st7511.c on your local system? Change value in line 93 from 0 to 160
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7511.c#L93

Will graphics test be better? Maybe you need to use other values like 80 or 40, but I assume 160 should be correct.

Dawid-GDA, 2019-04-26:
I changed /* default_x_offset = */ 160,
That's how it works! You do not even know how happy I am.

https://youtu.be/BKn4X09Z9AI

The first part of the movie with backlight. In the second I turned off.
Do you want to improve something.

For sure I will continue testing.
How could I repay you?

olikraus, 2019-04-26:
> For sure I will continue testing.

It would be cool if you could test *u8x8* flip mode. We might need to adjust the flip mode offset also:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7511.c#L94

Then of course *u8g2* flip mode and u8g2 graphics test should also work.

> How could I repay you?

Not required. But if you really want to pay me a beer, my email (see source code) is connected to paypal.

Dawid-GDA, 2019-04-29:
I tested the screen over the weekend and as of today:
The u8x8 flip mode example can not locate the flipmode_x_offset starting point. I changed to different values but you can not see anything.

I tested the examples from page_buffer I used the constructor:

> U8G2_ST7511_AVD_320X240_1_8080 u8g2(U8G2_R0, 9, 8, 2, 3, 4, 5, 6, 7, /*enable/WR=*/ 10, /*cs=*/ 12, /*dc=*/ 13, /*reset=*/ 11); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V

And it looks like this:

Clock.ino https://youtu.be/kIUqCqkCqOw
Weather.ino https://youtu.be/58FbjEiMGeM
GraphicsTest.ino https://youtu.be/1_MCI_sjMcc

HelloWorld.ino 
```
void loop(void) {
  u8g2.setFont(u8g2_font_ncenB14_tr);
  u8g2.firstPage();
  do {
    u8g2.setCursor(0, 20);
    u8g2.print(F("Hello1 World hello11!"));
    u8g2.setCursor(0, 60);
    u8g2.print(F("Hello2 World hello22!"));
    u8g2.setCursor(0, 100);
    u8g2.print(F("Hello3 World hello33!"));
    u8g2.setCursor(0, 140);
    u8g2.print(F("Hello4 World hello44!"));
  } while ( u8g2.nextPage() );
  delay(1000);
} 
```
[image]

olikraus, 2019-04-29:
Did you enable 16 bit mode? See U8g2 FAQ for this.

Dawid-GDA, 2019-04-30:
I checked and in fact was not active 16 Bit mode.
I uncommented #define U8G2_16BIT but nothing changed.
I ran the example HelloWorld.ino but the effect as in the picture above.

Dawid-GDA, 2019-04-30:
Please tell me why the constructor supports 16bit interface? In technical note ST7511 which I have provided is written:

> 8-bit parallel bi-directional interface with 6800 and 8080

Unless I'm wrong.

olikraus, 2019-04-30:
U8G2_16BIT is for the internal calculation of the pixel coordinates. It is mandatory to activate 16 bit mode if the screen dimension is higher than 240 in any direction.

You need to ensure to modify the right .h file (Arduino is a little bit strange here) and also ensure that all files are fully recompiled.

Dawid-GDA, 2019-04-30:
Compiles correctly. I uncommented on the local system:

https://github.com/olikraus/u8g2/blob/master/csrc/u8g2.h#L72

Still the same :(
The only thing that different HelloWorld examples in u8x8 and page_buffer.
This is in U8g2 reference:
```
u8g2.firstPage();
do {
.
.
} while (u8g2.nextPage());
```

olikraus, 2019-05-04:
hmm... it is not so clear to me, what is wrong. So you did uncomment "#define U8G2_16BIT", correct?
It does not make much sense, why "World" is skipped in your example.

Please ensure that the complete u8g2 is recompiled. Also ensure that u8g2 lib is there only once on your system. Sometimes it happens, that a lib is installed twice at different locations.

olikraus, 2019-05-05:
I have updated the x offset to 160, however, i am not sure about the other problems. Any progress?

Dawid-GDA, 2019-05-16:
I test the display but still without any positive solutions.
I do not know what to do to improve it. I do not have as much knowledge as you.
I ran the HelloWorld example.

```
void setup(void) {
  u8g2.begin();  
}

void loop(void) {
  u8g2.firstPage();
  do {
    u8g2.setFont(u8g2_font_logisoso38_tr);
    u8g2.drawStr(0,50,"0123456789x1x2x3x4x5x6x7");
    u8g2.drawStr(0,100,"0123456789y1y2y3y4y5y6y7");
    u8g2.drawStr(0,150,"0123456789z1z2z3z4z5z6z7");
  } while ( u8g2.nextPage() );
  //delay(1000);
}
```

With the constructor.
```
U8G2_ST7511_AVD_320X240_2_8080 u8g2(U8G2_R0, 13, 11, 2, 3, 4, 5, 6, A4, /*enable/WR=*/ 7, /*cs=*/ 10, /*dc=*/ 9, /*reset=*/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V
```

[image]

In the picture I drew the frames one green, the other red. I have the impression that the text displayed in the red frame obscures the text in the green frame. As if there were two buffers.
Which files can I modify to see the results on the screen. Or what should I do to find the error that causes it.

Dawid-GDA, 2019-05-16:
I am modifying the file u8x8_d_st7511.c
I thought the problem was with screen initialization so I started to "add" commands. All of which in the commentary have to add are the ones I added.

> static const uint8_t u8x8_d_st7511_320x240_init_seq[] = {
>     
>   U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */
>   
>   U8X8_CA(0xae, 0xa5),						/* SW Reset */
>   U8X8_CAAAA(0x61, 0x0f, 0x04, 0x02, 0xa5),	/* all power on */
>   U8X8_CAAAA(0x62, 0x0a, 0x06, 0x0f, 0xa5),	/* electronic volumne set 1 */
>   U8X8_CAAAA(0x63, 0x0f, 0x0f, 0xa5, 0xa5),		/* electronic volumne set 2 */
>   U8X8_CAAAA(0x66, 0x00, 0xa5, 0xa5, 0xa5),		/* electronic volumne set 2 */
>   U8X8_CA(0x12, 0xa5),						/* SLeeP OUT */
>   U8X8_DLY(50),
>   // skiping display on here, deviation from https://github.com/olikraus/u8g2/issues/876
>   // will be called later in u8x8_d_st7511_320x240_powersave0_seq
>   U8X8_CA(0x15, 0xa5),                          /* add - display on */
>   U8X8_CAAAA(0x22, 0x00, 0xa5, 0xa5, 0xa5),		/* monochrome display */
>   U8X8_CAAAA(0x24, 0x01, 0xa5, 0xa5, 0xa5),		/* memory control directions */
>   //U8X8_CAAAA(0x26, 0x00, 0xa0, 0x01, 0xdf),     /* add - col. addr. setting */
>   //U8X8_CAAAA(0x25, 0x00, 0x1d, 0x00, 0xa5),     /* add - page addr. setting */
>   //U8X8_CA(0x1b, 0xa5),                          /* add - inversion */
>   U8X8_DLY(50),
>   U8X8_CA(0x12, 0xa5),
>   
>   
>   U8X8_END_TRANSFER(),             	/* disable chip */
>   U8X8_END()             			/* end of sequence */
> };
> 

The operation of one command made me curious. Like after

> U8X8_CA (0x12, 0xa5), / * SLeeP OUT * /
> U8X8_DLY (50)

I added

> U8X8_CA (0x15, 0xa5), / * add - display on * /

This text on the screen first appeared very quickly and disappeared. And then came the second time but slower.

https://youtu.be/fSQWpEvzQIw

Why did this happen?

olikraus, 2019-05-16:
> This text on the screen first appeared very quickly and disappeared. And then came the second time but slower.

The display on is intentionally not part of the init sequence, because U8g2 will erase any old content first.

> first appeared very quickly 

If you activate the screen during init, then you will see the previous content from the last run

> and disappeared

This is, because u8g2 will erase the complete screen.

> And then came the second time but slower.

This is your arduino code in the "loop()" function

I have to look into your other problem. Just to confirm: You did activate 16 bit mode, right?

Dawid-GDA, 2019-05-16:
Thank you for the explanation.

> This is your arduino code in the "loop()" function

Yes this is my function. I chose a different font and added more lines for better illustration.

> I have to look into your other problem. Just to confirm: You did activate 16 bit mode, right?

Yes

I try to get the full code from the seller. Maybe he'll explain something.

olikraus, 2019-05-16:
One more request: Can you test u8x8 examples? Will the segmentation/hiding/skipping also happen in u8x8 examples? Or are u8x8 correct?

Dawid-GDA, 2019-05-17:
U8x8 example:

```
int i = 0;
int a = 0;
int c = 0;

void setup(void)
{
  /* U8g2 Project: SSD1306 Test Board */
  //pinMode(10, OUTPUT);
  //pinMode(9, OUTPUT);
  //digitalWrite(10, 0);
  //digitalWrite(9, 0);		
  
  /* U8g2 Project: KS0108 Test Board */
  //pinMode(16, OUTPUT);
  //digitalWrite(16, 0);	
  
  u8x8.begin();
 // u8x8.setPowerSave(0);
  
}

void loop(void)
{
  i = i + 1;
  a = a + 2;
  c = c + 3;
  
  u8x8.inverse();
  u8x8.setFont(u8x8_font_inb46_4x8_r);
  u8x8.setCursor(3,2);
  u8x8.print(i);
  u8x8.noInverse();
  u8x8.setFont(u8x8_font_inb46_4x8_r);
  u8x8.setCursor(25,2);
  u8x8.print(i);
  u8x8.setFont(u8x8_font_px437wyse700b_2x2_r);
  u8x8.setCursor(3,15);
  u8x8.print(a);
  u8x8.setFont(u8x8_font_7x14B_1x2_r);
  u8x8.setCursor(10,7);
  u8x8.print(c);
  u8x8.inverse();
  u8x8.setFont(u8x8_font_inr33_3x6_r);
  u8x8.drawString(0,20,"Hello World!");
  //u8x8.refreshDisplay();		// only required for SSD1606/7  
  delay(0);
}
```

Constructor:
```
U8X8_ST7511_AVD_320X240_8080 u8x8(13, 11, 2, 3, 4, 5, 6, A4, /*enable/WR=*/ 7, /*cs=*/ 10, /*dc=*/ 9, /*reset=*/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V
```
Movie:

https://youtu.be/s55_D9oKIpw

I do not see segmentation / hiding / skiping. If you want to send me an example, I will test it.
Is it possible to draw lines, squares, pictures in u8x8?
Maybe the problem lies in the conversion of positions. Examples of u8x8 use position_width / title_hight and u8g2 with pixel_width / pixel_hight.

olikraus, 2019-05-17:
> Is it possible to draw lines, squares, pictures in u8x8?

no, this is exactly the difference between u8g2 and u8x8.

> Maybe the problem lies in the conversion of positions.

hmm... i do not think so....

I need to review the code again....

olikraus, 2019-05-17:
Because u8x8 works, i have this idea: Maybe the x position should be resent with each new data transfer. That means the x position needs to be resent just before 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7511.c#L192
and
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7511.c#L196

olikraus, 2019-05-18:
ah, I think i found the bug...

olikraus, 2019-05-18:
I have created a new beta 2.26.7
You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2019-05-18:
I u8g2 now works, can you also test the u8g2 page mode FlipMode example? There must be a frame/border around the display all the time.

Dawid-GDA, 2019-05-20:
I downloaded the new version and installed it.
I did some tests. Here are the results:

FlipMode example with u8g2:

https://youtu.be/aAKBEdt8moI

Example UpdatePartly:

https://youtu.be/IG6KpF8OGWo

My example:

```
#include <Arduino.h>
#include <U8g2lib.h>

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_ST7511_AVD_320X240_1_8080 u8g2(U8G2_R0, 13, 11, 2, 3, 4, 5, 6, A4, /*enable/WR=*/ 7, /*cs=*/ 10, /*dc=*/ 9, /*reset=*/ 8); // Enable U8g2 16Bit Mode and connect RD pin with 3.3V/5V

void setup(void) {
  u8g2.begin();
}

static unsigned char b20_bits[] = {
   0xfc, 0xff, 0xff, 0x3f, 0x04, 0x00, 0x00, 0x20, 0x07, 0x00, 0x00, 0x2f,
   0x01, 0x00, 0x00, 0x2f, 0x01, 0x00, 0x00, 0x2f, 0x01, 0x00, 0x00, 0x2f,
   0x01, 0x00, 0x00, 0x2f, 0x01, 0x00, 0x00, 0x2f, 0x01, 0x00, 0x00, 0x2f,
   0x01, 0x00, 0x00, 0x2f, 0x01, 0x00, 0x00, 0x2f, 0x07, 0x00, 0x00, 0x2f,
   0x04, 0x00, 0x00, 0x20, 0xfc, 0xff, 0xff, 0x3f };

void loop(void) {
  u8g2.firstPage();
  do {
      u8g2.drawXBM(288, 2, 30, 14, b20_bits);
    u8g2.drawLine(0, 18, 319, 18);
    u8g2.drawLine(0, 19, 319, 19);
//    u8g2.drawLine(159, 20, 159, 239);
//    u8g2.drawLine(160, 20, 160, 239);
//    u8g2.drawLine(0, 129, 319, 129);
//    u8g2.drawLine(0, 130, 319, 130);
    u8g2.drawRFrame(1, 21, 158, 108, 10);
    u8g2.drawRFrame(161, 21, 158, 108, 10);
    u8g2.drawRFrame(1, 131, 158, 108, 10);
    u8g2.drawRFrame(161, 131, 158, 108, 10);

  } while ( u8g2.nextPage() );

}

```

[image]

The U8g2Logo example:

[image]

It's nice if the logo was bigger. It would look better.

I still test this library, but so far it looks very good. If you want me to test something, write.

olikraus, 2019-05-20:
Thanks for creating video and pics. I am not 100% sure whether HW Flip works (it looks like it doesn't work) but if you do not require hardware flip, then i assume all is ok.

> It's nice if the logo was bigger. It would look better.

U8g2 does not support scalable fonts. It is build for embedded systems, so as much as possible is precalculated. This applies also to the font. The font size is fixed had has to be defined at runtime. So from that perspective the logo is also fixed.

Dawid-GDA, 2019-05-28:
I'm testing your library all the time. Drawing frames and adding images is great. I need large numbers for the project and there is a little refreshing problem. This is probably due to the large screen size.
I have recorded two films for comparison:
The first example with the u8x8 constructor:

> U8X8_ST7511_AVD_320X240_8080

https://youtu.be/zpBVKST6RA0

The second example with the u8g2 constructor:

> U8G2_ST7511_AVD_320X240_1_8080

https://youtu.be/GEQ6F4niQKk

I really like u8x8 refreshing is much better than in u8g2 but I need pictures and drawing.

I changed the constructor to:

> U8G2_ST7511_AVD_320X240_2_8080

It is much faster but there is this paging effect. Is there any option for refreshing to look like in u8x8?

olikraus, 2019-05-28:
Which board do you use? Maybe `U8G2_ST7511_AVD_320X240_F_8080` also works, if there is sufficient RAM (this will require about 10KB).

> It is much faster but there is this paging effect.

You mean the refresh from top to bottom? It will be lesser with the `_F_` constructor.

You could also optimize the low level timing:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7511.c#L94
and
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st7511.c#L95
Play around with smaller values on your local copy.
Of course the complete 8080 is written on top of the Arduino HAL. It could be MUCH faster, but then it would not be compatible across the Arduino boards.

U8x8 is faster, because it transfers only the required data to the display. No local buffer is needed. Drawback is, that there is no graphics available and also the fonts are limited (only monospace, dimensions have to be multiple of 8)
To some extend reduce the amount of data transfered by u8g2 with UpdatePartly (discussed above). 

If your concern is, that too much time is taken for display refresh and too less remains for your application, have a look here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/StateBufferLoop/StateBufferLoop.ino
Well, this will also not solve the paging problem...

olikraus, 2019-05-28:
Hmm... I have checked u8g2 code. The 8080 mode optimization is only enabled on AVR systems:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L1243

For any none-AVR systems, it falls back to the standard 8080 implementation:
https://github.com/olikraus/u8g2/blob/master/cppsrc/U8x8lib.cpp#L1337

If I would know your board name, I might be able to improve the data transfer performance.

Dawid-GDA, 2019-05-29:
I use the Mega 2560 PRO MINI ATmega2560-16AU 16MHz. I also have ESP-12e and I am wondering if I will run the display on it to use the `_F_` constructor. But it may have too few pins. How would I buy ESP32 or will your library work?
I did a little experiment yesterday. On one example I have included two constructors.

> U8G2_ST7511_AVD_320X240_2_8080
> U8X8_ST7511_AVD_320X240_8080

U8g2 I used in setup () to draw graphics. However, u8x8 in loop () to display numbers. It looks like:

https://youtu.be/BCDCy_ok0I0

I just do not know how it will change the pictures? It may not work.
Change constructor from

> U8G2_ST7511_AVD_320X240_1_8080

on

> U8G2_ST7511_AVD_320X240_2_8080

it speeds up refreshing three times.
I think that this is related to the amount of memory allocated. Maybe you need to increase the amount of memory?

olikraus, 2019-05-29:
> I use the Mega 2560 PRO MINI ATmega2560-16AU 16MHz. 

ok, for this processor, U8g2 is already optimized.

> to use the _F_ constructor. But it may have too few pins. 

Hmm not clear what you mean. The `_F_` constructor (`U8G2_ST7511_AVD_320X240_F_8080`)  will just require more RAM. Just replace the constructor name, then observer whether you get a linker error (RAM Overflow). If not, compare the speed. That's all.

> it speeds up refreshing three times.
I think that this is related to the amount of memory allocated. Maybe you need to increase the amount of memory?

The `_F_` constructor will exactly do this ;-)

olikraus, 2019-06-12:
Can we close this issue?

olikraus, 2019-06-20:
... no feedback... closing...

closed 2019-06-20 by olikraus