olikraus/u8g2 #878: Hardware Scroll Vertically with Double Buffer or Similar?

xgarb, 2019-04-22:
Playing with the new hardware scroll and I've found it's possible to do vertical as well as the horizontal and diagonal.

[image]

It's a bit of a hack because it seems to work by using the diagonal movement in unused pages to push the visible contents of the screen up or down.

I've using a 32px height screen and I would like to use this to move two windows of 32px  up and down into view. I tried experimenting with the 64px constructor but I don't know if I can do 64px in the buffer and move this up and down in the 32px space.

My loop  looks like this:
```
  u8g2.sendF("c", 0x02e);    // disable scroll
  u8g2.clearBuffer();
  u8g2.setFont(u8g2_font_logisoso32_tf);
  u8g2.drawStr(0, 32, "000000");
  u8g2.sendBuffer();
  u8g2.sendF("caaaaaac", 0x029, 0, 4, 0, 7, -1, 255, 0x2f);  // down
  delay(2300);
  u8g2.sendF("caaaaaac", 0x029, 0, 4, 0, 7, 1, 255, 0x2f);  // up
  delay(2300);
```

Robiv8, 2019-04-22:
'class U8G2_SSD1305_128X32_NONAME_F_4W_HW_SPI' has no member named 'sendF'
or
'class U8G2_SSD1306_128X32_UNIVISION_F_HW_I2C' has no member named 'sendF'
hmm..?

olikraus, 2019-04-22:
sendF is very new. I assume it is there only in the latest beta release.
Example is here: https://github.com/olikraus/u8g2/blob/master/sys/arduino/u8g2_page_buffer/HardwareScroll/HardwareScroll.ino

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

olikraus, 2019-04-22:
"SendF" command was introduced here: https://github.com/olikraus/u8g2/issues/822

xgarb, 2019-04-23:
I just made a copy of the src directory from the Arduino folder from this repository, added it to the folder with the ino and included it like this:

```
//#include <U8g2lib.h>
#include "src/U8g2lib.h"
```

xgarb, 2019-04-23:
Can I move two 'screens' up and down?

> I've using a 32px height screen and I would like to use this to move two windows of 32px up and down into view. I tried experimenting with the 64px constructor but I don't know if I can do 64px in the buffer and move this up and down in the 32px space.

olikraus, 2019-04-23:
> Can I move two 'screens' up and down?

The commands are documented in the datasheet of the display.

closed 2019-05-05 by olikraus