olikraus/u8g2 #879: SSD1327 wrong size

arcbtc, 2019-04-24:
U8G2_SSD1327_EA_W128128_F_HW_I2C seems to be 128/96 not 128/128?
Tried multiple boards/screens/projects

I tried backdating U8g2 and noticed its a relatively new constructor

olikraus, 2019-04-24:
What exactly do you mean?

This constructor is for this display: https://www.lcd-module.com/fileadmin/eng/pdf/grafik/W128128-XR.pdf

Do you mean, that only 96 out of 128 rows are writable?

arcbtc, 2019-04-24:
This one seemed to answer my question, sorry about that
https://github.com/olikraus/u8g2/issues/783
I think the confusion was not finding the right constructor in the examples, which is where I usually get them from.

olikraus, 2019-04-24:
well, yes, the examples do not contain all the constructors..., so we can close this issue?

arcbtc, 2019-04-24:
Yes, thanks

closed 2019-04-24 by olikraus