olikraus/u8g2 #880: Waveshare SSD1327 128x128 renders as 128x96

nikuz, 2019-04-27:
Hello, 

OLED module:
https://www.waveshare.com/w/upload/8/80/1.5inch_OLED_Module_User_Manual_EN.pdf

I have the same problem as described in #783 but U8G2_SSD1327_MIDAS_128X128_1_HW_I2C constructor doesn't help me. Actually it looks even worse, it renders only few lines at the top of the screen, about 10 lines.

Please help.

olikraus, 2019-04-28:
I think the init sequence needs to be adjusted.
You could change the second value here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1327.c#L349

and here:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_ssd1327.c#L352

Will this improve something?

nikuz, 2019-04-28:
@olikraus I'm don't really know what I can write there. Can you suggest something? Thank you.

olikraus, 2019-04-28:
Just to be clear: You need to located u8x8_d_ssd1327.c on you local file system.
Then goto line 349 and modify the A1 command, for example like this:
U8X8_CA(0x0a1, **0x010**)
Changed value in **bold**. This will shift the vertical offset.
You can try other values like
U8X8_CA(0x0a1, **0x020**)
up to a value of
U8X8_CA(0x0a1, **0x07f**)

Recompile and check whether there is an improvement.

olikraus, 2019-04-28:
Ok, In found the init sequence from waveshere (https://www.waveshare.com/wiki/File:1.5inch_OLED_Moudle.7z):
```
static void OLED_InitReg(void)
{
  OLED_WriteReg(0xae);//--turn off oled panel

  OLED_WriteReg(0x15);    //set column address
  OLED_WriteReg(0x00);    //start column   0
  OLED_WriteReg(0x7f);    //end column   127

  OLED_WriteReg(0x75);    //set row address
  OLED_WriteReg(0x00);    //start row   0
  OLED_WriteReg(0x7f);    //end row   127

  OLED_WriteReg(0x81);    //set contrast control
  OLED_WriteReg(0x80);

  OLED_WriteReg(0xa0);    //gment remap
  OLED_WriteReg(0x51);    //51

  OLED_WriteReg(0xa1);    //start line
  OLED_WriteReg(0x00);

  OLED_WriteReg(0xa2);    //display offset
  OLED_WriteReg(0x00);

  OLED_WriteReg(0xa4);    //rmal display
  OLED_WriteReg(0xa8);    //set multiplex ratio
  OLED_WriteReg(0x7f);

  OLED_WriteReg(0xb1);    //set phase leghth
  OLED_WriteReg(0xf1);

  OLED_WriteReg(0xb3);    //set dclk
  OLED_WriteReg(0x00);    //80Hz:0xc1 90Hz:0xe1   100Hz:0x00   110Hz:0x30 120Hz:0x50   130Hz:0x70     01

  OLED_WriteReg(0xab);    //
  OLED_WriteReg(0x01);    //

  OLED_WriteReg(0xb6);    //set phase leghth
  OLED_WriteReg(0x0f);

  OLED_WriteReg(0xbe);
  OLED_WriteReg(0x0f);

  OLED_WriteReg(0xbc);
  OLED_WriteReg(0x08);

  OLED_WriteReg(0xd5);
  OLED_WriteReg(0x62);

  OLED_WriteReg(0xfd);
  OLED_WriteReg(0x12);

}

```

hmmm... there are several differences.

nikuz, 2019-04-28:
Ok, I'll play with this values. Thank you!

nikuz, 2019-04-30:
This constructor works well:
```C
static const uint8_t u8x8_d_ssd1327_ea_w128128_init_seq[] = {

    U8X8_START_TRANSFER(),             	/* enable chip, delay is part of the transfer start */

    U8X8_C(0x0ae), //--turn off oled panel

    U8X8_CAA(0x015, 0x000, 0x07f),    //set column address, start column 0, end column 127

    U8X8_CAA(0x075, 0x000, 0x07f),    //set row address, start row 0, end row 127

    U8X8_CA(0x081, 0x080),    //set contrast control

    U8X8_CA(0x0a0, 0x051),    //gment remap, 51

    U8X8_CA(0x0a1, 0x000),    //start line

    U8X8_CA(0x0a2, 0x000),    //display offset

    U8X8_CAA(0x0a4, 0x0a8, 0x07f),    //rmal display, set multiplex ratio

    U8X8_CA(0x0b1, 0x0f1),    //set phase leghth

    U8X8_CA(0x0b3, 0x000),    //set dclk, 80Hz:0xc1 90Hz:0xe1   100Hz:0x00   110Hz:0x30 120Hz:0x50   130Hz:0x70     01

    U8X8_CA(0x0ab, 0x001),    //

    U8X8_CA(0x0b6, 0x00f),    //set phase leghth

    U8X8_CA(0x0be, 0x00f),

    U8X8_CA(0x0bc, 0x008),

    U8X8_CA(0x0d5, 0x062),

    U8X8_CA(0x0fd, 0x012),

    U8X8_END_TRANSFER(),             	/* disable chip */
    U8X8_END()             			/* end of sequence */
};
```

@olikraus please add to the library.

olikraus, 2019-05-05:
I have created new constructors starting with "U8G2_SSD1327_WS_128X128".
There is also a new beta release 2.26.5. Can you test the same? Will it work?

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

nikuz, 2019-05-06:
@olikraus I'll test it when I'll get the new screen module. I just broke my own :( during mounting.

olikraus, 2019-05-06:
Oh...

nikuz, 2019-05-16:
@olikraus did you implement only SPI interface for this constructor?

olikraus, 2019-05-16:
No, all possible interfaces are there. See the Wiki page for this.

nikuz, 2019-05-17:
@olikraus I can't find the I2C interface for U8G2_SSD1327_WS_128X128 constructor in wiki as well in source code.

olikraus, 2019-05-19:
Ups, I missed this. I have created beta 2.26.8

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

nikuz, 2019-06-02:
@olikraus thank you. U8G2_SSD1327_WS_128X128_F_SW_I2C works well for this OLED module
https://www.waveshare.com/1.5inch-OLED-Module.htm

closed 2019-06-02 by nikuz

olikraus, 2019-06-02:
Maybe consider HW_I2C for better performance.