olikraus/u8g2 #882: Reducing RAM on AVR using PROGMEM

espHorst, 2019-05-01:
Dear olikraus,
thanks at first for this wonderful library! I would like to ask if it is 
possible to further reduce the RAM usage on AVR processors.
On AVR
```
static const uint8_t u8x8_d_max7219_init_seq[] = {
```
goes to ram. If you agree I can try doing a PR including
```
#ifdef __AVR__
    #include <avr/pgmspace.h>
#endif

#ifdef __AVR__
static const uint8_t PROGMEM u8x8_d_max7219_init_seq[] = { 
#else
static const uint8_t sbox[256] = {
#endif
```
to put static const to PROGMEM. Do you think that there are any issues?
Best regards, espHorst

olikraus, 2019-05-01:
Well, yes, I know, this is a missing thing... However, there are several things to consider.

 * pgmspace.h is already included: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L116
 * There are already several defines, which will help to use progmem area and which should be reused: https://github.com/olikraus/u8g2/blob/master/csrc/u8x8.h#L152-L185
 * We need to update https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_cad.c#L159 and use the progmem access macro
 * We need to modify ALL sequences, which are passed to u8x8_cad_SendSequence. If I use `fgrep -c u8x8_cad_SendSequence *.c | fgrep -v ":0" | cut -d ":" -f 2 | xargs | tr ' ' + | bc` it will give me 451 calls. This means we need to reliable update 451 arrays.

espHorst, 2019-05-01:
Hm. What about adding an additional dedicated `u8x8_cad_SendSequenceP()` with P for progmem. Then piece by piece memory critical sequences can be changed. I would try to start with u8x8_d_max7219_init_seq ;-)

olikraus, 2019-05-01:
The goal is to keep the code portable for all platforms. This means, u8x8_d_max7219_init_seq  should be usable on AVR but also on ARM and ESP. This means, adding a function for u8x8_d_max7219_init_seq only will not help. We need to do this for all 451 arrays.
I mean, I need a solution, which will work for all 451 arrays on all plattforms.

You can definitly do a local solution for your own project for one specific array. But then this is not something what I will takeover here.

espHorst, 2019-05-01:
Hi olikraus, thanks for the feedback - I will do a local solution for just this project and use a bigger uC (STM32F1) for the next project! Thanks a lot!

closed 2019-05-01 by espHorst

olikraus, 2019-05-01:
Sure, I am sorry for this, but this task indeed is a bigger topic.