olikraus/u8g2 #883: Saving SRAM by not including TWI if not required

espHorst, 2019-05-01:
Dear olikraus,
I hope you don't mind that I have opened a second issue on memory saving. My project is really low on memory and I don't wan't to do a new hardware ...
I'm using MAX7219 via SPI interface on an Arduino Mini Pro. Just by including "Wire.h" in the U8x8lib.h header file
```
/* Assumption: All Arduino Boards have "Wire.h" */
#define U8X8_HAVE_HW_I2C
```
about 7% of SRAM are used by the Wire (twi). Is there a chance not to include Wire.h ?
Maybe by putting the SPI initializers and the I2C Initializers in two files? For legacy reasons of course both can be included in the current U8x8lib.h file, in order to save memory one could include U8x8lib_SPI.h and U8x8lib_I2C, respectively.
What is your opinion on this?
Best regards,
espHorst.

olikraus, 2019-05-01:
This is a bug in the Arduino IDE Wire lib. I think I documented this already somewhere. 
Workaround is to comment out the above define and remove the inclusion of Wire.h in your project.

olikraus, 2019-05-01:
Ah, yes I remember: https://github.com/olikraus/u8g2/blob/master/doc/faq.txt#L171
and here: https://github.com/olikraus/u8g2/issues/369

espHorst, 2019-05-01:
Thanks a lot for the fast response!

closed 2019-05-01 by espHorst

olikraus, 2019-05-01:
:)