olikraus/u8g2 #884: UC1611S and CG16060D

witc, 2019-05-02:
Dear,

Do you know if the lib works with display CG160160D  containing driver us1611s?

Thank you.

Best Regards, Jan.

olikraus, 2019-05-02:
There is support for the UC1611 but not for your specific display dimensions: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#uc1611-ea_dogm240

However supporting your display dimensions should not be a big problem if you are willing to do some tests. We also need the datasheet for your display.

witc, 2019-05-02:
Thank you for response!

There is datasheet for display: http://www.cloverdisplay.com/pdf/CG160160D.pdf I would like to use parallel bus for this project, ideally 4-bit but I can see you support only 8 bit.. nevermind.

witc, 2019-05-02:
> There is support for the UC1611 but not for your specific display dimensions: https://github.com/olikraus/u8g2/wiki/u8g2setupcpp#uc1611-ea_dogm240
>
> However supporting your display dimensions should [...]

Sure I will do the tests...

olikraus, 2019-05-03:
todo: Update Init Sequence

olikraus, 2019-05-05:
I have created a new beta release 2.26.4. Can you test the new constructor?

The constructor should look like this (see also the examples)
```
U8G2_UC1611_CG160160_1_8080 u8g2(U8G2_R0, 8, 9, 10, 11, 4, 5, 6, 7,  /*enable=*/ 18, /*cs=*/ 3, /*dc=*/ 16, /*reset=*/ 17); // Connect RD/WR1 and CD1 pin with 3.3V, connect CS0 with cs
```
CS1 of the display must be connected to 3.3V permanently.
CS0 must be connected to your board and the pin number must be provided to the constructor
WR1 must be connected to 3.3V
WR0 must be connected to your board and the pin number must be provided as "enable" argument.
CD must be connected to your board and the pin number must be provided as "dc" argument.

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

witc, 2019-05-06:
Thank you really much. I will test it as soon as the my PCB will be ready. (I do not use arduino, I have my own designed board with STM MCU).

Best Regards Jan.

witc, 2019-05-17:
PCB is done so I tried it by this way:
` /* permanently to VCC*/
  HAL_GPIO_WritePin(DISP_CS1_GPIO_Port,DISP_CS1_Pin,true);
  HAL_GPIO_WritePin(DISP_PORT_WRITE_GPIO_Port,DISP_PORT_WRITE_Pin,true);
  osDelay(10);

    /* Save full display to RAM 160x160 */
    u8g2_Setup_uc1611_cg160160_f(&u8g2, U8G2_R2, u8x8_byte_8bit_8080mode, 
    u8x8_stm32L152_gpio_and_delay);
    u8g2_InitDisplay(&u8g2);
    osDelay(100);
    u8g2_SetPowerSave(&u8g2, 0);
    u8g2_SetContrast(&u8g2, 140);
    u8g2_ClearDisplay(&u8g2);
    u8g2_SetFont(&u8g2, u8g2_font_ncenR18_te);
    u8g2_ClearBuffer(&u8g2);

    u8g2_DrawStr(&u8g2, 10, 80, "Hello World");
    u8g2_SendBuffer(&u8g2);`

I had to add some delay as you can see, but the text is displayed on the LCD - it seems it works properly :-) 
Thanks!!

olikraus, 2019-05-17:
Great... so I will close this issue. Feel free to reopen...

closed 2019-05-17 by olikraus