olikraus/u8g2 #887: u8g2 scrolling  Temperature Graph

mariusmym, 2019-05-07:
Hi Oliver, hi guys !
So, i'm trying to make a temperature graph using this awesome library.
I want to mention from the start that i'm just an hobbyist when it comes to arduino / coding / programming stuff.
The code below works very well until enters the if statement, I put a lot of "Serial.print" in order to try to debug the code, without success.
Can you guys give me a solution ?
The library for the BME280 sensor is here : https://github.com/Seeed-Studio/Grove_BME280

the code (i put a lot of comments, i hope it won't bother you very much): 
`#include <Arduino.h>
#include <U8g2lib.h>
#include "Seeed_BME280.h"

#ifdef U8X8_HAVE_HW_SPI
#include <SPI.h>
#endif
#ifdef U8X8_HAVE_HW_I2C
#include <Wire.h>
#endif

U8G2_SH1106_128X64_NONAME_F_HW_I2C u8g2(U8G2_R0, /* reset=*/ U8X8_PIN_NONE);
BME280 bme280;

// ------ global variables -----

int16_t maxTemperature = 30;
int16_t arrLoad = 0;
int16_t graphOriginY = 60;
int16_t graphOriginX = 10;

int16_t displayHeight = u8g2.getDisplayHeight();
int16_t displayWidth = u8g2.getDisplayWidth();
const int16_t graphLength = displayWidth - graphOriginX;
const int16_t graphHeight = displayHeight - (displayHeight - graphOriginY);
int16_t arrCapacity = graphLength;
uint8_t valuesArr[118]; //118 (128 - 10)
int16_t graphPosX = graphOriginX;
int16_t mappedValue;

void setup() {
  Serial.begin(9600);
  u8g2.begin();
  u8g2.setContrast(255); // 0 = lowest ; 255 = highest
  bme280.init();
  
  // ---------- draw XoY axis ----------

  u8g2.drawLine(graphOriginX, 0 ,graphOriginX, graphHeight ); // verical line -> Y axis
  u8g2.drawHLine(graphOriginX, graphOriginY, graphLength); // horizontal line -> X axis
  u8g2.sendBuffer();
  Serial.print("Graph height : ");
  Serial.println(graphHeight);
  Serial.print("Graph length : ");
  Serial.println(graphLength);
  Serial.println("----------");

}

void loop() {
  uint8_t temperature = (uint8_t)bme280.getTemperature(); //dht.getTemperature(); // for dht22 sensor

    // --- check if array/graph limit is reached in order to delete the first value from the array ---
    if (arrLoad == arrCapacity){
      graphPosX = graphOriginX;
      // redraw the screen
    u8g2.clearBuffer();
    u8g2.drawLine(graphOriginX, 0 ,graphOriginX, graphHeight ); // draw verical line -> Y axis
    u8g2.drawHLine(graphOriginX, graphOriginY, graphLength); // draw horizontal line -> X axis
    u8g2.sendBuffer();
    // delete the first entry of the array
    for (int i = 0; i < arrCapacity - 1; i++){
      Serial.print("."); // check if the code reach this step
      valuesArr[i] = valuesArr[i+1]; 
    }
    for (int i = 0; i < arrCapacity - 1; i++){
      mappedValue = mapToGraph(valuesArr[i]); //map values from the array
      //u8g2.drawPixel(i, mappedValue); // draw values
      Serial.print(">"); // to check if the code reach this step
      u8g2.drawLine(graphPosX, mappedValue, graphPosX, graphOriginY); // draw a verical line instead of point
      graphPosX ++ ;
      u8g2.sendBuffer();
    }
    delay(2000);
    valuesArr[arrLoad] = temperature ; // asign last read temp. value at the end of the array
    mappedValue = mapToGraph(temperature);
    //u8g2.drawPixel(arrLoad, mappedValue); // draw last values 
    u8g2.drawLine(arrLoad, mappedValue, arrLoad, graphOriginY); // draw a verical line instead of point
    u8g2.sendBuffer();
    arrLoad -- ; // update array index (load) because it can't exceed graphLength
    Serial.print("arr load (if) : ");
    Serial.println(arrLoad);
    }

    // if array/graph capacity is not reached, print next temp. value
    else {
      valuesArr[arrLoad] = temperature;
      mappedValue = mapToGraph(temperature);
      //u8g2.drawPixel(graphPosX, mappedValue);
      u8g2.drawLine(graphPosX, mappedValue, graphPosX, graphOriginY); // draw a verical line instead of point
      u8g2.sendBuffer();
    }
  arrLoad ++ ;
  graphPosX++;
  Serial.print("arr load : ");
  Serial.println(arrLoad);
  Serial.print("graph Pos X : ");
  Serial.println(graphPosX);
  Serial.println("----------");
  delay(2000);
  
}

/* because the starting point (0,0) is in the upper left corner of the OLED sceen, 
maping is reversed (when the temp goes up the representation on the graph goes down) 
so we had to fix that by writting another function for mapping values */

int mapToGraph(float value){
   int16_t mapValue = ((value - 0) * (graphHeight - 0) / (maxTemperature - 0) + 0);  // (value - in_min) * (out_max - out_min) / (in_max - in_min) + out_min
   return graphHeight-mapValue ;
} `

olikraus, 2019-05-07:
In general you should use "sendBuffer()" only once. Anything else will only slowdown the graphics output (and you will complain about the low speed of u8g2).

But in general my suggestion is to restructure your code. Use two parts at least:
1. One part will fill and update the array
2. Output the array (and other information) to the display.

To simplify the code you could init the array with all "0" in setup().
Then always shift the array content to te left and add a new temperature value at the array end.
This will  avoid the if - else check in your code.

mariusmym, 2019-05-08:
Thank you for your quick response!
The "else" part works well, and olso the "if" one until some point, after that I receive a "soft reset" message on the Serial monitor 
 (don't know why), and the code start to act weird. 
I will try to follow your advices and see what happens :)!
Thank you for your help!

closed 2019-05-19 by olikraus