olikraus/u8g2 #888: Space out characters that aren't in the font ?

Randname666, 2019-05-08:
When `drawUTF8()` and  `print()` run into a character that's not in the font, seems it would simply skip displaying that character. Is it possible to change its behavior when encountering such a character, like displaying a "□", "�" or space it out ?

Usage environment:
Maybe u8g2 is not designed for such an intense usage, but here is going to build a device that should be able to display 2,500 Chinese characters with an Arduino Mega2560. Due to the limitation of the AVR compiler (max array size 32768), those 2,500 characters are separated into 5 fonts, each one contain 500. So the idea is, by doing such a thing:
```
void ChinesePrint(String Chinese,int x,int y){
  u8g2.setFont(u8g2_CN1_12pt);
  u8g2.setFontDirection(0);
  u8g2.setCursor(x,y);
  u8g2.print(Chinese);
  u8g2.setFont(u8g2_CN2_12pt);
  u8g2.setFontDirection(0);
  u8g2.setCursor(x,y);
  u8g2.print(Chinese);
  u8g2.setFont(u8g2_CN3_12pt);
  u8g2.setFontDirection(0);
  u8g2.setCursor(x,y);
  u8g2.print(Chinese);
  u8g2.sendBuffer();
  u8g2.clearBuffer();
}
```
If a character is not in font1, it would space out and when being printed in font2, it may show up in that space. If still not, there is font3, etc. So all characters can be displayed by constantly swapping the font. However, u8g2 doesn't work like that and it doesn't space out, so all the characters printed are piled up together.

olikraus, 2019-05-08:
I can add space if required. But the space will be of constant size (maybe a #define value).

> If a character is not in font1, it would space out and when being printed in font2, it may show up in that space. If still not, there is font3, etc. 

This does not make sense from my perspective. You should ensure that all required glyphs are in font1 and avoid font2 and font3.

It is not so complicated to create custom fonts with exactly the gylphs you need. I can also do this for you (or show you the required commands for this)

Randname666, 2019-05-08:
> You should ensure that all required glyphs are in font1 and avoid font2 and font3.

Actually Wenquanyi 12pt GB2312a included in u8g2 is already enough for me. 
Finding or creating a font is not a problem, the problem is putting it in ATMega2560: according to https://arduino.stackexchange.com/questions/24458/how-is-a-41kb-char-array-filling-up-256kb-of-program-memory/24465, the maximum array size is 32767, if larger than this compiler would complain array too large and quit (And it did). A font containing all glyphs required is way exceeding this limitation, so some tricks need to be done to put it into Mega2560.
Or maybe it's really not the way to deal will such large amount of glyphs by such a hardware.

olikraus, 2019-05-09:
ah, yes, that large array problem could be an issue. In U8g2 there is a macro which signals, whether big arrays (fonts) can be used or not:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2.h#L150

What about this: You create your own glyph drawing procedure, which select different fonts, based on the encoding of the glyph. 

```
my_draw_glyp(u8g2, x, y, encoding)
{
if ( encoding < 0x01000) {
  u8g2_SetFont(u8g2, font1)
  u8g2_DrawGlyph(u8g2, x,y,encoding)
} 
else if ( encoding < 0x02000) {
  u8g2_SetFont(u8g2, font2)
  u8g2_DrawGlyph(u8g2, x,y,encoding)
} 
...
```

You can then copy the draw string function from u8g2 and call your own glyph drawing function:
https://github.com/olikraus/u8g2/blob/master/csrc/u8g2_font.c#L758

The idea of drawing the same string multiple time with different font also seems to be a good idea under this font size restriction, but for better performance you should also have new fonts for this. 
The other limitation is: All glyphs must have the same width. But I assume, this is true for Chinese fonts.

Randname666, 2019-05-09:
Good idea, thanks. 
The only tiny issue for doing so is my fonts are separated by glyphs' statistic usage frequency rather than encoding. But rearranging them is nothing difficult.

closed 2019-05-09 by Randname666

olikraus, 2019-05-10:
There is also a u8g2 FN which tests whether a glyph is there or not. Wenn this fails, test the next font