olikraus/u8g2 #889: request for 64x32 LCD-Switch

thilda-ma, 2019-05-08:
Hello,
i would love to use the u8g2 lib with my new LCD-Switches.

But i think they are not in support right now.
The switch is called Q5.
Manual can be found here:
[http://www.lcd-solution.com/fiches_techniques/DataSheet-Q5-V0-9%20RdR01.pdf](url)
The manual says mode '0x49' is best for text...

Is it possible to add this one to the library?

Thanks in advance
thilda

olikraus, 2019-05-08:
I try to get one of these LCDs. And yes, should be possible to add this to u8g2.

thilda-ma, 2019-05-08:
I have 5 of them at my Place. Maybe i can lend you one, so you don't need to buy it.

olikraus, 2019-05-09:
I already placed a bid on ebay. But even without the device itself, I should be able to add support. I just need to find some time to implement this.

olikraus, 2019-05-18:
The display reached my lab...
Arduino Example Sources: http://bplagens.bplaced.net/Q5/

olikraus, 2019-11-23:
made a mistake during soldering... will put this on hold..

sephalon, 2020-05-05:
Could you please try if the driver for the [E³] SB6432 from pull request #1180 works for you as well? The protocol of these two LCD keys seems to be compatible.

olikraus, 2020-05-10:
Ok, I reopen this. I need to find some time for further testing of this device.