olikraus/u8g2 #890: Consider new font tools / format for bitmap fonts

puzrin, 2019-05-09:
Hi!

Current sutiation with embedded fonts is:

- There are no unified format, good enougth for embedded needs.
- Each author tends to make own set of tools, and users need additional efforts to use those.
- As a result, projects include a lot of custom files for different sizes, and may still need subsetting (for example "big font to display float numbers only").

IMO it would be more convenient to have "simple" font utility and links collection to popular TTF fonts.

https://github.com/littlevgl/lv_font_conv - may this be interesting for you? Convertor & spec were initially developped for lvgl needs, but don't depend on lvgl anyhow.

If we speak about core differences with your spec:

- Layout is more close to OpenType. May be 50 bytes more, compared to your one, but much more featured and extendable.
- No RLE for 1-bit pixels. I inspected your one, it's not very effective, and not extendible to multi-bit pixels. So, decided to ignore, but can incorporate it as special case if you wish.
- Tools are easy to install and use. You can leave font deals to users, instead of provide tons of variants in your repo.

Let me know what do you think. If you think it's not useful for u8g2 - feel free to close this issue.

AnHardt, 2019-05-09:
In my experience almost all automatically concerted, from TTF to small height, fonts are looking bad or are completely unreadable. The smaller the fonts height the worse. And usually i need a small font to fit the contend to the tiny screens i use.

puzrin, 2019-05-09:
@AnHardt
 
1. Tiny bitonal fonts are correct case, but far from 100% of all possible.
2. It's still possible find pixel-perfect fonts of fixed size in TTF format. For example, [unscii](http://pelulamu.net/unscii/)
3. It's possible to add BDF input support. It not exists now because nobody asked. But that's not font format limitation.

In short: your case is not unsolvable technical problem.

olikraus, 2019-05-09:
> There are no unified format, good enougth for embedded needs. 

True, but U8g2 offers a full chain from ttf to the internal format

>  Each author tends to make own set of tools, and users need additional efforts to use those.

Because it was needed. When I started dogmlib which then became u8glib which is now u8g2, there was nothing.
Additionally, u8g2 comes with many many prebuild fonts with different sets. So my hope is, the average use can just select a font.

When looking at your code I do not see a decoder written in C. How can it be used in embedded systems?

olikraus, 2019-05-09:
I spent some more time to understand your project, but I do not see how to add this to U8g2.
 * Javascript does not make sense for Arduino embedded systems
 * Documentation almost not existing
 * Although I did look a lot, I did not found a proper decoding software
 * Your format seems to be feature rich, however it also seems to be very huge. Will it be as space efficient as u8g2?
 
> May be 50 bytes more, compared to your one

Great, if it would be only 50 bytes. To me it looks, that it might be much more than 50 bytes. I am very much open to compare real data here.

For example let us use http://www.fontspace.com/total-fontgeek-dtf-ltd/glasstown-nbp
If I use all chars of glasstown, it will occupy 12218 bytes in the u8g2 font format. 
See here: https://github.com/olikraus/u8g2/wiki/fntgrpnbp

The otf version of glasstown is 49276 bytes in size, which is four times more than the u8g2 format. So your font format will be around 50K while the same font in the u8g2 format will be 12KB. This is a huge difference for an embedded system.

So as a conclusion: Nice project, good idea, but there is no C or C++ decoder and the font format seems to be four times bigger.

puzrin, 2019-05-09:
Images dump, to prove params are correct (images are pixel-perfect):

`./lv_font_conv.js --font glasstown_nbp.ttf -r 0x20-0x10FFFF --size 9 --bpp 1 --format dump -o 1111`

Seems it's 9px font.

Binary dump:

```
env DEBUG=* ./lv_font_conv.js --font glasstown_nbp.ttf -r 0x20-0x10FFFF --size 9 --bpp 1 --format bin -o test.bin
  font last_id: 878 +0ms
  font minY: -2 +2ms
  font maxY: 8 +1ms
  font glyphIdFormat: 1 +0ms
  font kerningScale: 1 +1ms
  font advanceWidthFormat: 0 +0ms
  font xy_bits: 4 +1ms
  font wh_bits: 4 +1ms
  font advanceWidthBits: 4 +0ms
  font monospaced: false +0ms
  font indexToLocFormat: 0 +16ms
  font.table.head table size = 44 +0ms
  font.table.cmap 8 subtable(s): 7 "format 0", 1 "sparse" +0ms
  font.table.cmap table size = 1116 +6ms
  font.table.loca table size = 1768 +0ms
  font.table.glyph table size = 4660 +0ms
  font font size: 7588 +31ms
```

Binary file size 7588.

Seems spec is not so bad :)

puzrin, 2019-05-09:
> When looking at your code I do not see a decoder written in C. How can it be used in embedded systems?

Let me explain. Full process consists of multiple steps:

- Install tools.
- Encode fonts.
- Draw fonts.

`lv_font_conv` cares about first two, and leave decoding to libs authors. I think, as author of u8g2 font spec, you know, writing good spec is non trivial task and requires a lot of experimenting.

Also, now `node.js` is available everywhere. Some people may find more easy to install package via npm, instead of build c sources. This also simplifies development. I don't insist on anything, only explain possibilities.

You may find useful this things:

- standalone font spec (and write your own tools)
  - may be useful if you decide add kerning support
- spec + encoder (and write decoder only)
- read it just for fun instead of book prior to sleep :)

Consider this project as some interesting information, and feel free use it as you wish. If you don't need it - no problem at all.

olikraus, 2019-05-09:
> Seems it's 9px font.

glasstown (which is a bitmap font in a ttf file) is rendered correctly only with  "ftview 16 glasstown_nbp.ttf". So probably size has to be 16?
Well, difficult to tell whether your font size is correct, without seeing the actual result of your conversion.

Nevertheless, I see that the result is already quite good.

> PS. If you wish to compete - i'd recommend try with 0x20-0x7F (ascii) range. 

This is also available in u8g2: https://github.com/olikraus/u8g2/wiki/fntgrpnbp
The size will be 1013 bytes.

> lv_font_conv cares about first two, and leave decoding to libs authors. 

Well, fine, but I have an existing font format with a good decoder, but I do not have the time to write a decoder for your format. I probably would give it a try, if there is a C code decoder, which just breaks down all rendering to a pixel draw function.

But yes, nice project...

puzrin, 2019-05-09:
> glasstown (which is a bitmap font in a ttf file) is rendered correctly only with "ftview 16 glasstown_nbp.ttf". So probably size has to be 16?

There is difference between term "font size" in OS/browsers and real images dimentions. TTF font of XX size can have imaes of ANY size inside. It's better look at `accent`/`descent`/`em` metrics and inspect content in editors like fontforge.

That's why i dumped raw images first, to check those are not distorted and pixel perfect. I had no intent to cheat with file size by reducing glyph size :)

>  Well, difficult to tell whether your font size is correct, without seeing the actual result of your conversion.

There is dumper of raw images + some tests. 99% result should be correct.

I can point some places, where loss can happen in your case:

- internal references may be 16 bit, instead of 8 bit delta-coded in my case
- i suspect, it worth try to disable your compression

> > PS. If you wish to compete - i'd recommend try with 0x20-0x7F (ascii) range.
> 
> This is also available in u8g2: https://github.com/olikraus/u8g2/wiki/fntgrpnbp
> The size will be 1013 bytes.
>

Hm... 800 bytes in my case. I'm really surprised. To be honest, i had no intent to reach best results for bitonal fonts (those are "small enougth" for ARM-s). Main focus was on anti-aliased fonts of 16-24-32px sizes + kerning, to draw high quality modern interfaces like "Material design" on simple stm32 SOCs. Bitonal format (--bpp 1) was added without much care.
 
> Well, fine, but I have an existing font format with a good decoder, but I do not have the time to write a decoder for your format. I probably would give it a try, if there is a C code decoder, which just breaks down all rendering to a pixel draw function.

I understand. All we have time constrains. I'm not good C coder and prefer do things where i'm more effective. May be some of your users decide to write decoder.

Anyway, it was very interesting to compare sizes with your help. And your info about pixel-perfect TTF sources is very valuable. Thank you very much!

Should i close this issue?

olikraus, 2019-05-09:
Thanks for pointing out your project. Let me know if there is a C/C++ decoder.

closed 2019-05-09 by olikraus

puzrin, 2019-05-09:
Hm... for full picture, 16px dumps.

Full:

env DEBUG=* ./lv_font_conv.js --font glasstown_nbp.ttf -r 0x20-0x10FFFF --size 16 --bpp 1 --format bin -o test.bin
  font.table.head table size = 44 +0ms
  font.table.cmap 8 subtable(s): 7 "format 0", 1 "sparse" +0ms
  font.table.cmap table size = 1116 +5ms
  font.table.loca table size = 1768 +0ms
  font.table.glyf table size = 7956 +0ms
  font font size: 10884 +39ms
```

Ascii:

```
env DEBUG=* ./lv_font_conv.js --font glasstown_nbp.ttf -r 0x20-0x7F --size 16 --bpp 1 --format bin -o test.bin
  ...
  font.table.head table size = 44 +0ms
  font.table.cmap 1 subtable(s): 1 "format 0", 0 "sparse" +0ms
  font.table.cmap table size = 124 +0ms
  font.table.loca table size = 204 +0ms
  font.table.glyf table size = 700 +0ms
  font font size: 1072 +4ms
```

Probably you are right, `--size` should be 16. But result is still good enougth (and without compression, raw bitstreams of bounding boxes). Very small loss for ascii and win for multiple languages.