olikraus/u8g2 #891: winstar display is mirrored

instrumentek, 2019-05-11:
Hi;
I have a display that has a horizontal mirror effect. Other than the mirror everything seems to work well. I have tried:
- HW, SW SPI modes, 
- Both U8G2 and U8X8 examples 
- changed to 16 bit mode as recommended
Hardware:
- teensy 3.5
- winstar WO256x128-TFH
- It uses ST75256 controller
- It is not listed so I used "U8G2_ST75256_JLX256128_F_4W"

any suggestions would be appreciated

olikraus, 2019-05-11:
Locate u8x8_d_st75256 on your local hard disk. Goto line 87:
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st75256.c#L87
and line 307 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st75256.c#L307

In both locations change 
`U8X8_CA( 0xbc, 0x00 )`
to
`U8X8_CA( 0xbc, 0x01 )`

Will this solve your problem help?

instrumentek, 2019-05-11:
Hi and thanks for you idea, I tried as you described but now it creates random pixels in the lower 1/3 of screen. however some text moved to the correct location but it is not displayed correctly
Before change:
[image]
After change
[image]

olikraus, 2019-05-11:
hmm... can you use the value 0x004 instead?
I mean In both locations change
`U8X8_CA( 0xbc, 0x00 )`
to
`U8X8_CA( 0xbc, 0x04 )`

In general, you can play with values 0 to 7. If some lines are not written or show random patterns, this can be ignored for now (we will fix this later).

instrumentek, 2019-05-12:
So I did a bunch of testing. It looks like line 307 at 0X02 it appears to work fine except the screen is 180 degree rotation. line 87 doesn't seem to do anything. 

line 87 (0-7 tried)- when line 307 (0) -  no changes at all seen with graphics test

line 87 (7) line 307(2) 180 degree rotation NOT mirrored seems good!
line 87 (6) line 307(2) same as above
line 87 (5) line 307(2) same as above
line 87 (4) line 307(2) same as above
line 87 (3) line 307(2) same as above
line 87 (2) line 307(2) same as above
line 87 (1) line 307(2) same as above
line 87 (0) line 307(2) same as above

line 87 (7) line 307(3) 180 degree rotation NOT mirrored some screen display is jibberish 
line 87 (7) line 307(4) same as above
line 87 (7) line 307(5) same as above
line 87 (7) line 307(6) same as above
line 87 (7) line 307(7) same as above

can you tell me what I should look for with changing line 87? and is there a way to rotate the screen to 180 degrees? or is the rotation function the only options here? 

Thanks again!

olikraus, 2019-05-12:
Line 87 and 307 should do the same. I actually was not sure about line 87, so i mentioned both.
Line 87 and 307 must have the same values.
So you say, value 2 works good? `U8X8_CA( 0xbc, 0x02 )`
Are there any random pixel visible?

instrumentek, 2019-05-12:
Yes 0x02 seems to work fine no random pixels. Tried again with 0x01 also no random pixels but still errors in how it displayed, not sure what happened there the first time.  But one minor issue is it is 180 degrees out of orientation. I was hoping there was some way to deal with that without using the rotation function
[image]

olikraus, 2019-05-12:
ok, great, so 
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st75256.c#L307
is
`U8X8_CA( 0xbc, 0x02 )`

Lets try to fix the 180 rotation, can you change
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st75256.c#L311
to
`U8X8_C( 0x008 )` 

BTW: These are all internal configuration commands for the display. You can read details in the ST7256 datasheet.

instrumentek, 2019-05-12:
Hi;
I think i see what you did. In the data sheet I found:
 0 0 0 1 DO 0 0 DO=0; LSB on bottom (Default) DO=1; LSB on top
001100 = C
and
00100 = 8
you changed the LSB from the top to the bottom? I tried it and it did not work correctly, see the picture.
[image]

olikraus, 2019-05-12:
ok, looks odd.
What about this: 

https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st75256.c#L307
`U8X8_CA( 0xbc, 0x01 )`

Lets try to fix the 180 rotation, can you change
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st75256.c#L311
`U8X8_C( 0x008 )` 

I think it is this:
180 Degree rotation:
`U8X8_CA( 0xbc, 0x02 )`
`U8X8_C( 0x00c )` 

No rotation:
`U8X8_CA( 0xbc, 0x01 )`
`U8X8_C( 0x008 )`

instrumentek, 2019-05-12:
Close but I have the gap at the bottom now. I tried:
U8X8_CA( 0xbc, 0x01 )
U8X8_C( 0x008 )
(IN PICTURE)
[image]

and 
U8X8_CA( 0xbc, 0x00 )
U8X8_C( 0x008 )

olikraus, 2019-05-12:
Looks good. To fix the offset, set the default_x_offset to 5 (5*8=40 lines):
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st75256.c#L255

Do not worry, the x_offset is used as y-offset here.

I will create a new constructor, which will include all the changes.

olikraus, 2019-05-12:
Can you change the default x value from 0 to 5?
https://github.com/olikraus/u8g2/blob/master/csrc/u8x8_d_st75256.c#L255
Will it work?

Actually the variable name default_x_value is wrong. It is actually the y value.

olikraus, 2019-05-12:
New constructors created, e.g.:
`U8G2_ST75256_WO256X128_1_4W_HW_SPI u8g2(U8G2_R0, /* cs=*/ 10, /* dc=*/ 9, /* reset=*/ 8);`

I have created new beta 2.26.6

You can download the latest U8g2 beta release from here: https://github.com/olikraus/U8g2_Arduino/archive/master.zip
1. Remove the existing U8g2_Arduino library (https://stackoverflow.com/questions/16752806/how-do-i-remove-a-library-from-the-arduino-environment)
2. Install the U8g2_Arduino Zip file via Arduino IDE, add zip library menu (https://www.arduino.cc/en/Guide/Libraries).

instrumentek, 2019-05-12:
The offset worked! thanks. I have not yet tried the new library but as soon as i do i will reply to this thread

instrumentek, 2019-05-13:
Hi i tried the library and used the "U8G2_ST75256_WO256X128_1_4W_HW_SPI" 
It works mostly but see the picture not quite:
[image]

but the one we changed manually works fine:
[image]

olikraus, 2019-05-13:
Did you enable 16 bit mode? I actually do not see a difference in the device code itself.

instrumentek, 2019-05-13:
you are correct all is working well. 
Thanks for your help

olikraus, 2019-05-13:
Great. Final request: can you execute u8g2 flipmode example? Independent from the rotation, there must be a frame at the border of the display.

instrumentek, 2019-05-13:
looks good. 
[image]
[image]
[image]
[image]

olikraus, 2019-05-13:
Super. Looks perfect. Great cooperation. Thanks for the pictures.

closed 2019-05-13 by olikraus